schreibfaul1/ESP32-MiniWebRadio #362: REMOVE DAC!!!

clear-sky, 2024-03-07:
Hi dear wolle!
in v 3.0 you have used bt module named KCX_BT_EMITTER
i leave in iran ! Iran is sanctioned and does not have access to international stores such as Ali Express!!!
i could not  find this module in local stores so decided to use another module 
i found f-6988 that uses bk3266 chip -this module can be used as reciever or transmitter
but i found another nice ability in this module! it has I2S input !!! this means DAC can be removed and make radio much more simpler!!!
datasheet:
http://25009430.s21d-25.faiusrd.com/61/ABUIABA9GAAg1ZzY_gUokK6IgQU.pdf
https://github.com/tomaskovacik/kicad-library/blob/master/library/datasheet/F-6988_BK3266/Bluetooth.transmitter.and.receiver.integrated.audio.module.BK3266SR.manual.V1.0_HYT.zh-CN.en.pdf

very useful link:
https://github.com/tomaskovacik/BK3254

clear-sky, 2024-03-07:
advantages:
--configurable as receiver or transmitter (ESP32-S3 restriction in receive mode can be removed)
--can add phone call ability
--improving play local files 
--EQ 
--IR

schreibfaul1, 2024-03-07:
At first glance, it looks complicated. The board has many functions and connections, but not everything is needed. I think it could work.  
I used this in another project, it works without programming, the housing can be easily removed
[image]

closed 2024-03-07 by clear-sky