schreibfaul1/ESP32-MiniWebRadio #363: how to connect ILI 9488 with SD CARD

Tymonek87, 2024-03-09:
Hi,

How should I connect ILI 9488 with SD CARD?
I have an error with my SD card all the time.
Which  pin should I use

Arne65, 2024-03-11:
Hello,

I think you've already figured it out yourself, but here are the pins again.

Display                                PIN ESP32-S2    ESP32-S3
 
SD_CLK   = SD_MMC_CLK        14                  13
SD_MISO  = SD_MMC_D0         2                   11
SD_MOSI  = SD_MMC_CMD    15                  14

The three resistors on the display should actually be removed. For me it only works with the resistors on one display and on another they have to be removed.
You have to find that out yourself. You may need a 10K pullup on D0.

You should first minimize SDMMC_FREQUENCY to 20 MHz or possibly even smaller since not all SD cards work with 40 or 80 MHz.

Have fun testing :)

Tymonek87, 2024-03-11:
Hello,

Yes, i have figured out ;)
But there are some other changes... For example, I have VS1053 and you have
not supported it anymore :(
I have to buy DAC.
I am trying to run V2.15 but I have an issue with the connection to Wifi. I
have problems with the credentials ( even if they are good) and the
stability of AP 192.168.4.1 connection.

Regards,
Tom

Arne65, 2024-03-11:
I think you will use a dev kit with a PCB antenna. Since I've had similar problems before, I only use ESP with ext. Antenna. But that could also be due to my location.
Maybe your WiFi signal from your router is too weak. Try moving your setup closer to your router.
I can't say with certainty to what extent version 2.15 still works with the VS1053 because I only use DAC.

Tymonek87, 2024-03-11:
Hi,

Which DAC will the Best for this project ?

Regards
Tom

schreibfaul1, 2024-03-11:
Hi Tom, it depends on the availability in your country. A good choice is PCM5102. If you want to connect an amplifier (for speakers) behind it, it is important to use a sufficiently powerful power supply that provides separate voltages for analogue and digital.

Tymonek87, 2024-03-13:
Do I need BT emiter in case od using PCM5102
??

schreibfaul1, 2024-03-14:
The BT transmitter is optional, it also works without it.

closed 2024-04-29 by github-actions[bot]