schreibfaul1/ESP32-MiniWebRadio #364: Problem with downloading /displaying stations logo (icon)

ovidiu169, 2024-03-10:
Hello everybody !!!

I’m facing from the beginning of this project with the fact that the radio says that the logo was downloaded on sd card, but is not appearing on the screen. Don’t tell me it’s a lack of PSRAM, because it’s wrong.  I have enough PSRAM. Moreover the process of displaying the logos it’s absolutely aleatory. Some are displayed, some are not. When the icons downloaded are not displayed, the radio displays the format icon instead… What’s happening in fact ?!?

ovidiu169, 2024-03-10:
[image]
[image]
[image]
[image]
[image]

steve6375, 2024-03-10:
Tick Tock Radio - 1989.jpg   is  NOT the same as Tick Tock Radio 1989.jpg 

So the name of the jpg is wrong?

ovidiu169, 2024-03-10:
Hi, Steve !
This is the name of jpeg downloaded. It not depends of me and I did not edited at all. 
Another issue for me is to understand why are two files with jpegs. One is called “m” and the other is “s”. I don’t get it…
I Will show you other examples soon.

ovidiu169, 2024-03-10:
Hi, Steve !
Look at this example - “Exclusively Kenny Rogers”. It has a “Exclusively Kenny Rogers.jpeg” downloaded to sd card on logo file, but is not displaying on the screen…☹️

ovidiu169, 2024-03-10:
[image]
[image]
[image]

steve6375, 2024-03-10:
m = medium, s = small
The name of the jpg must match the name of the station
If you look - the Exclusively Kenny Rogers .jpg has a space after the word Rogers
[image]

ovidiu169, 2024-03-10:
It worked for this station. But it’s not working for the others.  And what I don’t get it is the fact that I downloaded “as it is” from the station site. What could be wrong ?!?😑

steve6375, 2024-03-10:
The names must match exactly. Unless you show us the filenames and radio station names, how can we tell?

Unfortunately, I cannot look over your shoulder 😭

ovidiu169, 2024-03-10:
I double checked Al the file names accordingly. It’s still not working. Thank you for your kind support.
I felt from the very beginning this radio has a lot of enigmas…🥸

steve6375, 2024-03-10:
Well your eyesight seems questionable. You do not seem to be able to see hyphens or spaces.
So unless you show us the radio station names and the jog file names, it is difficult for us to check!
For instance a is different from A and a° or a^

ovidiu169, 2024-03-10:
Ok I will show you all the lists…

ovidiu169, 2024-03-10:
[image]
[image]
[image]
[image]

ovidiu169, 2024-03-10:
I corrected as good I could. Some of them are still not displaying… But I got your point. The file downloaded is very tricky - it has not all the time the same title as the station saved… Someone could tell as this…
Another problem is how to change the credentials when we have to connect to another network. Of course without downloading the sketch again . I read that is possible but I don’t imagine how, because I tried that and my radio is saying only the credentials are not correct…

steve6375, 2024-03-10:
Which icons don't work? Are we supposed to guess?

steve6375, 2024-03-10:
You know you can take out the SD card and connect it to an SD card reader and check the file names and stations.ini file contents using an editor like notepad, right?

ovidiu169, 2024-03-10:
One of them is iHeart. But maybe I missed an apostrophe. 
No, I didn’t know. I tried with Word and did not work. Someone told me to try Sublime Text as editor…

schreibfaul1, 2024-03-10:
The idea is that a station is selected via the radio browser. A logo is often included. If not, you can select an image via Google Image Search. You then enter the URL found as LogoUrl. The image is automatically scaled to the correct size [image] Click the up arrow to transfer the image to the SD card. When saving the station data, the station name and the name of the image on the SD card must be identical.
[image]
You can then do this with additional stations, but the table must then be saved on the SD card. The new stations are transferred to the NVS with "Load"
[image]

steve6375, 2024-03-10:
Well remove the apostrophe then!

ovidiu169, 2024-03-10:
Hi, Wolle !
I followed always the procedure you described. Usually, I am not selecting stations which have no logo. I have loaded everytime the new station on NVS. I downloaded logos with Google Image Search on my computer, but I have no clue how to send them on the SD card.
My mistake was I never imagine that the name of the station and that one of the jpeg are not identic…

ovidiu169, 2024-03-10:
I don’t expect. I was expecting that the icon to be somehow related with the station name. Otherwise why someone would deliver an icon which is not related with  ?!?… Sorry to be so naiv…

Arne65, 2024-03-10:
Hello

You either copy the station logos to the SD card under logos/m or you use filezilla. Please pay attention to them

ovidiu169, 2024-03-10:
Thank you, Arne ! That’s I will do !!!

steve6375, 2024-03-10:
#255 should have fixed space character bug? Has it regressed @Wolle?

schreibfaul1, 2024-03-11:
I will think about it. If the logo is not found, then the smurf or an mp3/aac logo is displayed.
I have set a different Cors proxy so that the logos are displayed quickly on the "radio browser page".

ovidiu169, 2024-03-11:
After I checked all the jpegs, everything was OK. Last evening.
I the morning, when I tried to access the browser, it didn’t work.
I used the second MiniwebRadio I have, and finally worked.
But, the problem is the “blank” picture of logo, nevertheless the station has a logo, the radio doesn’t see that..,
[image]
[image]

steve6375, 2024-03-11:
You can save the jpg file to the SD card using the file browser in the web UI.
Note that you must have only ONE web browser page open to the webradio ip address (192.168.0.213 in your case). make sure no other browser or other tab has the same address or you will get strange behaviour.

ovidiu169, 2024-03-11:
Of course I stopped the other one radio. I tried to copy/paste the picture on the “s” file, but it’s not possible 🙁

schreibfaul1, 2024-03-11:
It is not possible to copy the image of a website into your own. This is prevented by the "cors rule". A proxy must therefore be interposed. Sometimes the proxies are overloaded. I have chosen another one,
[image]

 which seems to be faster. It is not enough to update the index.h file, the browser must be forced to load the new web page.

ovidiu169, 2024-03-11:
OK !!!👍

closed 2024-04-25 by github-actions[bot]