schreibfaul1/ESP32-MiniWebRadio #365: Uhrzeit nach jedem Neustart verstellt

acer1111, 2024-03-10:
Hallo 
Vielen Dank für das großartige Projekt .
Ich habe es in verschiedenen Versionen nachgebaut .
Im Moment nutze ich es mit einem ESP32S3
N16R8 mit der Version 3 des Webradios.
Läuft soweit gut , nur die Uhrzeit ist nach jedem Neustart um eine Stunde nach vorn verstellt .
Woran kann das liegen? 
In der Version 2 war der Fehler etwas anders , dort verstellte sich die Uhrzeit nach einer gewissen Betriebszeit von selbst im mehrer Stunden , jedoch nach dem Neustart war sie zunächst wieder korrekt.
Kann  mir eventuell jemand sagen was ich machen kann das es funktioniert?

Linkenelis, 2024-03-11:
In webpage "I" button. At the bottom Timezone

Standard is Europe/Berlin time, I guess Wolle is on that time and I am lucky to be on the same time.
Or change it manually in main.cpp line 140: String              _TZString = "CET-1CEST,M3.5.0,M10.5.0/3";
in rtime.cpp you will find the stings
[image]

acer1111, 2024-03-12:
Vielen Dank für eure Antworten erst einmal!
Bin leider noch nicht zum Testen gekommen .
Ich weiß aber eins , die timezone ist in der settings.json gespeichert das habe ich schon überprüft .
Ich werd nochmal weiter schauen ob ich eine Ursache finde.
Leider hab ich vom programmieren nicht sehr viel Ahnung, bin er der Nachbautyp

acer1111, 2024-03-12:
Könnte es vielleicht an der Sommer-und Winterzeit Umstellung  liegen ?
Nur so eine Idee.
Ich weis aber leider nicht wie man das integrieren kann!

closed 2024-04-26 by github-actions[bot]