schreibfaul1/ESP32-MiniWebRadio #367: Bluetooth Botton ESP32-S2 no Function

Arne65, 2024-03-10:
Hi Wolle, great job.

Unfortunately, in V3 ESP32-S2 the Bluetooth receiver button has no function.
[image]

schreibfaul1, 2024-03-11:
Thanks for the hint, I have deactivated the button.

Arne65, 2024-03-11:
Hello Wolle,

Thanks for the quick reply.

Is there a reason why you disabled the button?

I use the ESP 32 and it worked great with the transmitter and BT from the ESP. The transmitter makes little sense to me. When using a BT speaker and playing files from the mobile device, the audio comes out in analogue form, so the receiver mode on the KCX Emitter has a huge disadvantage compared to the BT button on the MiniWebRadio.

schreibfaul1, 2024-03-11:
Hello Arno,
the previous "a2dp-sink" solution also has limitations. WiFi and BT use the same frequency range, the ESP32 cannot process both at the same time. And the S3 can only process BLE. In future, the BT emitter will be integrated via the display and will then offer roughly the same functionality as before. I don't have the time to implement this in the short term.

closed 2024-04-25 by github-actions[bot]