schreibfaul1/ESP32-MiniWebRadio #368: Station-Icon Problems in Webbrowser

HWH53, 2024-03-11:
Hello,
If configured and named correctly, the station icons are displayed on the TFT display, but not all icons are displayed in the web browser. The missing icons are than displayed as MP3.
Where can I find the differences between the display of the icons on the TFT display and the display in the web browser?
I'm currently still using version Radio V2 until I have the appropriate new hardware. 
Thanks for your feedback.
HWH53

schreibfaul1, 2024-03-11:
There is a logo folder on the SD card. The logos are stored there depending on the display size. Folder "s" is for displays with a resolution of 240x320px, folder "m" for 320x480px displays. You are simply asked whether there is a file that is identical to the station name. If so, this will be loaded.

HWH53, 2024-03-11:
Thanks for quick answer,
The logos are in the 's' folder because the display is 320*240. The file names are correctly assigned to the station names. I also copied them into the 'm' folder, same result. They also appear correctly on the TFT display, but some logos are not displayed in the browser on the website. Why? What are the differences between the selection for the display and the presentation on the website?
Thanks for the support!
HWH53

schreibfaul1, 2024-03-11:
Strange, maybe it helps to clear the browser cache

HWH53, 2024-03-12:
Thanks,
Problem solved. 
Browsercache was the problem!
Sorry, manchmal sieht man den Wald vor lauter Bäumen nicht;;;;

closed 2024-03-12 by HWH53