schreibfaul1/ESP32-MiniWebRadio #371: How to increase Low, Band, High

hevet, 2024-03-20:
In what file should something be changed so that these values could be increased to e.g. 15dB, and not +6dB as before?

schreibfaul1, 2024-03-20:
The limits are defined in `IIR_calculateCoefficients(int8_t G0, int8_t G1, int8_t G2)`. +15dB is a lot.

hevet, 2024-03-20:
So I change the I2S library?
if(G0 < -40) G0 = -40; // -40dB -> Vin*0.01
     if(G0 > 12) G0 = 12; // +6dB -> Vin*2
     if(G1 < -40) G1 = -40;
     if(G1 > 12) G1 = 12;
     if(G2 < -40) G2 = -40;
     if(G2 > 12) G2 = 12;

But in the web interface I still have a maximum of +6B.

hevet, 2024-03-21:
@schreibfaul1 Could you tell me what to change so that it can also be changed in the WEB interface?

schreibfaul1, 2024-03-22:
The above was just an example of how the filters are mounted in the audioI2S library. This is just maths. You can develop your own filters with completely different properties and filter curves.
The sliders and their ranges are defined in index.h.

closed 2024-05-06 by github-actions[bot]