schreibfaul1/ESP32-MiniWebRadio #373: Changing Display Brightness Error Message at Console and wrong Timezone at boot

HWH53, 2024-03-21:
Hello Wolle,
Thank you in advance for your excellent work on this radio project!
After the hardware upgrade, I have now switched to version 3.0. 

There are 2 small discrepancies for me:

1. Error messages on the console when I change the display brightness. The function is there, but unfortunately there are error messages.
See attached screenshots.
[image]
[image]

2. The time is 1 hour ahead every time you restart. This can only be corrected by changing the time zone via the website.
This is then always set to "Africa/Abijan", the first entry in the time zone list. The corect entry for the Timezone "Europa/Berlin" at the SD-Card i've checked.
[image]

Thanks in advance for the support.
Helmut

PS: Answers in Language German are welcome to me.

schreibfaul1, 2024-03-22:
Hallo Helmut,
das `ledcAttach(TFT_BL, 1200, 8); // 1200 Hz PWM and 8 bit resolution` für die Helligkeitssteuerung braucht nur einmal gesetzt zu werden, ich hoffe der Fehler ist nun beseitigt.

Das andere Problem ist schwieriger. Vermutlich ist eine alte  settings.json die Ursache.
Bitte lösche hier die Datei, 
[image]

gehe anschließend zu den Zeitzonen, wähle irgendeine aus und anschließend die für Europa/Berlin.
Dadurch wird die settings.json neu angelegt. Und drücke danach die Resset Taste.

HWH53, 2024-03-22:
Hallo Wolle,
die Änderungen in (Issue #373) funktionieren. Die lästigen Meldungen auf der Konsole sind somit behoben. Danke nochmals für den schnellen Support.
Zum Problem mit der Zeitzone bin dich deinem Vorschlag gefolgt und habe die "settings.json" gelöscht und diese neu anlegen lassen. Anscheinend hat dies funktioniert, obwohl ich keinen Unterschied zwischen der alten und der neuen "settings.json" feststellen kann. Die Uhrzeit ist derzeit korrekt. 
Sehr verwirrend ist die Ausgabe der Zeitzone auf der Webseite, bei jedem erneuten Öffnen der Webseite wird dort immer der erste Eintrag aus der Timezone-Liste  angezeigt. Wird dieses Feld nicht mit dem  aktuell hinterlegten Wert neu eingelesen? Ist ja eigentlich kein Problem, sondern nur Kosmetik.
Danke nochmals für schnelle Unterstützung
Helmut

schreibfaul1, 2024-03-22:
Auf der Webseite soll die zuletzt eingestellte Zeitzone erscheinen. Das muss ich dann nochmal genauer analysieren.