schreibfaul1/ESP32-MiniWebRadio #374: Wifi credentials are not correct

RobinMosedale, 2024-03-21:
Schreibfaul ,

Thank you for your efforts and responses so far:

It's a lengthy battle, and I'm still losing.

Having started with a standard no PSRAM, A VS1053, and an ILI9341 with MMC resistors, and stumbled serially with each of those impediments in turn.

That took several weeks of discovery that VS1053 was no longer supported, the experimentation with SD card direct connection, and the need to transition to  a DC
So the current configuration is:-
ESP32S3N16R8
ILI9341
PCM5102
MMC on ILI8341 has no resistors.

I'm struggling. As discussed before my only motive is to receive BBC  playlist streaming.  A basic functionality of ESP32N16r8/ILI9341+touch+PCM5102 with BBC URL's embedded would be sufficient, as your colleague's kitchen basic radio. SD Playlists, BT, IR control are all good, but not necessary
If you could point me to build to that I'd be more than grateful
The first thing to note is:
Startup page showing only half image
[image]

Main Issue: WiFi Credentials are not correct:
[image]

In addition, Message seems to be different colour to that in main.ccp

The access point is unobtainable, and not registering in any form on my router. I can't see whether it is indeed entering the access point code in the Wifi connect part of main.cpp

The Wifi is 2.4Ghz (Indeed initial attempts were made with 5.2GHz until it was discovered that the S3 can't operate on the frequency
The WiFi capability has been checked with a simple standalone Wifi connect on both hotspots.
SSID and Passwords checked. They have no unusual characters, one is a 4character SSID no spaces, no none alphanumeric. One has a space.
Attempted in proximity to Hotspots, less than 30cms adjacent.
Attached is Common.h

`// created: 10.Feb.2022
// updated: 08.Feb 2024

#pragma once
#pragma GCC optimize("Os") // optimize for code size

#define _SSID               "Xxxx"                        // Your WiFi credentials here
#define _PW                 "xxxxxxxxxxx"                // Or in textfile on SD-card
#define DECODER             1                               // (1)MAX98357A PCM5102A CS4344... (2)AC101, (3)ES8388, (4)WM8978
#define TFT_CONTROLLER      0                               // (0)ILI9341, (1)HX8347D, (2)ILI9486a, (3)ILI9486b, (4)ILI9488, (5)ST7796, (6)ST7796RPI
#define DISPLAY_INVERSION   0                               // (0) off (1) on
#define TFT_ROTATION        3                               // 1 or 3 (landscape)
#define TFT_FREQUENCY       40000000                        // 80000000, 40000000, 27000000, 20000000, 10000000
#define TP_VERSION          0                               // (0)ILI9341, (1)ILI9341RPI, (2)HX8347D, (3)ILI9486, (4)ILI9488, (5)ST7796, (3)ST7796RPI
#define TP_ROTATION         3                               // 1 or 3 (landscape)
#define TP_H_MIRROR         0                               // (0) default, (1) mirror up <-> down
#define TP_V_MIRROR         0                               // (0) default, (1) mittor left <-> right
#define AUDIOTASK_CORE      0                               // 0 or 1
#define AUDIOTASK_PRIO      2                               // 0 ... 24  Priority of the Task (0...configMAX_PRIORITIES -1)
#define I2S_COMM_FMT        0                               // (0) MAX98357A PCM5102A CS4344, (1) LSBJ (Least Significant Bit Justified format) PT8211
#define SDMMC_FREQUENCY     20000000                        // 80000000, 40000000, 27000000, 20000000, 10000000 not every SD Card will run at 80MHz
#define FTP_USERNAME        "esp32"                         // user and pw in FTP Client
#define FTP_PASSWORD        "esp32"
#define CONN_TIMEOUT        5000                             // unencrypted connection timeout in ms (http://...)
#define CONN_TIMEOUT_SSL    5000                            // encrypted connection timeout in ms (https://...)
#define BT_SINK_NAME        "MiniWebRadio"                  // ESP32 only

/**********************************************************************************************************************/

#include <Arduino.h>
#include <ArduinoOTA.h>
#include <Preferences.h>
#include <Ticker.h>
#include <SPI.h>
#include <SD_MMC.h>
#include <FS.h>
[... 302 lines cut ...]
    https://github.com/schreibfaul1/ESP32-DLNA-Client.git
    https://github.com/schreibfaul1/ESP32-IR-Remote-Control.git

;______________________________________________________________________________________________________________________________________________-
[env:esp32]
    board = ${esp32.board}
    board_build.partitions = ${esp32.board_build.partitions}
;______________________________________________________________________________________________________________________________________________-
[env:esp32s3]
    board = ${esp32s3.board}
    board_build.partitions = ${esp32s3.board_build.partitions}
;______________________________________________________________________________________________________________________________________________-
[env:esp32_ota]
    board = ${esp32.board}
    board_build.partitions = ${esp32.board_build.partitions}
    upload_protocol = espota ;update code over WiFi
    upload_port = "MiniWebRadio.local"
;______________________________________________________________________________________________________________________________________________-
[env:esp32s3_ota]
    board = ${esp32s3.board}
    board_build.partitions  = ${esp32s3.board_build.partitions}
    upload_protocol = espota ;update code over WiFi
    upload_port = "MiniWebRadio.local"

`

Arne65, 2024-03-21:
Hello, 
I have the following setting for ILI 9341

#define TFT_ROTATION        3                          
#define TP_VERSION           1                             
#define TP_ROTATION         1     

This is how the touch works

BBrz, 2024-03-21:
@RobinMosedale I have the same hardware config as you.
It connects to Wi-Fi (2.4GHz) and works ok, but I cannot acces it from the browser (laptop or phone).
I've used the wire connections from here: https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/338

RobinMosedale, 2024-03-21:
Thank you Arne65. I did manage to get it to log on twice and obtained the operation window and did something to touch. No more, so I'm not getting to the operational page any more

RobinMosedale, 2024-03-21:
IBBrz, The connections are now automatically generated by setting the processor, esp32N16R8, PCM5012 and ILI99341 (correct variant). They are nearly identical to your (MMC slightly different, but MMC seems fine).

What puzzles me is why the USB becomes unrecognizable when running, but fine in Burn/Boot mode. If I could get terminal output, I'd be able to see what's going on or inserting some serial messages

schreibfaul1, 2024-03-22:
If this happens: [image] this is often a contact problem with the SD_MMC. Please check the wiring again, especially the solder joints. And if these resistors exist,  
[image]
 it is better to bridge them

hevet, 2024-03-22:
@schreibfaul1 It's the same for me, and I have an adapter soldered directly to the pins. I loaded version 2 and the logo is displayed correctly. The terminal displays this error:
[  1164][E][tft.cpp:3782] decode_mcu(): pjpeg_decode_mcu() failed with status 29

schreibfaul1, 2024-03-22:
thank you, I was able to reproduce this, the `MiniWebRadioV3.jpg` image for the small displays is defective. I have replaced it. Please unpack the `content_on_SD_card.zip` and overwrite the file in SD_MMC  /common/s

RobinMosedale, 2024-03-22:
Thank you Shreibfaul1,

Welcome screen now fixed. Therefore the MMC card has been reading successfully and not a contributor to any of the faults.

Double checked Router and Wifi settings, and all 2.4GHz, well within range. Default access point 30cm adjacent. Two successful logons last night, and obtained operational screen, with an IP of 192.168.1.212. Dunno where the 192.168.4.1 displayed comes from, not accessible and not identifiable on my network. Even with the 192.168.1.212 shown, unable to log on to the ESP32 webserver at that address.
Credentials wrong still displayed. 4 access points on the SD card.

Connection time increased to
#define CONN_TIMEOUT        8000                             // unencrypted connection timeout in ms (http://...)
#define CONN_TIMEOUT_SSL    8000                            // encrypted connection timeout in ms (https://...)
although I suspect that these having nothing to do with initial logon
Still no serial terminal activity
[image]
[image]

schreibfaul1, 2024-03-22:
Please replace the main.cpp with the example from Espressif. https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiMulti/WiFiMulti.ino Does this work?

RobinMosedale, 2024-03-22:
Esp32ExceptionDecoder: disabling, firmware at D:\ESP32-MiniWebRadio\.pio\build\esp32s3\firmware.elf does not exist, rebuild the project?
--- Terminal on COM19 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Disconnected (ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22)))
Reconnecting to COM19 
Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

Esp32ExceptionDecoder: disabling, firmware at D:\ESP32-MiniWebRadio\.pio\build\esp32s3\firmware.elf does not exist, rebuild the project?
         Connected!
[   210][I][esp32-hal-psram.c:92] psramInit(): PSRAM enab[  4243][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  5369][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  6394][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  7418][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  8186][E][WiFiMulti.cpp:311] run(): [WIFI] Connecting Failed (6).
[  8195][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 202 - AUTH_FAIL
[ 13983][I][WiFiMulti.cpp:133] run(): [WIFI] scan done
[ 13990][I][WiFiMulti.cpp:138] run(): [WIFI] 5 networks found
[ 13996][I][WiFiMulti.cpp:262] run(): [WIFI] Connecting BSSID: D8:0D:17:EF:41:B6 SSID: Roise_4 Channel: 1 (-78)
[ 15091][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 16115][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 17139][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 18265][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 19050][E][WiFiMulti.cpp:307] run(): [WIFI] Connecting Failed.
[ 19290][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 20059][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 202 - AUTH_FAIL
[ 25847][I][WiFiMulti.cpp:133] run(): [WIFI] scan done
[ 25853][I][WiFiMulti.cpp:138] run(): [WIFI] 5 networks found
[ 25860][I][WiFiMulti.cpp:262] run(): [WIFI] Connecting BSSID: 00:0F:94:D5:A2:14 SSID: Rose Channel: 9 (-83)
[ 26966][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 27990][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 29014][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 30038][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 30914][E][WiFiMulti.cpp:307] run(): [WIFI] Connecting Failed.
[ 31062][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 31923][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 202 - AUTH_FAIL
[ 37711][I][WiFiMulti.cpp:133] run(): [WIFI] scan done
[ 37717][I][WiFiMulti.cpp:138] run(): [WIFI] 5 networks found
[ 37724][E][WiFiMulti.cpp:316] run(): [WIFI] no matching wifi found!
[ 41520][I][WiFiMulti.cpp:133] run(): [WIFI] scan done
[ 41526][I][WiFiMulti.cpp:138] run(): [WIFI] 5 networks found
[ 41532][I][WiFiMulti.cpp:262] run(): [WIFI] Connecting BSSID: 60:E3:27:EB:85:0C SSID: TP-LINK_EB850C Channel: 1 (-40)
[ 42645][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 43771][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 44796][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 45821][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 46587][E][WiFiMulti.cpp:307] run(): [WIFI] Connecting Failed.
[ 46946][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 47596][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 202 - AUTH_FAIL
[ 53385][I][WiFiMulti.cpp:133] run(): [WIFI] scan done
[ 53391][I][WiFiMulti.cpp:138] run(): [WIFI] 6 networks found
[ 53397][I][WiFiMulti.cpp:262] run(): [WIFI] Connecting BSSID: 00:0F:94:D5:A2:14 SSID: Rose Channel: 9 (-82)
[ 54513][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 55536][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 58142][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 58495][E][WiFiMulti.cpp:303] run(): [WIFI] Connecting Failed AP not found.
[ 59223][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 59505][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 202 - AUTH_FAIL
[ 65294][I][WiFiMulti.cpp:133] run(): [WIFI] scan done
[ 65300][I][WiFiMulti.cpp:138] run(): [WIFI] 5 networks found
[ 65306][I][WiFiMulti.cpp:262] run(): [WIFI] Connecting BSSID: D8:0D:17:EF:41:B6 SSID: Roise_4 Channel: 1 (-83)
[ 66396][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 67522][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 68547][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 69673][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 70361][E][WiFiMulti.cpp:307] run(): [WIFI] Connecting Failed.
[ 70697][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 71370][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 202 - AUTH_FAIL
[ 77159][I][WiFiMulti.cpp:133] run(): [WIFI] scan done
[ 77165][I][WiFiMulti.cpp:138] run(): [WIFI] 6 networks found
[ 77171][E][WiFiMulti.cpp:316] run(): [WIFI] no matching wifi found!
[... cut ...]
[ 83096][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 84119][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 85245][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 86034][E][WiFiMulti.cpp:307] run(): [WIFI] Connecting Failed.
[ 86270][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE

https://github.com/espressif/arduino-esp32/issues/8142

I'll revert to the simplest Arduino build, which has been successful and report back

RobinMosedale, 2024-03-22:
Bare Esp32N16R8.
Rerun your example main with the same result.

ELF file SHA256: d4978be755caf041

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376d71
=> 0x40376d71: esp_restart_noos at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158 (discriminator 1)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
=> 0x403c9700: ?? ??:0
load:0x403c9704,len:0xad0
=> 0x403c9704: ?? ??:0
load:0x403cc700,len:0x29d8
=> 0x403cc700: ?? ??:0
entry 0x403c9880
=> 0x403c9880: ?? ??:0
[   209][I][esp32-hal-psram.c:92] psramInit(): PSRAM enabled
[   248][I][WiFiMulti.cpp:89] addAP(): [WIFI][APlistAdd] add SSID: TP-LINK_EB850C
[   256][I][WiFiMulti.cpp:89] addAP(): [WIFI][APlistAdd] add SSID: Roise_4
[   264][I][WiFiMulti.cpp:89] addAP(): [WIFI][APlistAdd] add SSID: Rose
[  2908][I][WiFiMulti.cpp:133] run(): [WIFI] scan done
[  2914][I][WiFiMulti.cpp:138] run(): [WIFI] 4 networks found
[  2920][I][WiFiMulti.cpp:262] run(): [WIFI] Connecting BSSID: 60:E3:27:EB:85:0C SSID: TP-LINK_EB850C Channel: 1 (-45)
[  4020][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  5044][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  6068][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  7092][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  7970][E][WiFiMulti.cpp:311] run(): [WIFI] Connecting Failed (6).
[  7980][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 202 - AUTH_FAIL
[ 13768][I][WiFiMulti.cpp:133] run(): [WIFI] scan done
[ 13774][I][WiFiMulti.cpp:138] run(): [WIFI] 5 networks found
[ 13781][I][WiFiMulti.cpp:262] run(): [WIFI] Connecting BSSID: D8:0D:17:EF:41:B6 SSID: Roise_4 Channel: 1 (-63)
[ 14834][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 15858][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 16882][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 17906][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 18829][E][WiFiMulti.cpp:307] run(): [WIFI] Connecting Failed.
[ 18930][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 19839][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 202 - AUTH_FAIL
[ 25627][I][WiFiMulti.cpp:133] run(): [WIFI] scan done
[ 25633][I][WiFiMulti.cpp:138] run(): [WIFI] 5 networks found
[ 25639][I][WiFiMulti.cpp:262] run(): [WIFI] Connecting BSSID: 00:0F:94:D5:A2:14 SSID: Rose Channel: 9 (-87)
[ 26772][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 27795][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 30396][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 30689][E][WiFiMulti.cpp:303] run(): [WIFI] Connecting Failed AP not found.
[ 31482][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 31699][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 202 - AUTH_FAIL
[ 37288][I][WiFiMulti.cpp:133] run(): [WIFI] scan done
[ 37294][I][WiFiMulti.cpp:138] run(): [WIFI] 4 networks found
[ 37301][E][WiFiMulti.cpp:316] run(): [WIFI] no matching wifi found!
[ 40896][I][WiFiMulti.cpp:133] run(): [WIFI] scan done
[ 40902][I][WiFiMulti.cpp:138] run(): [WIFI] 4 networks found
[ 40908][I][WiFiMulti.cpp:262] run(): [WIFI] Connecting BSSID: 60:E3:27:EB:85:0C SSID: TP-LINK_EB850C Channel: 1 (-41)
[ 42012][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 43036][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 44060][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 45084][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 45959][E][WiFiMulti.cpp:307] run(): [WIFI] Connecting Failed.
[ 46108][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[ 46969][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 202 - AUTH_FAIL

There must be some fault in multiwifi

Suggest a simple wifi logon in main.cpp and I'll try it

RobinMosedale, 2024-03-23:
Looks as though we need a station close before retrying

RobinMosedale, 2024-03-24:
Re-emphasising issue with WifiMulti, fails to connect:

I have extensively checked the hardware under suspicion of it having a weak wifi receiver on another thread.

I have run tests back to back.

They are under Arduino builds as they are simple

The first is a simple WiFi.begin, with a time to connect:
The chip connects consistently in 500ms

The test is repeated employing the WiFiMulti.run() and it never connects

This seems fairly convincing evidence that on the same hardware, running in the same environment, within the same timeframe, one works, but the other fails. Does this suggest something odd with WiFiMulti?

If you would like further test runs or evidence to isolate please let me know
This is the simple wiFi.begin() programme

`#include <WiFi.h>

const char* ssid = "TP-LINK_EB850C";
const char* password = "blanked_out";

int btnGPIO = 0;
int btnState = false;
unsigned long startTime;
unsigned long elapsedTime;

void setup()
{
Serial.begin(115200);
delay(10);

// Set GPIO0 Boot button as input
pinMode(btnGPIO, INPUT);

// We start by connecting to a WiFi network
// To debug, please enable Core Debug Level to Verbose

Serial.println();
Serial.print("[WiFi] Connecting to ");
Serial.println(ssid);

WiFi.begin(ssid, password);
// Auto reconnect is set true as default
// To set auto connect off, use the following function
// WiFi.setAutoReconnect(false);
startTime=millis();
// Will try for about 10 seconds (20x 500ms)
int tryDelay = 500;
int numberOfTries = 20;

// Wait for the WiFi event
while (true) {
    
    switch(WiFi.status()) {
      case WL_NO_SSID_AVAIL:
        Serial.println("[WiFi] SSID not found");
        break;
      case WL_CONNECT_FAILED:
        Serial.print("[WiFi] Failed - WiFi not connected! Reason: ");
        return;
        break;
      case WL_CONNECTION_LOST:
        Serial.println("[WiFi] Connection was lost");
        break;
      case WL_SCAN_COMPLETED:
        Serial.println("[WiFi] Scan is completed");
        break;
      case WL_DISCONNECTED:
        Serial.println("[WiFi] WiFi is disconnected");
        break;
      case WL_CONNECTED:
        Serial.print("Time to connect - ms: ");
        elapsedTime=millis()-startTime;
        Serial.println(elapsedTime);
        Serial.println("[WiFi] WiFi is connected!");
        Serial.print("[WiFi] IP address: ");
        Serial.println(WiFi.localIP());
        return;
        break;
      default:
        Serial.print("[WiFi] WiFi Status: ");
        Serial.println(WiFi.status());
        break;
    }
    delay(tryDelay);
    
    if(numberOfTries <= 0){
      Serial.print("[WiFi] Failed to connect to WiFi!");
      // Use disconnect function to force stop trying to connect
      WiFi.disconnect();
      return;
    } else {
      numberOfTries--;
    }
}
}

void loop()
{
// Read the button state
btnState = digitalRead(btnGPIO);

if (btnState == LOW) {
  // Disconnect from WiFi
  Serial.println("[WiFi] Disconnecting from WiFi!");
  // This function will disconnect and turn off the WiFi (NVS WiFi data is kept)
  if(WiFi.disconnect(true, false)){
    Serial.println("[WiFi] Disconnected from WiFi!");
  }
  delay(1000);
}
}`

This is the output:

[WiFi] Connecting to TP-LINK_EB850C
[WiFi] WiFi is disconnected
Time to connect - ms: 500
[WiFi] WiFi is connected!
[WiFi] IP address: 192.168.1.212

This is the WiFiMulti programme:

`/*
 *  Based on the following examples:
 *  WiFi > WiFiMulti: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiMulti/WiFiMulti.ino
 *  WiFi > WiFiScan: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiScan/WiFiScan.ino
 *  Complete project details at our blog: https://RandomNerdTutorials.com/
 *  
 */

#include <WiFi.h>
#include <WiFiMulti.h>

WiFiMulti wifiMulti;

// WiFi connect timeout per AP. Increase when connecting takes longer.
const uint32_t connectTimeoutMs = 10000;

void setup(){
  Serial.begin(115200);
  delay(10);
  WiFi.mode(WIFI_STA);
  
  // Add list of wifi networks
  wifiMulti.addAP("ESP32_NAT_Router", "Blanked_out");
  wifiMulti.addAP("TP-LINK_EB850C", "Blanked_out");
  wifiMulti.addAP("Rosem", "Blanked_out");
  wifiMulti.addAP("Roie_4", "Blanked_out");
  wifiMulti.addAP("Rose", "Blanked_out");
  wifiMulti.addAP("Rosem", "Blanked_out");
  wifiMulti.addAP("Rosie_3", "Blanked_out");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0) {
      Serial.println("no networks found");
  } 
  else {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i) {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
      delay(10);
    }
  }

  // Connect to Wi-Fi using wifiMulti (connects to the SSID with strongest connection)
  Serial.println("Connecting Wifi...");
  if(wifiMulti.run() == WL_CONNECTED) {
    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }
}

void loop(){
  //if the connection to the stongest hotstop is lost, it will connect to the next network on the list
  if (wifiMulti.run(connectTimeoutMs) == WL_CONNECTED) {
    Serial.print("WiFi connected: ");
    Serial.print(WiFi.SSID());
    Serial.print(" ");
    Serial.println(WiFi.RSSI());
  }
  else {
    Serial.println("WiFi not connected!");
  }
  delay(1000);
}`

This is the output (truncated......................)
WiFi not connected!
WiFi not connected!
WiFi not connected!
WiFi not connected!
WiFi not connected!
WiFi not connected!
WiFi not connected!
WiFi not connected!
WiFi not connected!
WiFi not connected!
WiFi not connected!

RobinMosedale, 2024-03-24:
Lastly, I'll attempt to replace the ESP32N16R8 with another. However, originals are hard to find in the Uk, most being a close clone
When achieved, I'll repeat.
A search of GITHUb throws up several references to WiFiMulti failingin a loop

RobinMosedale, 2024-03-28:
[image]
Wolle,

I am very pleased to let you know, that my +Replacement ESP32N32R8 Devkitc-1 arrived this afternoon.
The issues with AUTH Failure is due to a poor antenna on the original. Similarly failure to raise an AP is likely to be the pooor antenna too.

I noticed that the original ESP32 would connect if I held a finger against the antenna, and disconnect when finger removed. 

The salutary tale is that the market is flooded with poor quality ESP32 clones. On close inspect, it could be distinguished with the Boot and Reset buttons not in the published position.
On rewiring with my new DevkitC-1 Esp32N32R8, the radio connects quickly with a decent bit rate.
In addition it is worth reminding people that a decent power supply helps with WiFi Connection. I swapped mine for a decent 3Amp supply.

All I need to do now is:-
a) Figure out which setting is appropriate for the TFT
"20:13:49 Touchpoint not valid x=12, y=10
20:13:51 Touchpoint not valid x=14, y=13
20:13:53 Touchpoint not valid x=46, y=55
20:13:53 Touchpoint not valid x=14, y=25"

b) Insert the BBC Radio stations in the list.

I'm more than grateful for not only your efforts but also your patience.

closed 2024-03-28 by RobinMosedale