schreibfaul1/ESP32-MiniWebRadio #375: Audio distortion with PCM5102A

FranZatGIT, 2024-03-22:
Dear Schreibfaul1, 
First of all - thank you very much for the great project. You've definitely invested a lot of time here. After a few teething problems (especially with a bad SD card), I got the web radio to work. I use the PCM5102A as a DA. What really bothers me is slight distortion - a kind of ringing. No matter whether radio, MP3 or FLAC. You can also hear it quite well in the example audio files. Is this due to the DA converter? Can YOU recommend another? Or is that jitter coming from the ESP? 

BTW. It looks too me that the AP is not working ( I cannot see any AP WIFI when in AP Mode)

many Thanks in advance

Franz

FranZatGIT, 2024-03-23:
To clarify my statement. The crackling noise is not a separate noise that is brought in by a digital bus. It is part of the music/language and modulates with it - only in the high tones. With a small speaker it's not noticeable, but with good headphones it is. It's not a big problem that you couldn't live with. Like a bad MP3, it just doesn't sound clear. Maybe my HIFI demands are just too high ;)

Arne65, 2024-03-23:
There shouldn't really be any background noise. Check your power supply. You shouldn't use USB power adapters for mobile devices or a step down converter because sometimes you don't really get a clean DC voltage. This can be easily checked with an oscil

FranZatGIT, 2024-03-29:
solved the issue - Audio has no distortion / Jitter when PCM is using 3.3V instead of 5V.

[image]

closed 2024-03-29 by FranZatGIT