schreibfaul1/ESP32-MiniWebRadio #376: No WIFI in AP Mode

FranZatGIT, 2024-03-22:
Dear Schreibfaul1,
It looks too me that the AP mode  is not working. I cannot find any AP WIFI when in AP Mode ( when display is showing the AP IP)
With wifi credentials in the common.h the radio works fine.

regards Franz

schreibfaul1, 2024-03-22:
So I can find the "MiniWebRadio" entry in the WIFI settings on my smartphone and connect to it. In a browser, the page is displayed under the IP 192.168.4.1, I can select the network and enter the password. [image]

The noise during audio playback is presumably transmitted from the digital to the analogue part via the operating voltage?

FranZatGIT, 2024-03-23:
Thank you for your quick reply! But well, it's a little trickier. When I tried to take a few screenshots yesterday it even worked. But I generally have a problem with the WIFI with the ESP32. Sometimes it works like a charme  for hours. Sometimes the WIFI doesn't want to connect or has connection problems and drops out.
It is exactly the problem that is described here: 

https://github.com/espressif/arduino-esp32/issues/2144

maybe it is somhow related to my Asus Router environement and DHCP. When I restart the router  mostly it works - I have the same issues in several ESP 8266 Projects. I did a workaround in setting a fixed IP

Here is the monitor output when a connection to hard coded WIFI (skynet) failed and AP is not opening a Hotspot: 
[image]

regarding audio destroiton I respond in the other thread

RobinMosedale, 2024-03-23:
You me too

RobinMosedale, 2024-03-24:
You're not alone
https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/379

FranZatGIT, 2024-03-29:
Bad Quality of esp32-s3-devkitc-1 caused the WIFI issues

closed 2024-03-29 by FranZatGIT