schreibfaul1/ESP32-MiniWebRadio #377: Display dunkel

ha0248, 2024-03-22:
Hallo Scgreibfaul1(Wolle),
ich versuche seit einer Woche verzeifelt das Mini Web Radio nach zu bauen, bin zwar in VScode und Platformio ein Anfänger, habe auch schon das ESP32-vs1053_ext mit Nextion Touch Senderverstellung gebaut, aber dies hier will einfach nicht laufen. Ich benutze ESP32S3 1N16MB/8Psram, TFT IL9488, microSD-Cardreader mit hama 16 MB SD und PCM 5102.
Ich kann uploaden, das läuft einwandfrei durch, das Display wir hell, zeigt aber nichts an.
Könnte es daran liegen, das ich das PSRAM nicht in "oktal" umgenannt habe? Das konnte ich nicht, weil der Run menuconfig in meiner Platformio nicht vorhanden ist. Wenn es draran liegen sollte, wie kriege ich denn diesen run menuconfig in Platformio rein? Ich konnte im Internet nichts darüber finden.
Gruß Dieter

schreibfaul1, 2024-03-22:
Hallo Dieter,
der Punkt menuconfig ist nicht aktiv. Es gibt noch zu viele Fehler weil einige Links zu notwendigen Bibliotheken fehlen. Daher benutzen wir zur Zeit die vorkompilierten Arduino Versionen. Das PSRAM wird in jedem Fall richtig erkannt.
Ich denke eher, das ist ein Standardfehler in Zusammenhang mit dem Display. Hier gibt es eine Erklärung dazu https://github.com/Bodmer/TFT_eSPI/discussions/898
Du kannst einfach MISO unbeschaltet lassen
[image]

Ich habe selber das Display im Einsatz und finde es ganz brauchbar.

ha0248, 2024-03-25:
Hallo Schreibfaul1(Wolle),
am NODEMCU Esp32s geht das TFT IL9488, ich hatte es auch schon so verdrahtet wie im Foto. Allerdings krieg ich auch mit dem NodeMCU das Radio nicht hin, es kommt immer der Hinweis: "SD CARD MOUNT FAILING", was wohl auch die Ursache mit dem TFT am ESP32S3 war.  Das Löten einer MMC gestaltet sich als schwierig. Gibt es denn kein SD Card Leser der am Mini Web Radio fuktioniert?
Gruß Dieter

HWH53, 2024-04-05:
Hallo ha248,
ich verwende das "Adafruit Micro SD SPI or SDIO Card Breakout Board - 3V ONLY!" von der Fa. Eckstein.
[https://eckstein-shop.de/AdafruitMicroSDSPIorSDIOCardBreakoutBoard-3VONLY](url)
Funktioniert einwandfrei!
Gruß
Helmut

ha0248, 2024-04-08:
Hallo Schreibfaul1(Wolle),
so, das Mini Web Radio läuft jetzt!!!!!! Prima!!!! Ich habe mir einen Espressif Esp32s1N8 gekauft, den SD Card-Leser habe ich fertig gelötet bei Ama..... gefunden. Die Wroom ESP32s3 sind China Clone, funktionieren nicht mit dem Mini Web Radio, muss man erstmal wissen. Nach drei Tagen Betrieb kann ich sagen, läuft toll, keine Wlan Aussetzer wie mit ESP32 Dev kit gebaute Radios. Ich habe noch eine Frage: wo finde ich den Programm-Teil des VU Meters? Ich möchte mir auch so was in andere Radios bauen, vielleicht finde ich dort ein paar Hinweise und Denkanstöße.
Vielen Dank im voraus und allerbeste Grüße
Dieter

schreibfaul1, 2024-04-10:
Super! das freut mich, manchmal ist schon die richtige HW entscheidend.
Die Werte für die Aussteuerungsanzeige kommen aus der audioI2S Bibliothek.
Dir Abfrage könnte so aussehen:
````c++
uint16_t vu = audio.getVUlevel();
uint8_t left = (vu & 0xF0) >> 8;
uint8_t right = vu & 0x0F;
````
Die Werte für einen Kanal liegen zwischen 0 ...127

vG
Wolle

schreibfaul1, 2024-04-25:
I don't know exactly what happens on the other side of the serial interface. But a loop with Serial.write() should take a lot of time. Wouldn't it be enough to just pass the values, `Serial2.printf("barR.val=%\nbarL.val=%i\n", left, right);` ?

closed 2024-06-09 by github-actions[bot]