schreibfaul1/ESP32-MiniWebRadio #378: Conflict with USB

RobinMosedale, 2024-03-23:
USB becomes unavailable:
 Executing task: C:\Users\robin\.platformio\penv\Scripts\platformio.exe device monitor --environment esp32s3 --port COM19 

UserSideException: could not open port 'COM19': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)

 *  The terminal process "C:\Users\robin\.platformio\penv\Scripts\platformio.exe 'device', 'monitor', '--environment', 'esp32s3', '--port', 'COM19'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it

ILI98341/PCM5102/esp32n16r8 standard configuration

USB becomes inoperable as soon as image is loaded.
Windows declares  an unknown device on the USB
No serial terminal output can be obtained.Probably a pin conflict with the standard Devkit esp32n16r8

common.h extract:
`// created: 10.Feb.2022
// updated: 08.Feb 2024

#pragma once
#pragma GCC optimize("Os") // optimize for code size

#define _SSID               "TP-LINK_EB850C"                        // Your WiFi credentials here
#define _PW                 "ju87bstuka"                // Or in textfile on SD-card
#define DECODER             1                               // (1)MAX98357A PCM5102A CS4344... (2)AC101, (3)ES8388, (4)WM8978
#define TFT_CONTROLLER      0                               // (0)ILI9341, (1)HX8347D, (2)ILI9486a, (3)ILI9486b, (4)ILI9488, (5)ST7796, (6)ST7796RPI
#define DISPLAY_INVERSION   0                               // (0) off (1) on
#define TFT_ROTATION        1                               // 1 or 3 (landscape)
#define TFT_FREQUENCY       40000000                        // 80000000, 40000000, 27000000, 20000000, 10000000
#define TP_VERSION          1                               // (0)ILI9341, (1)ILI9341RPI, (2)HX8347D, (3)ILI9486, (4)ILI9488, (5)ST7796, (3)ST7796RPI
#define TP_ROTATION         1                               // 1 or 3 (landscape)
#define TP_H_MIRROR         0                               // (0) default, (1) mirror up <-> down
#define TP_V_MIRROR         0                               // (0) default, (1) mittor left <-> right
#define AUDIOTASK_CORE      0                               // 0 or 1
#define AUDIOTASK_PRIO      2                               // 0 ... 24  Priority of the Task (0...configMAX_PRIORITIES -1)
#define I2S_COMM_FMT        0                               // (0) MAX98357A PCM5102A CS4344, (1) LSBJ (Least Significant Bit Justified format) PT8211
#define SDMMC_FREQUENCY     20000000                        // 80000000, 40000000, 27000000, 20000000, 10000000 not every SD Card will run at 80MHz
#define FTP_USERNAME        "esp32"                         // user and pw in FTP Client
#define FTP_PASSWORD        "esp32"
#define CONN_TIMEOUT        500                             // unencrypted connection timeout in ms (http://...)
#define CONN_TIMEOUT_SSL    2000                            // encrypted connection timeout in ms (https://...)
#define BT_SINK_NAME        "MiniWebRadio"                  // ESP32 only

/**********************************************************************************************************************/

#include <Arduino.h>
#include <ArduinoOTA.h>
#include <Preferences.h>
#include <Ticker.h>
#include <SPI.h>
#include <SD_MMC.h>
#include <FS.h>
#include <FFat.h>
#include <Wire.h>
#include <ESPmDNS.h>
#include <WiFiClient.h>
#include <WiFiMulti.h>
#include <vector>
#include "index.h"
#include "index.js.h"
#include "accesspoint.h"
#include "websrv.h"
#include "rtime.h"
#include "IR.h"
#include "tft.h"
#include "ESP32FtpServer.h"
#include "AC101.h"
#include "ES8388.h"
#include "WM8978.h"
#include "DLNAClient.h"
#include "KCX_BT_Emitter.h"

#ifdef CONFIG_IDF_TARGET_ESP32
    // Digital I/O used
        #define TFT_CS           22
        #define TFT_DC           21
        #define TFT_BL           32  // at -1 the brightness menu is not displayed
        #define TP_IRQ           39  // VN
        #define TP_CS             5
        #define SD_MMC_D0         2  // cannot be changed
        #define SD_MMC_CLK       14  // cannot be changed
        #define SD_MMC_CMD       15  // cannot be changed
        #define IR_PIN           35  // IR Receiver (if available)
        #define TFT_MOSI         23  // TFT and TP (VSPI)
        #define TFT_MISO         19  // TFT and TP (VSPI)
        #define TFT_SCK          18  // TFT and TP (VSPI)

        #define I2S_DOUT         25
        #define I2S_BCLK         27
        #define I2S_LRC          26
        #define I2S_MCLK          0  // mostly not used

        #define I2C_DATA         -1  // some DACs are controlled via I2C
        #define I2C_CLK          -1
        #define SD_DETECT        -1  // some pins on special boards: Lyra, Olimex, A1S ...
        #define HP_DETECT        -1
        #define AMP_ENABLED      -1

        #define BT_EMITTER_RX    33  // TX pin - KCX Bluetooth Transmitter (-1 if not available)
        #define BT_EMITTER_TX    36  // RX pin - KCX Bluetooth Transmitter (-1 if not available)
        #define BT_EMITTER_LINK  34  // high if connected                  (-1 if not available)
        #define BT_EMITTER_MODE  13  // high transmit - low receive        (-1 if not available)
#endif

#ifdef CONFIG_IDF_TARGET_ESP32S3
    // Digital I/O used
        #define TFT_CS            8
        #define TFT_DC           12
        #define TFT_BL           10 // at -1 the brightness menu is not displayed
        #define TP_IRQ           39
        #define TP_CS            15
        #define SD_MMC_D0        11
        #define SD_MMC_CLK       13
        #define SD_MMC_CMD       14
        #define IR_PIN            4  // IR Receiver (if available)
        #define TFT_MOSI         18  // TFT and TP (FSPI)
        #define TFT_MISO          2  // TFT and TP (FSPI)
        #define TFT_SCK          17  // TFT and TP (FSPI)

        #define I2S_DOUT          9
        #define I2S_BCLK          3
        #define I2S_LRC           1
        #define I2S_MCLK          0

        #define I2C_DATA         -1  // some DACs are controlled via I2C
        #define I2C_CLK          -1
        #define SD_DETECT        -1  // some pins on special boards: Lyra, Olimex, A1S ...
        #define HP_DETECT        -1
        #define AMP_ENABLED      -1

        #define BT_EMITTER_RX    45  // TX pin - KCX Bluetooth Transmitter (-1 if not available)
        #define BT_EMITTER_TX    38  // RX pin - KCX Bluetooth Transmitter (-1 if not available)
        #define BT_EMITTER_LINK  19  // high if connected                  (-1 if not available)
        #define BT_EMITTER_MODE  20  // high transmit - low receive        (-1 if not available)

#endif
`

JmanNZ, 2024-03-27:
I had a similar issue the other usb port worked for me

RobinMosedale, 2024-03-28:
Thank you. That worked. Something must be clashing with the USB ports

closed 2024-03-29 by RobinMosedale