schreibfaul1/ESP32-MiniWebRadio #379: Access Point not initiating on failure to connect

RobinMosedale, 2024-03-23:
The Webradio is consistently failing to connect, see elsewhere.

However, if that occurs it is supposed to initiate an access point to enable editing of the configuration.

It fails to raise that access point.
It is  inaccessible via 192.168.4.1

[image]
[image]
[image]

Arne65, 2024-03-23:
If you have problems with the WiFi, I would recommend that you load a simple WiFi program onto the esp. If you have problems with that too, get a new ESP.
I hope you have also connected to the EsP's WiFi via your PC. If not, you will never get a connection from your PC to 192.168.4.1

RobinMosedale, 2024-03-23:
Thank you Arne65,

However, the failure to connect has already been raised. This is report confines itself to failing to raise an Access point on failure.

The Esp32 has been checked with a simple wifi connect programme. It connect straight away. However when running WifiMulti as in main.cpp it fails repeated with Auth expire. Tests run in conjunction with schreibfaul1.

See
https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/374
Page down to the bottom
That is of little consequence here. This is reporting a further error.
This is a completely different problem.

FranZatGIT, 2024-03-24:
Hello Schreibfaul, I was opening another thread before but I will continue to respond in this one. I spent quite a lot of time to find the issue but was not abel to get that working. I have general issue with the WIFI of the Webradio like i mentioned in #376  and the same as RobinMosedale. 
some different flavors:
1. The board connects fine and everything works as it should for long time ( but that happens just occasional)
2. The board connects but get slow stream notifications and finally stops
3. The board is not connecting at all with this "Reason: 2 - AUTH_Expire"

The Webradio Radio AP starts according Monitor output but the WIFI somtimes could not be found or very late and the List of available Networks takes also a while to be shown.

I tried: 

- 2 different esp32-s3-devkitc-1 N18R8 boards (but bought at the same time) 
- 3 different SD cards
- 2 different TFT
- 4 different WIFI networks

1. Network - my Asus WIFI with Mesh of 3 devices => here somtimes it works as mentioned above
2. Huawei 4G Hotspot => Never connect
3. Samsung 20FE Hotspot => Never connects
4. Samsung S5 (very old) = >never connects

Below you find the 4 screenshots of the monitor output  with errors for the 4 different networks:

[image]

[image]

[image]

[image]

FranZatGIT, 2024-03-24:
Just want to add - 4 Networks were just  20cm to 3m far from the Webradio. Just one network was online at a time. With the uncountable attemts I found no pattern when it works and when not. Just the chance that it works was a bit higher right after rooter reboot...

Thanks so much for this project - I really want to use that Radio but in this status this is not possible. I belive the nature of the issue is not related to your code itself. It looks like a general WIFI / library bug

regards Franz

kerat71, 2024-03-25:
I have exactly the same problems.
I have been following the project for a long time, by the way MiniWebRadio Version 24 , Apr.10 2019
still working at this very moment :-)
Thank you, Schreibfaul1!

RobinMosedale, 2024-03-25:
Indeed my original fault report conducted some simple tests on the chip to show that it connected as you say.

Unfortunately I deleted my git clone and now the build references an external board git repository which has replaced the ES32Devkit n8r16 with another, and I've no idea how to correct the reference. See

https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/381

schreibfaul1, 2024-03-26:
Now I need to find out what the cause is. I have not changed the routines for WiFiMulti, but the Arduino version is now V3.0.0-alpha3 and was previously V2.0.14. I use a Fritzbox as an internet router and do not have the problems.   The encryption is set to `WPA2(CCMP)`, `WPA2 + WPA3` does not work at all. Going back to the old Arduino V2 is time-consuming.

RobinMosedale, 2024-03-26:
We're all more than grateful, Wolle.
It's a shame that you're good work is being hit by external changes.
I'm embarrassed that I seem to be encountering and raising them

RobinMosedale, 2024-03-28:
Replaced the ESP32 with a genuine Espressif Devkit. The issue is due to poor clone with poor antenna
See 
https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/373

closed 2024-03-28 by RobinMosedale

FranZatGIT, 2024-03-28:
Dear all, 

i can confirm - I ordered another ESP32 s3  and it works fine!