schreibfaul1/ESP32-MiniWebRadio #380: Version 2 does not connect to wifi.

hevet, 2024-03-24:
I want to make version 2 with the VS1053 module. I uploaded this version to a set that works and connects to Wi-Fi in version 3, but the same set does not connect to Wi-Fi in version 2. The error in the terminal is:

[  3394][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 1C:BF:CE:66:A9:A0 SSID: wifi-A9A0 Channel: 1 (-33)
[  6566][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[  8438][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (6).
00:00:07 WiFI_info:   WiFi credentials are not correct
[  8500][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 8 - ASSOC_LEAVE
00:00:07 Accesspoint: IP: 192.168.4.1
00:00:14 setup: ....  6 WiFi networks found
00:00:14 setup: ....  wifi-A9A0 (-34)
00:00:14 setup: ....  T-Mobile (-48)
00:00:14 setup: ....  Orange_Swiatlowod_9D10 (-82)
00:00:14 setup: ....  ZTE_T3000_0C4ED9 (-84)
00:00:14 setup: ....  Igula_Jedrula (-92)
00:00:14 setup: ....  PLAY_Swiatlowodowy_4394 (-94)

Arne65, 2024-03-24:
Maybe it will help to delete the flash and then reload the program

hevet, 2024-03-24:
Unfortunately this doesn't help.

closed 2024-05-09 by github-actions[bot]