schreibfaul1/ESP32-MiniWebRadio #381: esp32s3 (board: ESP32-S3-DevKitC-1-N16R8: Could not find one of 'package.json' manifest files in the package

RobinMosedale, 2024-03-25:
This is a clean load today:

Executing task: C:\Users\robin\.platformio\penv\Scripts\platformio.exe run --environment esp32s3 

Processing esp32s3 (board: ESP32-S3-DevKitC-1-N16R8; platform: https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package

 *  The terminal process "C:\Users\robin\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'esp32s3'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

RobinMosedale, 2024-03-25:
The referenced board repository was changed 2 days ago, and no longer references ESP32-S3-DevKitC-1-N16R8 but replaced with 
[4d_systems_esp32s3_gen4_r8n16.json] in the referenced board git file.
(https://github.com/Jason2866/platform-espressif32/blob/Arduino/IDF5/boards/4d_systems_esp32s3_gen4_r8n16.json)
Perhaps freeze certain dependencies otherwise build are at the mercy of changes outside the author's control

schreibfaul1, 2024-03-26:
I have pinned it, possible changes now have no effect. Unfortunately the repository https://github.com/platformio/platform-espressif32 does not work with IDF 5.1 or higher.

RobinMosedale, 2024-03-26:
Fabulous. Very many thanks

schreibfaul1, 2024-03-27:
What is missing is the file **..../.../platformio/packages/framework-arduinoespressif32/package.json**. Strange. Try deleting the .platformio/packages folder. The folder is automatically recreated. Is the json file then available?

closed 2024-03-27 by RobinMosedale

dave6024t, 2025-02-18:
I am using PlatformIO under VSC and the Arduino framework for ESP32-S3 development.

I have adapted [esp32-s3-devkitc-1-n16r8v.json](https://github.com/handledexception/platform-espressif32/blob/esp32-s3-devkitc-1-n16r8v/boards/esp32-s3-devkitc-1-n16r8v.json) for my genuine Espressif ESP32-S3-DEVKITC-1-N32R8V dev kit (which I understand can currently use only 16MB of its 32MB of flash).  After building my test app, I constantly got boot messages caused by the Watchdog timer RTC resetting every 10 seconds.  This was resolved after correcting the specification in the JSON file of the flash interface from Quad SPI to Octal SPI, as listed in [this Espressif DevKit User Guide](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitc-1/user_guide_v1.0.html), changing "memory_type" from "qio_opi" to "opi_opi".  I also changed  "flash_mode": "qio" to "flash_mode": "opi".  I left "partitions": "default_16MB.csv" unchanged.  According to the user guide, most of this series of DevKits DO use Quad SPI for the Flash, but those with larger 16MB or 32MB flash use Octal SPI.

I also found that Serial.print messages where not seen on the terminal, until I changed "-DARDUINO_USB_CDC_ON_BOOT=1" to "-DARDUINO_USB_CDC_ON_BOOT=0" in the the extra_flags section.