schreibfaul1/ESP32-MiniWebRadio #382: Cover beim Start nur auf halben Bildschirm

acer1111, 2024-03-25:
Hallo Wolle
Ich habe das Projekt erfolgreich nachgebaut 
Alles funktioniert soweit gut .
Auch das Problem mit der geänderten Uhrzeit
Hat sich wie von selbst gelöst.
Was mich im Moment noch ein wenig stört
ist , beim Start wird das Display-Cover nur halb auf dem Bildschirm angezeigt ,also nur die obere Hälfte ist ausgefüllt,
Ich weis leider nicht wie ich es ändern kann.
Vielleicht hast du ja einen Tip für mich .
Vielen  Dank 
[image]

RobinMosedale, 2024-03-25:
see https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/374

It's an fault in the image binary. It's been fixed, but be aware there's now another fault in the build referencing an board configuration is an external git repository which has been changed and the board reference is incorrect so won't build
See 
https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/381

schreibfaul1, 2024-03-26:
Ja, das ist wie Robin schreibt, eine defekte Datei die den Fehler in jpeg Dekoder erzeugt.
Bitte entpacke die Datei "content_on_SD_card.zip" erneut und ersetze die "MiniWebRadioV3.jpg" auf der SD Karte im Ordner /common/s

acer1111, 2024-03-26:
Prima das war es
Ich hab die Datei ersetzt und es funktioniert bestens !
Vielen Dank

Solar320, 2024-03-29:
...............................................
> Hallo Wolle Ich habe das Projekt erfolgreich nachgebaut Alles funktioniert soweit gut . Auch das Problem mit der geänderten Uhrzeit Hat sich wie von selbst gelöst. Was mich im Moment noch ein wenig [...]

Gleich vorweg, das mit dem ersetzen aller Dateien hat bei mir null effekt.

Das mit dem V3 Bild habe ich auch gehabt und auch immer noch. Ich habe das v2 Bild eingestellt und das funktioniert. 
ca. Zeile 1880  in der Main.cpp  drawImage("/common/MiniWebRadioV2.jpg", 0, 0); // Welcomescreen

Zusätzlich habe ich ein eigenes Bild rein gemacht, das funktioniert auch.

[image]

Dann habe ich das original MiniRadioV3 Bild genommen habs kopiert und in eine neue Datei 320x240 300dpi hineinkopiert gespeichert und?!?! es geht auch nicht.

Exakt das gleiche habe ich auch mit dem eigens angepassten Bild gemacht. Es muss irgendwas mit den Bilddaten selbst zu tun haben das es mit der V3 nicht funktioniert.

steve6375, 2024-03-29:
Use Microsoft Paint to save the jpg
Some other programs save jpg as progressive jpg for websites and that is not compatible

Solar320, 2024-03-29:
As I said, my own picture converted to 320x240 300dpi and saved with Paint.net is working. I can try it again, using the normal Micr. Paint.

Solar320, 2024-03-29:
[image]

I used paint3d because I have no normal Paint on windows 11. I opened it and save it as jpg renamed it to vtest. changed the line
drawImage("/common/MiniWebRadioVtest.jpg", 0, 0); // Welcomescreen
and now you see the result :-)

[image]

acer1111, 2024-03-30:
Hallo
Ich habe mir die Datei : Content_on_sd-card.zip erneut von GitHub runtergeladen ,
Dann habe ich sie entpackt und die entsprechende Datei für das Cover ersetzt ,
Danach war alles ok und das Bild wurde richtig angezeigt

Solar320, 2024-03-30:
Das habe ich auch mehrfach versucht, ohne Erfolg. Allerdings könnte es sein das die Datei zwischenzeitlich aktualisiert wurde und jetzt funktioniert. Wie gesagt, der Tip von Steve hat geholfen, einfach das Originalbild im common/s Ordner mit Paint oder 3D Paint öffnen und "speichern unter" an einem anderen Ort und danach die Datei auf der SD Karte damit überschreiben. 

Derzeit kann ich auch nicht so ohne weiteres Updates mit Github Desktop holen, weil ich Änderungen im Projekt gemacht habe und der will mir immer alles überschreiben. Mein ESP hat ein Neopixel was unter umständen grell aufleuchtet, das habe ich mit der neopixel bibliothek auf eine angenehmere Helligkeit eingestellt.  Ich hab das noch nicht gecheckt wie oder ob es geht meine Änderungen beizubehalten. aber das ist OT.

closed 2024-05-14 by github-actions[bot]