schreibfaul1/ESP32-MiniWebRadio #385: ILI9341 Touch coordinates

RobinMosedale, 2024-03-29:
Wolle,

My apologies for raising this topic again.
I've searched through all who have had issues with the Touch panel on these TFT systems.

I'm getting random coordinates.
the tft log enabled.
The extract below is from touching the same point on the TFT screen: The lower left signal strength symbol centre.
`[162498][I][tft.cpp:5292] loop(): tp_pressed x=98, y=68
[163380][I][tft.cpp:5292] loop(): tp_pressed x=22, y=15
[165177][I][tft.cpp:5292] loop(): tp_pressed x=68, y=48
20:46:00 Touchpoint not valid x=68, y=48
[165951][I][tft.cpp:5292] loop(): tp_pressed x=45, y=21
20:46:01 Touchpoint not valid x=45, y=21
[166695][I][tft.cpp:5292] loop(): tp_pressed x=47, y=43
20:46:02 Touchpoint not valid x=47, y=43
[169085][I][tft.cpp:5292] loop(): tp_pressed x=14, y=4
[169885][I][tft.cpp:5292] loop(): tp_pressed x=26, y=17
[170585][I][tft.cpp:5292] loop(): tp_pressed x=55, y=35
20:46:06 Touchpoint not valid x=55, y=35
[172568][I][tft.cpp:5292] loop(): tp_pressed x=25, y=15
20:46:08 Touchpoint not valid x=25, y=15
[173426][I][tft.cpp:5292] loop(): tp_pressed x=55, y=31
20:46:08 Touchpoint not valid x=55, y=31
[174179][I][tft.cpp:5292] loop(): tp_pressed x=40, y=24
20:46:09 Touchpoint not valid x=40, y=24
[174902][I][tft.cpp:5292] loop(): tp_pressed x=108, y=76
[175725][I][tft.cpp:5292] loop(): tp_pressed x=31, y=23
[176918][I][tft.cpp:5292] loop(): tp_pressed x=102, y=65
20:46:12 Touchpoint not valid x=102, y=65`

Tapping across a row horizontally left to right, or vertically top to bottom along a reasonable straight path, generates incomprehensible coordinates.

I've two ILI9341. Both exhibit.

I've tried: 
(0)ILI9341, (1)ILI9341RPI, (2)HX8347D, (3)ILI9486, (4)ILI9488, (5)ST7796, builds
I've tried various speeds
80000000, 40000000, 20000000

exactly similar results.
It was also experienced on another less reliable processor, this is an authentic Espressif N32r8 configured N16r8

I haven't tried mirroring, but the coordinates aren't consistent enough

The TFT/TP has been powered directly (common earth) 5V, and via 3.3v from the ESP32

Have you seen this before, and any suggestions please?
Robin

schreibfaul1, 2024-03-29:
The touch controller only returns voltage values. The position is calculated from the voltage values x/y. I can imagine that MISO is probably not connected correctly. The display itself does not need MISO.

RobinMosedale, 2024-04-01:
Thank you, Wolle
So connections were:-
TFT MISO not Connected
TP MISO ->2
TFT and TP CLK -> 17
TP MOSI (DIN)  and TFT MOSI -> 18
as well as TP IRQ ->39
and TP CS ->15
TP and TFT Orientation landscape 3
in addition both TP orientations mirrored
#define TP_H_MIRROR         1                               // (0) default, (1) mirror up <-> down
#define TP_V_MIRROR         1                               // (0) default, (1) mittor left <-> right

and we have not only the TP functioning, but in the correct locations

Very many thanks

closed 2024-04-01 by RobinMosedale