schreibfaul1/ESP32-MiniWebRadio #386: Wifi connection

Jefke1, 2024-03-29:
Hello Wolle,

My own WiFi signal is hard to find, when it does connect I get the error connection to NTP failed, trying again and then keeps rebooting..... V2 plays here in the same place and has no connection problems.
ESP32 Chip: ESP32-S3
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.00o Mar 02/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 40 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Software reset via esp_restart

00:01:29
00:01:29        ***************************    
00:01:29        *     MiniWebRadio V3     *    
00:01:29        ***************************    
00:01:29
00:01:29 setup: ....  Arduino is pinned to core 1
00:01:29 tft_info: .  init ILI9341
00:01:29 setup: ....  Init SD card
00:01:29 setup: ....  SD card found, 14811.3 MB by 14832.0 MB free
[   787][I][esp32-hal-ledc.c:131] ledcAttachChannel(): LEDC attached to pin 10 (channel 0, resolution 8)
00:01:29 setup: ....  seek for stations.csv
00:01:29 setup: ....  stations.csv found
00:01:29 setup: ....  seek for WiFi networks
[  1173][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: Orange-486AE
[  6856][I][WiFiMulti.cpp:134] run(): [WIFI] scan done
[  6862][I][WiFiMulti.cpp:139] run(): [WIFI] 12 networks found
[  6869][I][WiFiMulti.cpp:263] run(): [WIFI] Connecting BSSID: 54:67:51:A4:86:B3 SSID: Orange-486AE Channel: 6 (-82)
[  6880][W][STA.cpp:544] disconnect(): STA already disconnected.
[  6999][I][WiFiMulti.cpp:283] run(): [WIFI] Connecting done.
00:01:35 WiFI_info:   Connecting WiFi...
00:01:35 WiFI_info:   mDNS name: MiniWebRadio
00:01:35 setup: ....  connected to , IP address is 192.168.0.131
00:01:35 ftpServer:   Buffers allocated: 4622 bytes
00:01:35 connection to NTP failed, trying again
[  7053][W][STA.cpp:157] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037b7de
=> 0x4037b7de: esp_cpu_wait_for_intr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/cpu.c:121
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
=> 0x403c9700: ?? ??:0
load:0x403c9704,len:0xad0
=> 0x403c9704: ?? ??:0
load:0x403cc700,len:0x29d8
=> 0x403cc700: ?? ??:0
entry 0x403c9880
=> 0x403c9880: ?? ??:0
[   419][I][esp32-hal-psram.c:92] psramInit(): PSRAM enabled

schreibfaul1, 2024-03-30:
-82dB is far too weak, good values are higher than -50dB
[image]

Jefke1, 2024-03-30:
Okay, I understand that.
Now I have upload it to an esp32 wrover ie with an external antenna.... the result is very strange.

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1448
load:0x40078000,len:14820
=> 0x40078000: ?? ??:0
load:0x40080400,len:4
=> 0x40080400: _init at ??:?
load:0x40080404,len:3356
=> 0x40080404: ?? ??:0
entry 0x4008059c
=> 0x4008059c: ?? ??:0
[   454][I][esp32-hal-psram.c:92] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-D0WD
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.00o Mar 02/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 40 MHz
CPU speed 240 MHz
SDMMC speed 40 MHz
TFT speed 40 MHz
PSRAM total size: 4194304 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************
00:00:00        *     MiniWebRadio V3     *
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 14811.3 MB by 14832.0 MB free
[   874][I][esp32-hal-ledc.c:131] ledcAttachChannel(): LEDC attached to pin 32 (channel 0, resolution 8)
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  1291][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: Orange-486AE
[  8241][I][WiFiMulti.cpp:134] run(): [WIFI] scan done
[  8247][E][WiFiMulti.cpp:137] run(): [WIFI] no networks found
[  8253][E][WiFiMulti.cpp:317] run(): [WIFI] no matching wifi found!
[ 13270][E][main.cpp:1608] connectToWiFi(): WiFi timeout
[ 16286][E][main.cpp:1614] connectToWiFi(): WiFi timeout
00:00:16 WiFI_info:   WiFi credentials are not correct
[ 16345][W][STA.cpp:544] disconnect(): STA already disconnected.
00:00:16 Accesspoint: IP: 192.168.4.1
00:00:22 setup: ....  no WiFi networks found

schreibfaul1, 2024-03-30:
It looks as if no signal is being received. ESP32s with the option of connecting an external antenna usually have a solder bridge. Perhaps this is missing.
[image]

Jefke1, 2024-04-01:
Hello Wolle,
I tested several boards, Esp32 Devkit, Esp32_IE, ESP32_S3
I used the simple code "Wifi" from Espressif. All boards connected immediately, I have a different location and the RSSID is better but still no connection, I adjusted the time(Wifi) in the code but unfortunately it didn't work. 
Is there a way to disable WifiMulti??
I have had the opportunity to test Bluetooth, it works well, but the volume is a bit too low for me. The Bluetooth button does not appear on the display even though it is on the SD card.

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1448
load:0x40078000,len:14820
=> 0x40078000: ?? ??:0
load:0x40080400,len:4
=> 0x40080400: _init at ??:?
load:0x40080404,len:3356
=> 0x40080404: ?? ??:0
entry 0x4008059c
=> 0x4008059c: ?? ??:0
[   454][I][esp32-hal-psram.c:92] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-D0WD
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.00o Mar 02/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 40 MHz
CPU speed 240 MHz
SDMMC speed 40 MHz
TFT speed 40 MHz
PSRAM total size: 4194304 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 14811.3 MB by 14832.0 MB free
[   873][I][esp32-hal-ledc.c:131] ledcAttachChannel(): LEDC attached to pin 32 (channel 0, resolution 8)
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  1271][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: Orange-486AE
[  1290][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: Orange-486AE
[  8821][I][WiFiMulti.cpp:134] run(): [WIFI] scan done
[  8827][I][WiFiMulti.cpp:139] run(): [WIFI] 3 networks found
[  8834][E][WiFiMulti.cpp:317] run(): [WIFI] no matching wifi found!
[ 13851][E][main.cpp:1608] connectToWiFi(): WiFi timeout
[ 16867][E][main.cpp:1614] connectToWiFi(): WiFi timeout
00:00:16 WiFI_info:   WiFi credentials are not correct
[ 16926][W][STA.cpp:544] disconnect(): STA already disconnected.
00:00:16 Accesspoint: IP: 192.168.4.1
00:00:23 setup: ....  1 WiFi networks found
00:00:23 setup: ....  Orange-486AE (-44)
00:00:23 ftpServer:   Ftp server waiting for connection on port 21

schreibfaul1, 2024-04-02:
Sure you can replace "WifFimulti" with WiFi.begin(SSID, PW). What surprises me is that Orange-486AE is not connected. Are there special characters in the password? Maybe there are also data leftovers in the nvs, try "Erase Flash and Upload" instead of upload

Jefke1, 2024-04-03:
I tried everything, nothing helped, No special characters  password, erase flash and upload, too bad.
I also tested V2, no connection problems on the same esp.
It seems that the signal drops out when connecting to WiFi. weird.
ESP32 Chip: ESP32-D0WD
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.00p Mar 31/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 40 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 4194304 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 14811.3 MB by 14832.0 MB free
[   873][I][esp32-hal-ledc.c:131] ledcAttachChannel(): LEDC attached to pin 32 (channel 0, resolution 8)
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  1270][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: Orange-486AE
[  9404][I][WiFiMulti.cpp:134] run(): [WIFI] scan done
[  9410][E][WiFiMulti.cpp:137] run(): [WIFI] no networks found
[  9417][E][WiFiMulti.cpp:317] run(): [WIFI] no matching wifi found!
[ 14434][E][main.cpp:1608] connectToWiFi(): WiFi timeout
[ 17450][E][main.cpp:1614] connectToWiFi(): WiFi timeout
00:00:17 WiFI_info:   WiFi credentials are not correct
[ 17509][W][STA.cpp:544] disconnect(): STA already disconnected.
00:00:17 Accesspoint: IP: 192.168.4.1
00:00:24 setup: ....  16 WiFi networks found
00:00:24 setup: ....  Orange-486AE (-44)
00:00:24 setup: ....  Orange-1d6e6 (-56)
00:00:24 setup: ....  Orange-5cd0a (-66)
00:00:24 setup: ....  Guest-Orange-5cd0a (-66)
00:00:24 setup: ....  Proximus-Home-212702 (-68)
00:00:24 setup: ....  Proximus-Home-212702 (-73)
00:00:24 setup: ....  Guest-Orange-7a6c7 (-74)
00:00:24 setup: ....  Orange-7a6c7 (-75)
00:00:24 setup: ....  Proximus-Home-212702 (-79)
00:00:24 setup: ....  telenet-05670 (-80)
00:00:24 setup: ....  WiFi-2.4-BB4B (-86)
00:00:24 setup: ....  telenet-4E281C7 (-88)
00:00:24 setup: ....  WiFi-2.4-4F80 (-91)
00:00:24 setup: ....  Proximus Public Wi-Fi (-91)
00:00:24 setup: ....  Proximus Public Wi-Fi (-92)
00:00:24 setup: ....  Proximus-Home-6B61 (-93)
00:00:24 ftpServer:   Ftp server waiting for connection on port 21
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1448
load:0x40078000,len:14820
=> 0x40078000: ?? ??:0
load:0x40080400,len:4
=> 0x40080400: _init at ??:?
load:0x40080404,len:3356
=> 0x40080404: ?? ??:0
entry 0x4008059c
=> 0x4008059c: ?? ??:0
[   454][I][esp32-hal-psram.c:92] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-D0WD
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.00p Mar 31/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 40 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 4194304 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 14811.3 MB by 14832.0 MB free
[   873][I][esp32-hal-ledc.c:131] ledcAttachChannel(): LEDC attached to pin 32 (channel 0, resolution 8)
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  1270][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: Orange-486AE
[  8204][I][WiFiMulti.cpp:134] run(): [WIFI] scan done
[  8210][E][WiFiMulti.cpp:137] run(): [WIFI] no networks found
[  8217][E][WiFiMulti.cpp:317] run(): [WIFI] no matching wifi found!
[ 13234][E][main.cpp:1608] connectToWiFi(): WiFi timeout
[ 16250][E][main.cpp:1614] connectToWiFi(): WiFi timeout
00:00:16 WiFI_info:   WiFi credentials are not correct
[ 16309][W][STA.cpp:544] disconnect(): STA already disconnected.
00:00:16 Accesspoint: IP: 192.168.4.1
00:00:22 setup: ....  17 WiFi networks found
00:00:22 setup: ....  Orange-486AE (-45)
00:00:22 setup: ....  Orange-1d6e6 (-52)
00:00:22 setup: ....  Guest-Orange-5cd0a (-66)
00:00:22 setup: ....  Proximus-Home-212702 (-69)
00:00:22 setup: ....  Orange-5cd0a (-74)
00:00:22 setup: ....  Orange-7a6c7 (-75)
00:00:22 setup: ....  Guest-Orange-7a6c7 (-75)
00:00:23 setup: ....  Proximus-Home-212702 (-75)
00:00:23 setup: ....  Proximus-Home-212702 (-78)
00:00:23 setup: ....  telenet-05670 (-79)
00:00:23 setup: ....  WiFi-2.4-BB4B (-85)
00:00:23 setup: ....  Proximus-Home-6B61 (-87)
00:00:23 setup: ....  telenet-4E281C7 (-90)
00:00:23 setup: ....  Proximus Public Wi-Fi (-92)
00:00:23 setup: ....  WiFi-2.4-4F80 (-93)
00:00:23 setup: ....  Proximus Public Wi-Fi (-94)
00:00:23 setup: ....  telenet-8493442 (-95)
00:00:23 ftpServer:   Ftp server waiting for connection on port 21

schreibfaul1, 2024-04-03:
That is strange. There are no changes in the WiFi code between V2 and V3. The Orange-486AE network is perfectly fine with -45dB. However, the connection problem could be due to the fact that the new IDF V5.1 is now being used. I think your router is rejecting the connection. Can you change the settings in the router (WPA2/WPS3. CCMP...)?

Jefke1, 2024-04-03:
I was thinking in the same direction, I think the modem I have is an outdated type. time for replacement. 
I can adjust some settings if necessary.

Jefke1, 2024-04-10:
Hi Wolle, 
Installed a new modem today, everything works perfectly. 
Sorry for the inconvenience.

closed 2024-04-10 by Jefke1