schreibfaul1/ESP32-MiniWebRadio #387: Inserting BBC M3u

RobinMosedale, 2024-04-01:
Wolle,

I'm delight that I've finally managed to build your sophisticated radio successfully.

The object was to access BBC streaming radio stations

I've verified that this link works:-
https://garfnet.org.uk/download/radio/20231029-bbc-radio-norewind.m3u
It is a playlist, and I've edited the stations.csv successfully:

Hide | Cy | StationName | StreamURL
-- | -- | -- | --
  | D | 0N 70s | 0n-80s.radionetz.de:8000/0n-70s.mp3
* | D | 0N 80s | 0n-80s.radionetz.de:8000/0n-80s.mp3
* | D | 0N 90s | 0n-90s.radionetz.de:8000/0n-90s.mp3
* | D | 0N Charts | 0n-charts.radionetz.de:8000/0n-charts.mp3
  | D | 0N Dance | 0n-dance.radionetz.de:8000/0n-dance.mp3
  | D | 0N Disco | 0n-disco.radionetz.de:8000/0n-disco.mp3
* | D | 1000 Oldies | c3.auracast.net:8010/stream
* | D | Eurodance | www.laut.fm/eurodance
  | D | extra-radio 88.0 | https://www.extra-radio.de/stream/listen.m3u
  | UK | BBC Radio 4 | https://garfnet.org.uk/download/radio/20231029-bbc-radio-norewind.m3u
  | D | MacSlon's Irish Pub Radio | macslons-irish-pub-radio.stream.laut.fm/macslons-irish-pub-radio
  | RU | Europa Plus Top 40 | https://hls-01-europaplus-top40.emgsound.ru/21/playlist.m3u8
  | GR | Άνοιξη 100.7 | http://solid1.streamupsolutions.com:55023/stream
  | RU | НАШЕ Радио | nashe1.hostingradio.ru/nashe-128.mp3
  | RU | Радио Русские Песни | listen.rusongs.ru/ru-mp3-128
  | BG | Свежа България | http://31.13.223.148:8000/fresh.mp3
  | CH | SWISS POP | https://stream.srg-ssr.ch/rsp/aacp_48.asx
  | BG | BGRADIO | http://play.global.audio/bgradio_low.ogg
  | D | knixx.fm | http://s1.knixx.fm:5347/dein_webradio_vbr.opus
  | UK | Select a station here | https://a.files.bbci.co.uk/ms6/live/3441A116-B12E-4D2F-ACA8-C1984642FA4B/audio/simulcast/dash/nonuk/pc_hd_abr_v2/ak/bbc_radio_fourfm.mpd
  | UK | BBC | https://garfnet.org.uk/download/radio/20231029-bbc-radio-norewind.m3u
`
Of course, this is a playlist, and when I select this in the play window, it plays successfully from the first entry, which is the World Service.
However, I'd like to select the next station, which isn't possible on the radio play list.
What is the most effective way of loading this m3u, on a playlist on the SD, or break out the .m3u into the individual stations?

schreibfaul1, 2024-04-02:
Hi Robin, 
you don't need to edit the .cvs. You can do this via the website
 [image] 
1) Add: name and URL of the station 
2) Save: the new table in the NVS 
3) Load: the updated table into the MWR (or reboot)

RobinMosedale, 2024-04-02:
Fabulous.

I did tyy that but it seemed to have trouble with one BBC4 m3ug.
Perhaps didn't complete the actions correctly.

Just done BBC Radio 4, BBC World, and BBC Radio 4 Long wave and all working.

[image]
(That's a brand new ILI9341, I haven't yet taken the protective screen off for fear of damage, BUT IT'S FINE

I'm absolutely delighted.

This was the objective to receive BBC streaming and hook up to either internal amp or external amp.

3xESP32's, 1xPCM5102, 3xILI9341 , 2xPCM5102, since January, but we're there.

Absolutely delighted.

Very, very many thanks, Wolle

closed 2024-04-02 by RobinMosedale