schreibfaul1/ESP32-MiniWebRadio #388: Hilfe zu "OTA" (Host MiniWebRadio.local Not Found) bzw. "connection to NTP" failed

TubaFonse, 2024-04-01:
Hallo Wolle, 
ich wollte heute (ich weiß, es ist der 1. April) meinem MiniWebRadio die aktuelle Firmware verpassen, 
aber jetzt klappt leider gar nichts mehr: 

1.) die aktuelle Version vom 31.03. lässt sich nicht flashen (OTA-Version, aber die bringe ich nicht zum laufen): 

18:42:52 [DEBUG]: Options: Namespace(esp_ip='MiniWebRadio.local', host_ip='0.0.0.0', esp_port=3232, host_port=20273, auth='', image='.pio\\build\\esp32s3\\firmware.bin', spiffs=False, debug=True, progress=True, timeout=10)
18:42:52 [INFO]: Starting on 0.0.0.0:20273
18:42:52 [INFO]: Upload size: 2304208
Sending invitation to MiniWebRadio.local failed
18:42:55 [ERROR]: Host MiniWebRadio.local Not Found
*** [upload] Error 1

Hier bitte ich um "Nachhilfeunterricht" - was muss ich einrichten, damit der Upload-Port funktioniert? 

2.) die Version vom 26.03. hat bisher bei mir funktioniert env:esp32s3 (jedoch nur die Option "ohne OTA"). 
Diese habe ich heute versuchsweise nochmal compiliert und geflasht - leider mit einer Bootloop bei "connecting to ntp": 
00:00:10 WiFI_info:   Connecting WiFi...
00:00:10 WiFI_info:   mDNS name: MiniWebRadio
00:00:10 setup: ....  connected to FRITZ!Box7490LF, IP address is 192.168.178.31
00:00:10 ftpServer:   Buffers allocated: 4622 bytes
00:00:10 connection to NTP failed, trying again
[ 10673][D][WiFiGeneric.cpp:1055] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 10690][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 8 - ASSOC_LEAVE

Details zu 1.)  
Processing esp32s3 (board: ESP32-S3-DevKitC-1-N16R8; platform: https://github.com/Jason2866/platform-espressif32.git#348a508; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ESP32-S3-DevKitC-1-N16R8.html
PLATFORM: Espressif 32 (2024.3.10+sha.348a508) > Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.0+sha.4123e20
 - framework-arduinoespressif32-libs @ 5.1.3+sha.e7771c75bd.sha.270b5dc
 - tool-esptoolpy @ 4.7.2
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 3.2.0
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-riscv32-esp @ 12.2.0+20230208
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 47 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32-audioI2S @ 2.0.0+sha.e9a294e
|-- ESP32-KCX-BT-EMITTER @ 0.0.0+20240128094011.sha.338bc78
|-- ESP32-DLNA-Client @ 0.0.0+20231205185914.sha.cdde817
|-- ESP32-IR-Remote-Control @ 1.0.10+sha.6cb54a9
|-- FFat @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- ac101
|-- es8388
|-- ftpsrv
|-- rtime
|-- tftLib
|-- websrv
|-- wm8978
|-- ArduinoOTA @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- FS @ 2.0.0
|-- Preferences @ 2.0.0
|-- SD_MMC @ 2.0.0
|-- SPI @ 2.0.0
|-- Ticker @ 2.0.0
[... 152 lines cut ...]
[ 10619][D][WiFiMulti.cpp:344] resetFails(): [WIFI] Resetting failure flags
00:00:10 WiFI_info:   Connecting WiFi...
00:00:10 WiFI_info:   mDNS name: MiniWebRadio
00:00:10 setup: ....  connected to FRITZ!Box7490LF, IP address is 192.168.178.31
00:00:10 ftpServer:   Buffers allocated: 4622 bytes
00:00:10 connection to NTP failed, trying again
[ 10673][D][WiFiGeneric.cpp:1055] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 10690][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[ 10699][D][WiFiGeneric.cpp:1055] _eventCallback(): Ard�x��x�x�x�xx�xxx�x�x�������xx���xxxxx��xxx�xx����x�x�x��xESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420639b8
=> 0x420639b8: panic_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/panic_handler.c:121
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
=> 0x403c9700: ?? ??:0
load:0x403c9704,len:0xad0
=> 0x403c9704: ?? ??:0
load:0x403cc700,len:0x29d8
=> 0x403cc700: ?? ??:0
entry 0x403c9880
=> 0x403c9880: ?? ??:0
[   418][I][esp32-hal-psram.c:92] psramInit(): PSRAM enabled

TubaFonse, 2024-04-01:
weitere Details zu 1.) 
Processing esp32s3 (board: ESP32-S3-DevKitC-1-N16R8; platform: https://github.com/Jason2866/platform-espressif32.git#348a508; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ESP32-S3-DevKitC-1-N16R8.html
PLATFORM: Espressif 32 (2024.3.10+sha.348a508) > Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.0+sha.4123e20
 - framework-arduinoespressif32-libs @ 5.1.3+sha.e7771c75bd.sha.270b5dc
 - tool-esptoolpy @ 4.7.2
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 3.2.0
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-riscv32-esp @ 12.2.0+20230208
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 47 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32-audioI2S @ 2.0.0+sha.e9a294e
|-- ESP32-KCX-BT-EMITTER @ 0.0.0+20240128094011.sha.338bc78
|-- ESP32-DLNA-Client @ 0.0.0+20231205185914.sha.cdde817
|-- ESP32-IR-Remote-Control @ 1.0.10+sha.6cb54a9
|-- FFat @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- ac101
|-- es8388
|-- ftpsrv
|-- rtime
|-- tftLib
|-- websrv
|-- wm8978
|-- ArduinoOTA @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- FS @ 2.0.0
|-- Preferences @ 2.0.0
|-- SD_MMC @ 2.0.0
|-- SPI @ 2.0.0
|-- Ticker @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Retrieving maximum program size .pio\build\esp32s3\firmware.elf
Checking size .pio\build\esp32s3\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.8% (used 68060 bytes from 327680 bytes)
Flash: [======    ]  59.8% (used 2310377 bytes from 3866624 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = espota
Uploading .pio\build\esp32s3\firmware.bin
23:25:04 [DEBUG]: Options: Namespace(esp_ip='MiniWebRadio.local', host_ip='0.0.0.0', esp_port=3232, host_port=50254, auth='', image='.pio\\build\\esp32s3\\firmware.bin', spiffs=False, debug=True, progress=True, timeout=10)
23:25:04 [INFO]: Starting on 0.0.0.0:50254
23:25:04 [INFO]: Upload size: 2310736
Sending invitation to MiniWebRadio.local ..........
23:26:45 [ERROR]: No response from the ESP
*** [upload] Error 1

Arne65, 2024-04-02:
Moin Moin ,

PIO Set upload/monitor/test port von Auto auf Deinen Kommunikationsanschluss (COM) umstellen dann sollte das funktionieren

TubaFonse, 2024-04-02:
@schreibfaul1 
über USB kann ich ja flashen, das ist nicht das Problem, MWR verbindet sich auch mit dem WLAN, aber gleich danach läuft er in die Boot-Schleife, weil offensichtlich der Zugriff auf NTP nicht mehr funktioniert: 
00:00:10 setup: .... connected to FRITZ!Box7490LF, IP address is 192.168.178.31
00:00:10 ftpServer: Buffers allocated: 4622 bytes
00:00:10 connection to NTP failed, trying again

schreibfaul1, 2024-04-02:
das gab es eine Änderung
https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/354
das war vor zwei Monaten.

closed 2024-04-03 by TubaFonse

schreibfaul1, 2024-04-04:
prima, dann ist das Problem erledigt :-)