schreibfaul1/ESP32-MiniWebRadio #389: DLNA Issue with FritzBox and NAS (but Win10/11PC ok)

TubaFonse, 2024-04-01:
Ich glaube ich habe den Fehler beseitigt. Innerhalb des Verzeichnisbaums gibt es einen Ebenenzähler. Insgesamt gibt es 6 Ebenen. Wenn ein Server nicht antwortet wird die Ebene um eins zurückgezählt.  Aber wenn die Ebene 0 ist...
Und zusätzlich habe ich den Timeout für die Antwort des Servers verdoppelt, von 500ms auf 1000ms. 

V2 benutzt die Soap32 Lib von Yellobyte, für V3 habe ich eine eigene geschrieben. Mal sehen, wie ich dem DLNA Server sage, dass er ausführlicher antworten soll.

_Originally posted by @schreibfaul1 in https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/350#issuecomment-1953203113_

TubaFonse, 2024-04-01:
Hallo Wolle, 
Issue #350 konnte ich etwas eingrenzen, 
dazu habe ich 3 Medienserver eingerichtet: 
1.) FritzBox7490 als Medien-Server für Musik eingerichtet, einen USB-Stick mit MP3's beschrieben und an der FritzBox indiziert: 
- in der V2 (mit Soap32 Lib von Yellobyte) kann ich die MP3's abspielen, 
- in der V3 werden die MP3-Dateien nicht gefunden (die darüber liegenden Verzeichnisse schon). d.h. Select Level 2 findet nichts. 
2.) Synology NAS DS213J DSM Version 7 als Medien-Server für Musik eingerichtet,
- in der V2 (mit Soap32 Lib von Yellobyte) kann ich die MP3's abspielen, 
- in der V3 werden die MP3-Dateien nicht gefunden (die darüber liegenden Verzeichnisse schon). d.h. Select Level 2 findet nichts: 
[image]

3.) Windows10 bzw. Win11 als Medien-Server für Musik eingerichtet: 
- in der V3 kann ich die MP3's abspielen: 
[image] 

Vielleicht kann ja "jemand" versuchen, den Fehler zu reproduzieren. 
Ich kann erst dann weitere Details liefern, wenn das Flashen der aktuellen Firmware wieder funktioniert. 

Grüße, Alfons.

schreibfaul1, 2024-04-02:
Hallo Alfons,
das konnte ich reproduzieren, hab selber eine FritzBox.
Meine beiden anderen DLNA Server, Plex und MiniDLNA, bereiten keine Probleme. Ich schaue mir das an.

schreibfaul1, 2024-04-02:
Hi Alfons,
hab den Fehler gefunden, nach "Full Clean" wird es gehen.

vG Wolle

closed 2024-04-03 by TubaFonse