schreibfaul1/ESP32-MiniWebRadio #390: Error connecting to ntp.

hevet, 2024-04-02:
Since yesterday, the radio won't turn on. The log shows an error connecting to ntp and it loops. Nothing was changed, I restarted the router but nothing helped. Only the startup logo appears on the display.

00:06:22 WiFI_info:   Connecting WiFi...
00:06:22 WiFI_info:   mDNS name: MiniWebRadio
00:06:22 setup: ....  connected to T-Mobile, IP address is 192.168.1.109
00:06:22 ftpServer:   Buffers allocated: 4622 bytes
00:06:22 connection to NTP failed, trying again
[  6995][W][STA.cpp:157] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE

hevet, 2024-04-02:
@TubaFonse As a temporary solution, comment ESP.restart on line 2486 of the main.cpp file.

schreibfaul1, 2024-04-02:
There has indeed been a change to NTP recently. Instead of one NTP server as before, there are now three to choose from. The following has changed in platformio.ini: [image] The files rtime.cpp, rtime.h and main.cpp have been adapted. Please check whether everything is really up to date.

hevet, 2024-04-02:
@schreibfaul1 I have everything up to date, I downloaded the project again and compiled it.  The time on the display is displayed correctly, it looks as if it could not quickly download the time before starting.  As I commented ESP.restart, the radio starts correctly and the time is also loaded correctly.  Now there was a time change on the weekend but I don't know if it has any significance.

schreibfaul1, 2024-04-02:
The change was made two months ago. And the error is now occurring. I have changed the SNTP mode. Now the NTP connects faster. Does that help?

TubaFonse, 2024-04-02:
No - It doesn't help: 
[ 10292][I][WiFiMulti.cpp:262] run(): [WIFI] Connecting BSSID: C8:0E:14:0C:93:11 SSID: FRITZ!Box7490LF Channel: 11 (-47)
[ 10377][D][WiFiGeneric.cpp:1055] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 10389][D][WiFiGeneric.cpp:1055] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 10391][D][WiFiGeneric.cpp:1119] _eventCallback(): STA IP: 192.168.178.31, MASK: 255.255.255.0, GW: 192.168.178.1
[ 10398][I][WiFiMulti.cpp:282] run(): [WIFI] Connecting done.
[ 10400][D][WiFiMulti.cpp:283] run(): [WIFI] SSID: FRITZ!Box7490LF
[ 10402][D][WiFiMulti.cpp:284] run(): [WIFI] IP: 192.168.178.31
[ 10403][D][WiFiMulti.cpp:285] run(): [WIFI] MAC: C8:0E:14:0C:93:11
[ 10405][D][WiFiMulti.cpp:286] run(): [WIFI] Channel: 11
[ 10407][D][WiFiMulti.cpp:344] resetFails(): [WIFI] Resetting failure flags
00:00:09 WiFI_info:   Connecting WiFi...
00:00:10 WiFI_info:   mDNS name: MiniWebRadio
00:00:10 setup: ....  connected to FRITZ!Box7490LF, IP address is 192.168.178.31
00:00:10 ftpServer:   Buffers allocated: 4622 bytes
00:00:10 connection to NTP failed, trying again
[ 10443][D][WiFiGeneric.cpp:1055] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 10446][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[ 10448][D][WiFiGeneric.cpp:1055] _eventCallback(): Arduino Event: 3 - STA_STOP

hevet, 2024-04-03:
It seems that the ntp error has been fixed, but I will check it at home because it works at work.

schreibfaul1, 2024-04-03:
Thanks for the feedback, I suspect that the timezone_string in "settings.json" is missing or empty. That may have been the reason for this error.

TubaFonse, 2024-04-03:
Dear Wolle - My MiniWebRadio is playing now with the actual NTP-Workaround, many thanks! 
Best regards, Alfons

closed 2024-04-03 by hevet