schreibfaul1/ESP32-MiniWebRadio #392: Feature request: Multiple alarms

MyAir, 2024-04-07:
Hi Wolle
Do you ever plan to implement more than one alarm in your ESP32-MiniWebRadio? 

I had a Sangean radio alarm clock that died on me after 15 years of service. This alarm clock had two alarms where I could set each alarm per weekday. I need one alarm for Monday and Friday at a later time and another one for Tuesday, Wedensday and Thursday at an earlier time, but no commercial vendor offers that feature anymore. So I had to resort to building my own. This is where I came across your ESP32-MiniWebRadio and decided to order the parts (ESP32-S3, ILI9341 and MAX98357A) to build my own. While the parts were in transit I found another project called "Internet Radiowecker mit Touchscreen" from az-delivery. This sketch looked a bit more managable to me as a novice and already has two alarms that can be set. Pretty much what I need. So I managed to build myself a protoype based on that code which almost does what I need. But that project has other shortcomings like audio interruptions when updating the screen, alarms can only be set in the web interface and a quite outdated layout of both the TFT and the web interface. Plus it's not maintained anymore like your project is. So I would very much like to come back to your ESP32-MiniWebRadio eventually as it looks quite a bit better designed than the one I'm currently using. But only having one alarm is quite the show-stopper for me at the moment. 

My ideal WebRadio alarm clock would have the following additional features besides what you already built:
- 2 (or more) Alarms that can be set/repeated per individual weekday. Pretty much what you already implemented but more than one alarm.
- Tickbox to completely enable/disable each alarm.
- Maybe an additional single alarm that is active once and is not repeated.
- A snooze hard-button to stop the sound for 10min or so.
- Each alarm automatically stops after 1 hour with no user input.
- A on/off hard-button. If music is playing from an alarm, it stops the alarm and turns off. Otherwise it plays/stops the station/music-source last played.
- Hard-buttons for alarm 1 and 2. Pressing each brings me to the respective alarm settings panel. Long-Pressing enables/disables each alarm.

Before I invest hundreds of hours, trying to fully understand your code and figuring out how to add one or two more alarms, I was wondering if you might already have similar plans for your project. I might be able to implement some of the hard-buttons myself eventually, but figuring out how to add additional alarms in your design might be a bit too much for me to implement myself. But first I'd have to add SD card support to my hardware design anyway to even try out your code. So I'll stick to my prototype code for the moment and keep an eye on your wonderful project to see what else is coming.

I would apreciate some feedback if there is anything I would like to see is already planned for a future release.

Kind regards

Thomas

schreibfaul1, 2024-04-10:
Hi Thomas,
thanks for your suggestions. At the moment there are no plans to change anything about the alarm function, that is something for the future. In the loop() _alarmdays and _alarmtime are queried and compared with the current time. Perhaps this is a starting point for your own developments.
The use of buttons for different functions would be unproblematic with the ESP32-S3, but fails due to the lack of GPIOs with the ESP32.
Initially planned are a progress indicator for files being played from the SD card, as well as skipping forwards and backwards. And the reactivation of menuconfig for better settings in the WiFi area and for memory allocation. And the integration of the BT emitter into the display.

MyAir, 2024-04-12:
Thanks for the update
In that case I'll update my hardware and see if I can get your software up and running as well. 
I'll have a look at _alarmday and _alarmtime to see if I can manage upgrading to a second alarm myself.

closed 2024-04-12 by MyAir