schreibfaul1/ESP32-MiniWebRadio #394: A1S Audiokit with V3.01b

steevemadison75, 2024-04-14:
Hi , the last working version that work on my A1S Audiokit is v2.17 , i tried yesterday to compile your new version with visual code
and i get " no space left on device " , if i remember the audiokit has 4MB flash and 4MB psram , i did setup my platform.ini as usual ,
(same config on v2.17)
board_build.partitions = boards/miniwebradio4MB.csv    ; 4MB Flash
board = ESP32-Dev-4MB            ; 4MB Flash

i tried to comment AC101.h and WM8978.h in common.h to not use those library ,
i also put -1 on all bluetooth BT_EMITTER_ as i dont have this module , i even tried to comment 
the KCX_BT_Emitter.h , by doing this i get compilation error so i uncomment it

did find one version of the V3 that i could compile , it look like it worked , but i get garbage sound , (similar to a FM radio not on station) 

got any clue ??

PS im not new at your webradio , i was one of the first to use a rasberry tft , and you help me to calibrate the screen and i request
some function in the web search by tag search

thank for your work

schreibfaul1, 2024-04-15:
Hi @steevemadison75,
probably in Arduino V3 there is no default for the I2C frequency. I have now manually entered the 400KHz that used to be set automatically. My AI Thinker board with AC101 is now working. These are the values in common.h:

````c++
#ifdef CONFIG_IDF_TARGET_ESP32
    // Digital I/O used
        #define TFT_CS 22
        #define TFT_DC 5
        #define TFT_BL -1
        #define TP_IRQ 12
        #define TP_CS 13
        #define SD_MMC_D0 2 // cannot be changed
        #define SD_MMC_CLK 14 // cannot be changed
        #define SD_MMC_CMD 15 // cannot be changed
        #define IR_PIN -1
        #define TFT_MOSI 23 // TFT and TP (VSPI)
        #define TFT_MISO 19 // TFT and TP (VSPI)
        #define TFT_SCK 18 // TFT and TP (VSPI)

        #define I2S_DOUT 25 // pin 25 AC101, pin 26 ES8388
        #define I2S_DIN 35 // pin not used
        #define I2S_BCLK 27
        #define I2S_LRC 26  // pin 26 AC101, pin 25 ES8388
        #define I2S_MCLK 0

        #define I2C_DATA 33
        #define I2C_CLK 32
        #define SD_DETECT 34 // some pins on special boards: Lyra, Olimex, A1S ...
        #define HP_DETECT 39
        #define AMP_ENABLED 21

        #define BT_EMITTER_RX    -1  // TX pin - KCX Bluetooth Transmitter (-1 if not available)
        #define BT_EMITTER_TX    -1  // RX pin - KCX Bluetooth Transmitter (-1 if not available)
        #define BT_EMITTER_LINK  -1  // high if connected                  (-1 if not available)
        #define BT_EMITTER_MODE  -1  // high transmit - low receive        (-1 if not available)
#endif
````
If you have the new board with ES8388 you have to change it a bit. And yes, the board has 4MB FLASH and PSRAM, please perform a "Full Clean" beforehand so that the libraries are fetched again. The BT_Emitter triggered a click every second, this has now also been eliminated.

schreibfaul1, 2024-04-17:
Yes, unfortunately that is the case at the moment. We are still using a pre-compiled Arduino version. Once the bugs have been fixed and settings can be made via menuconfig again, the WiFi data throughput will be the same as with version 2.

steevemadison75, 2024-04-18:
oh , ok , thank's for your time , i'll wait for a fix at later time , keep up the good work.

P.S i'm using your radio everyday and i really enjoy it

closed 2024-06-01 by github-actions[bot]