schreibfaul1/ESP32-MiniWebRadio #395: Display und SD Probleme

MyAir, 2024-04-14:
Hi Wolle

Ich hab mich jetzt dran gemacht und dein V3 MiniWebRadio mal auf meine Hardware (ESP32-S3-N16R8 China Klon, ILI9341, [DevKitC-1 + Touchscreen Combiner PCB](https://github.com/DustinWatts/ESP32-S3_TFT_Combiner), MAX98357A bzw im Moment PCM5102A zum Testen) draufgespielt. Alles noch default vom neusten Master branch, nur ein paar anpassungen im common.h wegen den Pins, Wifi Credentials, DAC und TFT bzw. Touch controller. Beim ESP32-S3 musste ich noch eine Brücke löten um 5V Output am Pin zu haben. Bei der Antenne musste ich ein Teil der Antennenbahn frei kratzen und ein 8.5mm langer Draht anlöten um überhaupt eine Wifi Verbindung hizubekommen. Diese China ESP32-S3 scheinen einen sehr schlechten Wifi empfang zu haben und den Hack mit der Antenne hab ich neulich im Netz gefunen und bei meinen anderen Projekten schon erfolgreich angewendet. 

Da mein Combiner Board den RESET Pin vom TFT fix mit Pin 5 vom ESP32 verdrahtet musste ich in tft.cpp erst mal übergangsmässig ein `pinMode(5, INPUT_PULLUP);` codieren um überhaupt etwas anderes als ein weiss leuchtendes Display zu bekommen. Das muss ich mal noch anständig in die Software einpflegen aber scheint für den Moment mal zu funktionieren.

Das ganze Pinout vom Combiner Board ist ewas anders als dein Default und sieht in common.h wie folgt aus:
```
    // Digital I/O used
        #define TFT_CS           10
        #define TFT_DC            6
        #define TFT_BL            7 // at -1 the brightness menu is not displayed
        #define TP_IRQ           16
        #define TP_CS            15
        #define SD_MMC_D0        14
        #define SD_MMC_CLK        4
        #define SD_MMC_CMD        9
        #define IR_PIN            4  // IR Receiver (if available)
        #define TFT_MOSI         11  // TFT and TP (FSPI)
        #define TFT_MISO         13  // TFT and TP (FSPI)
        #define TFT_SCK          12  // TFT and TP (FSPI)
```
Plus, wie gesgt, der Pin 5 für den RESET Pin als INPUT_PULLUP.

Nachdem ich die Software geflashed habe ist aber die Anzeige komplett zerhackt und unlesbar. Auf dem PCM5102A höre ich zwar Musik von Hitradio SKW (scheint der default Sender zu sein) aber die Anzeige ist in 99% der Versuche extrem korrupt und unlesbar.

Der Start Screen sieht bei mir meistens so aus
[image]
und anschliessend wenn alles gestartet ist so:
[image]
Das einzige was ich halbwegs erkennen kann und Sinn macht auf dem Display ist der VU Meter der sich gemäss Sound beweget. 

Beim Rumprobieren hab ich 2 mal per Zufall einen anständigen farbigen Start Screen und danach folgendes hinbekommen. 
[image]
Einmal nach explizitem "Erase Flash and Upload". Aber ich vermute das ist eher Zufall mit dem Erase and Upload. Im Log seh ich bei Jedem Versuch immer Fehler das Dateien nicht gefunden werden können. Diese Dateien sind aber auf der SD vorhanden und lesbar wenn ich die SD am PC anschliesse und öffne.

Ich verwende den internen SD Slot vom ILI9341 und hab wie beschrieben die 3 SMD Wiederstände durch Drahtbrücken ersetzt und dann mit ca 20 cm langen DuPont Kabeln MISO, MOSI und SCK and die Pins 14, 9 und 4 vom ESP32-S3 angeschlossen. Wo der CS Pin vom ILI9341 hinkomen soll hab ich immer noch nicht herausgefunden. Aber da mir beim Abschneiden der Drahtbrücken die Lötpads vom CS Pin SMD Wiederstand auf dem Board abgerissen sind ist CS gar nicht mehr verbunden. Und auch bevor mir dieser Fehler mit den Lötpads passiert ist hatte ich bereits die gleichen Fehler. Bei den MISO, MOSI und SCK Leitungen mess ich inklusive meinen Lötbrücken gerade mal 0.1 Ohm vom Stecker bis zum jeweiligen Lötpunkt des SD Slots. Also scheint noch alles nötige korrekt verbunden zu sein. Die 3V3 und GND für die SD müssten ja vom TFT Board schon korrekt daher kommen denn da hab ich nix geändert.

Im Log sieht zuerst alles ziemlich normal aus bevor dann auf einmal Fehler wegend er SD Karte raportiert werden:
```
ESP32 Chip: ESP32-S3
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.01b  Apr 12/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 20 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 3742.0 MB by 3781.5 MB free
[   765][I][esp32-hal-ledc.c:131] ledcAttachChannel(): LEDC attached to pin 7 (channel 0, resolution 8)
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
[  1197][I][WiFiMulti.cpp:89] addAP(): [WIFI][APlistAdd] add SSID: MyAir
[  9885][I][WiFiMulti.cpp:133] run(): [WIFI] scan done
[  9886][I][WiFiMulti.cpp:138] run(): [WIFI] 3 networks found
[  9888][I][WiFiMulti.cpp:262] run(): [WIFI] Connecting BSSID: B8:BE:F4:1D:40:B8 SSID: MyAir Channel: 6 (-54)
[ 11957][W][WiFiGeneric.cpp:1081] _eventCallback(): Reason: 203 - ASSOC_FAIL
[... 84 lines cut ...]
Ich hab bereits mit sehr vielen Kombinationen von TFT_FREQUENCY und SDMMC_FREQUENCY rumgespielt und bin auch schon auf 10'000'000 runter gegangen ohne Erfolg. Mein letzter Versuch wo die Anzeige hablwegs intakt ist war mit 40Mhz TFT und 20Mhz SDMMC. 
SD Karten hab ich schon 2 verschiedene Probiert, eine ältere 4Gb und eine neuere 32Gb, FAT32 formatiert und beide mit dem Inhalt vom neusten Content_Of_SD_Card.zip bespielt. Macht alles keinen Unterschied.

Jetzt steh ich etwas am Berg wo hier der Fehler liegt. 

Ich vermute mal dass das TFT Display generell OK sein sollte und die Störungen beim Lesen von der SD Karte her kommen. Zumindest funktioniert das Display tadellos wenn ich eine Software von einem anderen Radiowecker Projekt draufspiele das keine SD Karte verwendet.

Ich schätze mal das am Anfang die stations.csv halbwegs gelesen werden kann und dann die Daten aus irgend einem Grund korrupt daher kommen. Darum sieht wohl der Start Screen so zerhackt aus. Aber da die ganzen "file ... not found" Feher aus Unterverzeichnissen der SD Karte kommen frag ich mich ob das evtl. was mit SPIFFS zu tun haben könnte. Ich hab mal was gelesen dass Unterverzeichnisse dort Probleme machen können. stations.csv liegt ja im Root Verzeichnis und alle Fehler sind wegen Files aus Unterverzeichnissen. Wo der start Screen her kommt hab ich noch nicht gesucht. Zumindest liegt der nicht im Root Verzeichnis. 

Hast du eine Idee was der Fehler sein könnte bzw. in welche Richtung ich suchen soll?

Siehst du offensichtliche Probleme bei meinem Pinout vom TFT das vom Combiner Board vorgegeben ist?

Die Pins 4, 9 und 14 waren grad noch frei und gut erreichbar am ESP32 um die SD anzuschliessen. Sind die evtl. ungeeignet? An der anderen Seite vom ESP hab ich noch die ganze Reihe frei und könnte auch dort mühsam noch Pins anschliessen falls nötig.
Soll ich mal einen Versuch mit einem verlöteten MicroSD Adapter und einer neuen MicroSD probieren ob das was bringt?

Ich hätte zwar einen Billig Logic Analyzer, China Oszi und ein brauchbares Multimeter rumliegen aber wüsste nicht wo ich da was sehe und was richtig oder falsch aussieht.

Du hast in einem anderen Issue mal was erwähnt das beim ESP32 nicht alle Pins frei verwendet werden können. Aber für den ESP32-S3 hab ich noch nie irgendwelche Hinweise gesehen welche Pins wofür verwendet werden können und welche nicht. Seh ich das richtig dass im Prinzip beim ESP32-S3 alle GPIOs für alles benutzt werden können und höchstens die 2 USB und JTAG behindern könnten?

Sorry wenn ich dich grad mit Fragen überflute. Aber ich bin Dankbar für jeden Hinweis.

Danke und gruss

Thomas

schreibfaul1, 2024-04-15:
Hallo Thomas,
danke für deinen Erfahrungsbericht. Dass die chinesischen Boards einen schlechten Empfang haben, habe ich hier schon oft gelesen. 
Der +5V Pin ist nur als Eingang gedacht, das habe ich auch. Auf dem Board gibt es eine Diode, die ich einfach überbrückt habe, um die 5V am Pin zu erhalten.

Wenn ich mir das untere Bild angucke sind die Farben invertiert. Das kannst du in common.h ändern. Wäre die SD Karte beschädigt oder die Verkabelung nicht i.O. würdest du soweit gar nicht kommen. Also wird die SD Karte erkannt. Allerdings kann nichts gelesen werden, das ist komisch. Formatiere bitte die SD neu und entpacke die .zip Datei noch einmal. Guck mal bitte, ob die Datenstruktur auf der SD dann so ist, wie es das Entpackprogramm anzeigt.
CS wird von SD_MMC nicht benutzt, kann also frei bleiben.

MyAir, 2024-04-15:
Hi Wolle
Also das mit dem invertierten Display war glaub ich tatsächlich der Fall. Aber das hat mich auch noch nicht ans Ziel geführt.
Mit der 4Gb SD Karte im SD Slot vom Display bring ich relativ konstant folgende Screens hin wenn ich einen "Erase flash and upload" mache:
[image]
und danach
[image]

Aber danach kommen immer die üblichen "File not found" und "Filesystem not mounted" Fehler im Log. Auch wenn das Display ausnamsweise mal nicht korrupt ist. Nach einem Reset oder wenn ich den USB frisch einstecke bekomm ich wieder das alte zerstörte Startbild und auch das Bild wenn dann das Webradio abgespielt wird sieht dann wieder gleich kaputt aus. Selten einmal bekomm ich auch einen farbigen Startscreen hin wenn ich den ESP mal für ein paar Minute ausgesteckt lasse und frisch anstecke.

Ich hab jetzt 3 verschiedene SDs probiert. Eine davon ist eine brandneue 32Gb micro SD Karte die ich sowohl am SD Slot vom Display als auch mit direkt angelöteten Kabel am MicroSD Adapter angeschlossen hatte. Über den MicroSD Adapter hatte ich weniger konsistente Ergebnisse und in den meisten Fällen nur die zerrissenen Screens. Egal ob ich den Flash vorher gelöscht habe oder einfach eingesteckt.

Eine SD hab ich sogar mal komplett formatiert und neu bespielt und auch geprüft ob die Files im Log auch wirklich vorhanden und lesbar sind. Auch die Datenstruktur ist bei allen SD identisch mit dem .zip File.

Ich habe auch bereits den ESP32, das Combiner Board und das Display übers Kreuz mit meinem zweiten Hardware Set getauscht das mit meiner Radiowecker Software erfolgreich läuft. Das Ergebnis is immer das Gleiche. Im besten Fall bekomme ich einen schön farbigen Startbildschirm wenn ich den Flash vorher lösche und gleich neu bespiele. Ansonsten zerrissene Bilder oder zum Teil auch nur einen weissen Screen.

Was ich auch schon erfolglos getestet habe ist ein 1A und ein 2A USB Ladegerät direkt anzuschliessen um einen Brownout auszuschliessen. Mit meinen DACs hatte ich schon brownout Probleme wenn ich den PC USB Anschluss verwende. Aber bei meinen Tests hatte ich in den meisten Fällen gar keinen DAC angeschlossen. 

Eben hab ich noch spasseshalber die SDMMC Geschwindikeit auf 80Mhz erhöht und am 2A USB Lader angeschlossen und hab sogar da beim ersten Versuch den farbigen Startscreen und den "normalen" Betreibsscreen bekommen aber ohne die Icons etc. die ja zu diesem Zeitpunkt noch nie von der SD gelesen werden konnten. Und nach einem Reset wieder zurück beim alten zerrissenen Screen. Das Log kann ich halt nicht sehen wenn ich direkt am USB Lader angeschlossen bin aber von den fehlenden Icons geh ich davon aus dass auch da die selben Fehler auftreten. 

Ich bin also unterm Strich immer noch gleich weit. Beim Aufstarten kann deine Software die SD noch teilweise lesen und je weiter im Boot-Vorgang desto weniger ist lesbar. Zumindest sieht es für mich so aus. Die Hardware kann ich glaub ich aufgrund meiner diversen Austauschversuchen so gut wie ausschliessen. Da müssten grad beide China ESP32-S3 den gleichen Hardware-Fehler haben und/oder ich hab beim verlöten vom Combiner Board bei beiden den gleichen Fehler gemacht. Aber beide Fehler müssten dann so versteckt sein dass sie bei meiner anderen Radiowecker Software keine Rolle spielen. Das wäre dann doch ein arger Zufall wenn dem so wäre. Und mein `pinMode(5, INPUT_PULLUP);` Hack mit dem Pin 5 der fix mit dem RESET Pin vom Display verbunden ist wird ja wohl auch kaum zu so einem komischen Verhalten führen denk ich.

Könntest Du nochmal einen Blick auf mein common.h werfen ob Dir da was auffällt was vom Pinout und den Settings nicht funktionieren kann?

```
#define DECODER             1                               // (1)MAX98357A PCM5102A CS4344... (2)AC101, (3)ES8388, (4)WM8978
#define TFT_CONTROLLER      0                               // (0)ILI9341, (1)HX8347D, (2)ILI9486a, (3)ILI9486b, (4)ILI9488, (5)ST7796, (6)ST7796RPI
#define DISPLAY_INVERSION   0                               // (0) off (1) on
#define TFT_ROTATION        3                               // 1 or 3 (landscape)
#define TFT_FREQUENCY       40000000                        // 80000000, 40000000, 27000000, 20000000, 10000000
#define TP_VERSION          1                               // (0)ILI9341, (1)ILI9341RPI, (2)HX8347D, (3)ILI9486, (4)ILI9488, (5)ST7796, (3)ST7796RPI
#define TP_ROTATION         1                               // 1 or 3 (landscape)
#define TP_H_MIRROR         0                               // (0) default, (1) mirror up <-> down
#define TP_V_MIRROR         0                               // (0) default, (1) mittor left <-> right
#define AUDIOTASK_CORE      0                               // 0 or 1
#define AUDIOTASK_PRIO      2                               // 0 ... 24  Priority of the Task (0...configMAX_PRIORITIES -1)
#define I2S_COMM_FMT        0                               // (0) MAX98357A PCM5102A CS4344, (1) LSBJ (Least Significant Bit Justified format) PT8211
#define SDMMC_FREQUENCY     20000000                        // 80000000, 40000000, 27000000, 20000000, 10000000 not every SD Card will run at 80MHz
#define FTP_USERNAME        "esp32"                         // user and pw in FTP Client
#define FTP_PASSWORD        "esp32"
#define CONN_TIMEOUT        500                             // unencrypted connection timeout in ms (http://...)
#define CONN_TIMEOUT_SSL    2000                            // encrypted connection timeout in ms (https://...)

    // Digital I/O used
[... cut ...]
        #define SD_DETECT        -1  // some pins on special boards: Lyra, Olimex, A1S ...
        #define HP_DETECT        -1
        #define AMP_ENABLED      -1

        #define BT_EMITTER_RX    45  // TX pin - KCX Bluetooth Transmitter (-1 if not available)
        #define BT_EMITTER_TX    38  // RX pin - KCX Bluetooth Transmitter (-1 if not available)
        #define BT_EMITTER_LINK  19  // high if connected                  (-1 if not available)
        #define BT_EMITTER_MODE  20  // high transmit - low receive        (-1 if not available)

```

MyAir, 2024-04-17:
Hi Wolle

Ich hab soeben rausgefunden dass meine Pinbelegung offenbar suboptimal ist! 

Im meiner Verzweiflung hab ich jetzt mal rumgespielt mit anderen Pins für die SD Karte. 
Zuerst hab ich folgendes probiert.
```
        #define SD_MMC_CLK       36 //SCK
        #define SD_MMC_D0        37//MISO
        #define SD_MMC_CMD       38 //MOSI
```
Aber das hat gar nicht funktioniert. Sogar der Monitor hat nur noch Schrott ausgespuckt dass ich das Log gar nicht mehr sehen konnte.

Aber mit diesem Setup bin ich einiges weiter gekommen:
```
        #define SD_MMC_CLK       39 //SCK
        #define SD_MMC_D0        40 //MISO
        #define SD_MMC_CMD       41 //MOSI
```
Hier bekomme ich einen anständigen Start-Screen und auch alle Icons können von der SD korrekt gelesen werden. Aber nur wenn ich die Firmware mit "Erase flash and upload" hochlade. Danach stimmt mal alles, ausser dass das Touch Panel wohl noch falsch invertiert ist.

Nach einem Reset bekomm ich wieder den alten, zerstörten Screen. Nur im Log seh ich ebenfalls keine "file not found" Fehler mehr. 
Bei einem Power Cychle (USB raus und wieder rein) bekomm ich meistens nur einen weissen Screen und ebenfalls keine Lesefehler mehr im log. 

Somit kann ich also mit grösserer Wahrscheinlichkeit sagen dass bei meinem Pinout einiges nicht optimal ist. Zumindest für mein China ESP32-S3. 

Kannst du etwas dazu sagen wie Du zu deinem Pinout für den ESP32-S3 und all die Komponenten (TFT, DAC, BT etc.) gekommen bist? 
Gibt es irgendwelche Dokumentation dazu wo ich nachelsen könnte was beim ESP32-S3 auf welchem Pin möglich ist und was nicht bzw. welche pins ich besser nicht verwenden soll für SD, TFT, Touch und I2S?

Ich hab zwar schon einige pinout Diagramme vom ESP32-S3 gefunden mit lustigen Bezeichnungen bei den GPIOs wie FSPIIO4-7 oder SPIIO6-7 etc. Aber ich blick da überhaupt nicht durch. Die Dokumentationen von Expressif versteh ich auch nicht wirklich was da relevant ist in meinem Fall und was nicht. 
Bis jetzt hab ich einfach darauf geachtet möglichst keine U0... und U1... pins zu belegen da diese wahrscheinlich von den USB Anschlüssen belegt sind. Obwohl ich GPIO15-18 bereits für Touch und I2S verwende was aber gemäss Diagramm schon für U0RTS, U0CTS, U1TXD und U1RXD belegt ist.
Hier wäre ich dankbar für ein paar Hinweise worauf ich achten muss und wo es egal ist was ich verwende.

Für mich heisst das jetzt wohl dass ich mir ein neues Combiner Board machen muss das mit deinem Pinout überein stimmt. Es sei denn ich kann Software mässig noch was machen um mein Pinout trotzdem zum Laufen zu kriengen. Das Pinout für den TFT und Touch ist bei mir ja vom Combiner Board fix und passt wohl auch nicht ganz wenn ich trotzdem immer wieder diese korrupte Anzeige bekommen. Das dürfte noch eine Weile dauern bis ich KiCad verstehe und ein neues Board bestellt hab. Aber bevor ich das mache wäre es schön wenn Du mich da noch etwas erleuchten könntest wegen dem Pinout.

Ein experimental Setup auf einem Breadboard ist auch kaum möglich da der ESP32 auf einem Breadboard kaum mehr eine Wifi Verbindung bekommt da die Antenne zu nah an den ganzen Metallstreifen vom Breadboard ist. Und den ganzen ESP über längere Kabel am Breadboard anzuschliessen ist wohl auch nicht die beste Idee wegen Signalverlust und Störungen etc. So muss ich Dein Projekt wohl erst mal auf Eis legen bis ich das geschafft habe. Schade. 

Da mach ich jetzt besser mit meiner anderen Radiowecker SW weiter und versuch evtl. deine Technik den Audio Task auf einem eigenen Core laufen zu lassen abzukupfern weil bei dieser SW der Ton ständig unterbricht wenn das TFT aktualisert wird. Auch wieder eine eigene Mission für mich...

Danke und Gruss
Thomas

schreibfaul1, 2024-04-17:
Schade, dass du solche Probleme mit der HW hast. Ich denke immer noch, es ist ei Problem mit dem SD Kartenadapter. Da gibt es irgendwo eine Berührung, einen Übergangswiderstand oder die Betriebsspannung ist nicht stabil.
Ich benutze mit Erfolg solche Adapter:
[image]

MyAir, 2024-05-02:
Hi Wolle

Wie schon gesagt hat die neue IDF 5.1.2 also all meine Probleme mit der SD Karte behoben.
Betreffend der komischen Anzeige auf dem ILI9341 hab ich das Problem jetzt auch beheben können. Da ja wie gesagt der RST Pin durch mein Combiner Board mit dem Pin 5 vom ESP32 verbunden war hatte ich ja ursprünglich einen ``pinMode(5, INPUT_PULLUP);`` in tft.cpp codiert. Das war da wohl nicht ganz an der richtigen Stelle. Ich habe das Statement jetzt mal ins main.cpp, direkt vor den ``tft.begin(TFT_CS, TFT_DC, FSPI, TFT_MOSI, TFT_MISO, TFT_SCK);`` Aufruf verschoben. Seit da ist auch die Display Anzeige nicht mehr gestört. Weiss jetzt nicht ob das vom Timing her etwas knapp war gleich innerhalb vom tft.cpp den RESET Pin auf Pullup zu setzen aber so funktioniert das erstmal. Ich hab auch schon Hinweise im Netz gelesen dass der ILI9341 manchmal etwas empfindlich reagiert wenn der RST zu früh oder zu spät auf Pullup gesetzt wird.

Einzig beim Touch hab ich noch Probleme mit der Ausrichtung und zum Teil sehr hohen Y koordinaten Werten. Vielleicht muss ich da mal noch die TFT Frequenz reduzieren. Aber ich komme in den nächsten Wochen nicht mehr dazu das zu testen.

Ich schliesse das Issue somit mal da bei mir vorerst mal alles läuft.

closed 2024-05-02 by MyAir