schreibfaul1/ESP32-MiniWebRadio #396: Radio reset due to discharge of static electricity

clear-sky, 2024-04-15:
If at a distance of less than one meter, the static radio electricity of the body is discharged due to contact with metal objects, the radio will be reset. Have you had such an experience?

schreibfaul1, 2024-04-15:
An electrostatic discharge must not lead to a reset. Please take a look at the log at the very beginning: 
[image]

 There you can see the field strength with the connected network. It should never be lower than -50dB. Some users complain about poor reception with Chinese S3 boards.

clear-sky, 2024-04-15:
Hi Dear wolle!
thanks for the answer  
you are right -- my mcu is ESP-S3-12K 8m --my radio is on my desk and distance between radio and wifi router is 1m but rssi is something between -49 to -54
sometimes when i sit on the chair or touch metal objects radio resets!
what is your opinion using an antenna? something like this one:
[image]

schreibfaul1, 2024-04-15:
Have a look at the issue
"Display und SD Probleme #395"
The user has the same problems and has scratched the aerial free and soldered on a short piece of wire. No idea if it helps, but it might work.

clear-sky, 2024-04-16:
Hi dear Wolle!
thanks for the answer-I plan to use a metal box for the radio, so the WiFi reception will be worse, I will also use a WiFi antenna, if the result is good, I will announce here.

closed 2024-04-16 by clear-sky

clear-sky, 2024-04-22:
Hi dear Wolle!
finally my radio is finished!
i made MWR in a metal case and used wifi antenna  
reception is good (there is no reset)-i can not say excellent because i have problem in in serial monitor and can not see real resault
[image]
[image]
[image]
[image]
[image]
[image]

schreibfaul1, 2024-04-22:
This has turned out very nicely.

The monitor speed is 512000. I have increased it so that not too much time is used for the output. The serial terminal in VSCode gets the `monitor_speed` from platfornio.ini

clear-sky, 2024-04-22:
Hi dear valera39!
that box below of radio is for switch off and on my pc1 pc2  project radio and etc via 5 SSR and high and low voltage sensor with lm324 
as you see FM radio is pre installed!!!
RTC is DS3234 with spi bus

valera39, 2024-04-22:
Is FM information displayed on the display?

clear-sky, 2024-04-22:
FM info is several parts: freq and vol info can be obtained from data bus(I2C)
other is called RDS that contains some info like station name-program info-program start and ending time etc
i live in iran and rds only gives me station name 
bad news is fm audio path is directly from rda5807fp to pcm5102 though there is no possibility to have level meter
fm radio and rtc work but not combined to wolles code!
This design is not final and may have changes. I made this to make my desk a little quieter!!!
Mr. Wolle may consider a place as a gallery for users to post their final designs so that others can better get ideas.

schreibfaul1, 2024-04-22:
If you would like to publish the photos, I would be happy to do so with a link on the homepage

clear-sky, 2024-04-22:
I will definitely send you more complete pictures

clear-sky, 2024-04-22:
hi dear Wolle!
I tried to make the radio with simplest tools that every electronic user have it
sent some pictures! please check your mail!

clear-sky, 2024-04-22:
now i can say that wifi antenna is working well!!!
this is my rssi
`[  9423][I][WiFiMulti.cpp:263] run(): [WIFI] Connecting BSSID: 02:10:18:01:00:02 SSID: darmstadt Channel: 1 (-31)`