schreibfaul1/ESP32-MiniWebRadio #397: music center?

valera39, 2024-04-15:
Dear Wolle!
Thanks to your hard work, MiniWeb Radio is now able not only to play internet radio, but also to play files from SD card and bluetooth. Do you have any desire and possibilities to add FM Radio, thus turning the project into a real music center?
With respect and further success!

schreibfaul1, 2024-04-17:
Hi valera39,
FM radio no longer plays a major role in some countries, rather DAB+. I don't know what HW is available for it. I'll make a note of the idea for later.

clear-sky, 2024-04-22:
In developed countries, DAB has replaced with FM radio, but in other countries, FM radio will work for years
FM radio is old but reliable

valera39, 2024-04-22:
For example, China still has SW-band radios and Poland has LW-band radios.

clear-sky, 2024-04-22:
note that ONLY RDA5807FP has I2S output there is another module with RDA5807M that has only analog output
[image]

clear-sky, 2024-04-22:
may be 74hc157 not be necessary -esp32 has 2 i2s and one of them can be used as fm input - i think wolle is skilled enough to do it-in this condition level meter will be available in fm mode

clear-sky, 2024-04-23:
the advantage of using 74hc157 is that it does not need any software  Engineering of esp32 ! besides using different menu helps the user to know what he/she is using insteed of using a text like internet radio or fm radio 
this is simple and nice looking idea:
[image]

valera39, 2024-04-23:
[clear-sky](https://github.com/clear-sky) Very cool, I wish the esteemed Wolle would pay attention to it.

closed 2024-06-07 by github-actions[bot]