schreibfaul1/ESP32-MiniWebRadio #398: IR Remote Control issue, only with ESP32 TTGO-T8 V1.7

TubaFonse, 2024-04-20:
Hallo Wolle @schreibfaul1 - "Großer Meister" bitte um Hilfe , diesmal in Deutsch; ist für mich leichter, mein Problem etwas klarer zu schildern - es wäre super, wenn dies jemand mit dem ESP32 TTGO-T8 ausprobieren könnte, da ich alleine nicht mehr weiter komme ... 

Hardware: ESP32 TTGO-T8 V1.7 Modul (D0WDQ6-V3 ext 4M-Flash ext 4M-PSRAM) + ILI9341 + PCM5102A + IR VS1838B. 

Mit der IR-Fernbedienung funktioniert das MiniWebRadio-V3 (das Umschalten der Sender und sonstiger Funktionen) nur sporadisch, d.h. mehrmaliges Drücken der gleichen Taste auf der FB ist nötig, bis der ESP32 irgendwann einmal den Code/das Signal vom VS1838B erkennt. Schon bei den IR-Settings über das WEB-Interface (Programmieren der Tastenfunktionen) habe ich das gleiche sporadische Fehlverhalten. Auf dem Serial Monitior sehe ich den Code entsprechend auch nur sporadisch (habe mit CORE_DEBUG_LEVEL=4 d.h. DEBUG compiliert). Bei diesen Aussetzern wird kein Fehler angezeigt - d.h. von 10x Drücken der Taste auf der FB wir das Signal wenn ich Glück habe nur 3x vom ESP32 erkannt und am Serial Monitor korrekt ausgegeben. 
Für den IR-Pin habe ich statt GPIO 35 am ESP32 versuchsweise mal einen anderen freien GPIO (z.B. IO 4) probiert - jedoch ohne Erfolg - um Konflikte mit etwaigen Restriktionen auf dem Board auszuschließen, da dort der IO 35 beim ESP32 TTGO-T8 mit jeweils 100kOhm zwischen VBAT und GND beschaltet ist. Die Versorgungsspannungen, auch am IR-Empfänger sind stabil und sauber, die Batterien der FB sind in Ordnung, identisches Fehlverhalten habe ich auch mit anderen Fernbedienungen (NEC-Code). Was nun? 

Mittlerweile befürchte ich, dass evtl. das ESP32 TTGO-T8 V1.7 Modul trotz 4M-Flash und 4M-PSRAM im aktuellen MWR-V3 Funktionsumfang an der Leistungsgrenze ist, wenn diese Aussetzer mit der IR-Fernbedienung auftreten - was meinst Du?  
Anregung/Vorschlag: 
Es wäre gut, wenn man Funktionen wie das VU-Meter, die ggf. viel Prozessorleistung schlucken, bei der Konfiguration z.B. common.h vor dem Compilieren deaktivieren zu können, um die Fehlersuche bei etwaigen Performance-Problemen zu erleichtern. Auch könnten mit dieser Option aktuelle Funktionen des MWR-V3 auf älterer ESP32-Hardware evtl. wieder zum Laufen gebracht werden. 
Das "VU-Meter" finde ich ja ganz "nett", würde aber lieber darauf verzichten, bevor Grundfunktionen des tollen MWR-V3 nicht mehr stabil funktionieren. 
Frage: Wäre aus Deiner Sicht des "Fachmanns" eine solche individuelle "Konfigurierbarkeit" mit vertretbarem Aufwand realisierbar? 
Ich bin mir sicher, dass dies auch bei anderen Nutzern in diesem Forum großes Interesse wecken könnte. 

Vergleich: Mit einem ESP32-_S3_ WROOM-1-_N16R8_ funktioniert mit gleicher Fernbedienung und gleichem IR-Empfänger alles sofort beim ersten Tastendruck - Performance-Engpass oder Konflikte mit Restricted GPIO's wären beim schwächeren _ESP32_ mit 4M-Flash und 4M-PSRAM demnach denkbar - aber wie ist dies mit den Möglichkeiten in PlatformIO messbar und darstellbar? Als Anfänger kenne ich mich da zu wenig aus und hoffe auf Deine geduldige Hilfe ... 

Danke & Grüße - Alfons.

TubaFonse, 2024-04-20:
Jetzt habe ich mal eine kurze, ungewöhnliche Ausgabe auf dem Serial Monitor, während das Radio ein paar Stunden lief 
 - OHNE dass ich die Fernbedienung betätigt habe!!! - 
vielleicht hilft das zur Fehlersuche weiter (compiliert mit CORE_DEBUG_LEVEL=4): 

21:15:20 StreamTitle: Chairmen Of The Board - Give Me Just A Little More Time
[10722646][D][IR.cpp:152] loop(): something went wrong, intval_l 1552459160, intval_h 0, pulsecounter 24
[10725741][D][IR.cpp:152] loop(): something went wrong, intval_l 3, intval_h 0, pulsecounter 24

Gruß, Alfons.

TubaFonse, 2024-04-20:
beim Sender-Umschalten (über Touch-Display) hängt sich das aktuelle MWR-V3 auch manchmal auf: 

00:09:03 StreamTitle: Peter, Paul & Mary - Lemmon Tree
00:09:05 action: ...  switch to station 8
00:09:05 AUDIO_info:  Connect to new host: "http://frontend.streams.arabella.at/arabella-goldenoldies/stream/mp3"
[1582149][V][ssl_client.cpp:342] stop_ssl_socket(): Cleaning SSL connection.
00:09:05 AUDIO_info:  buffers freed, free Heap: 94292 bytes
[1582164][D][NetworkManager.cpp:131] hostByName(): DNS found IPv4 88.209.21.46
00:09:05 AUDIO_info:  Connection has been established in 44 ms, free Heap: 94080 bytes
00:09:05 AUDIO_info:  redirect to new host "http://edge13.streams.arabella.at/arabella-goldenoldies/stream/mp3"
00:09:05 AUDIO_info:  Connect to new host: "http://edge13.streams.arabella.at/arabella-goldenoldies/stream/mp3"
[1582233][V][ssl_client.cpp:342] stop_ssl_socket(): Cleaning SSL connection.
00:09:05 AUDIO_info:  buffers freed, free Heap: 91052 bytes
[1582326][D][NetworkManager.cpp:131] hostByName(): DNS found IPv4 57.128.47.6
00:09:05 AUDIO_info:  Connection has been established in 114 ms, free Heap: 94020 bytes
[1584859][E][Audio.cpp:3659] parseHttpResponseHeader(): timeout
00:09:10 HTML_info:   "change_state=0"
00:09:10 HTML_info:   "getmute"

TubaFonse, 2024-04-20:
Die häufigen "txt overflow" (Abschneiden des StreamTitle) auf dem Display, 
kann das noch bereinigt werden? Abhängig vom Decoder (AAC) oder von der Größe des Displays? 
Oder sind das Fehler, die vom jeweiligen Sender verursacht, d.h. nicht ordnungsgemäß generiert / übertragen werden? 

00:17:14 action: ...  switch to station 7
00:17:14 AUDIO_info:  Connect to new host: "http://edge05.streams.arabella.at/arabella-austropop"
[2070992][V][ssl_client.cpp:342] stop_ssl_socket(): Cleaning SSL connection.
00:17:14 AUDIO_info:  buffers freed, free Heap: 94728 bytes
[2071067][D][NetworkManager.cpp:131] hostByName(): DNS found IPv4 51.210.216.57
00:17:14 AUDIO_info:  Connection has been established in 95 ms, free Heap: 94136 bytes
00:17:14 icy-descr:   Arabella Austropop
00:17:14 StationName: Arabella Austropop
00:17:14 icy-url: ..  https://www.arabella.at
00:17:14 AUDIO_info:  AACDecoder has been initialized, free Heap: 65984 bytes , free stack 4588 DWORDs
00:17:14 lastURL: ..  http://edge05.streams.arabella.at/arabella-austropop
00:17:14 AUDIO_info:  stream ready
00:17:14 AUDIO_info:  syncword found at pos 0
00:17:14 AUDIO_info:  Channels: 2
00:17:14 AUDIO_info:  SampleRate: 22050
00:17:14 AUDIO_info:  BitsPerSample: 16
00:17:14 AUDIO_info:  BitRate: 71662
00:17:14 AUDIO_info:  AAC HeaderFormat: ADTS
00:17:14 AUDIO_info:  AAC Codec: MPEG-4 LowComplexity
00:17:14 StreamTitle: DEPECHE AMBROS - I BIN A WEH
00:17:14 Audiocodec:  AAC
00:17:18 txt overflow, winHeight=100, strlen=28, written=23, str=DEPECHE AMBROS - I BIN A WEH
00:17:18 HTML_info:   "change_state=0"
00:17:18 HTML_info:   "getmute"
00:17:29 StreamTitle: ANDREAS GAUDMANN - PANIK
00:17:29 txt overflow, winHeight=100, strlen=24, written=19, str=ANDREAS GAUDMANN - PANIK

TubaFonse, 2024-04-20:
Bei MP3-Sendern erscheinen auch viele Fehlermeldungen (siehe MP3 decode error) im Serial Monitor, bevor gespielt wird - ist das "normal" oder sollte da auch noch was optimiert werden, um die Stabilität / Performance zu verbessern? z.B.: 

23:49:16 set_IR_cmd:  IR command 0x52, IR Button Number 0x08
23:49:25 ir_code: ..  IR address 0x00, IR command 0x4a
23:49:27 ir_result:   Stationnumber 9
23:49:27 action: ...  switch to station 9
23:49:27 AUDIO_info:  Connect to new host: "http://c3.auracast.net:8010/stream"
[404319][V][ssl_client.cpp:342] stop_ssl_socket(): Cleaning SSL connection.
23:49:27 AUDIO_info:  buffers freed, free Heap: 94656 bytes
[404350][D][NetworkManager.cpp:131] hostByName(): DNS found IPv4 213.32.121.34
23:49:27 AUDIO_info:  Connection has been established in 56 ms, free Heap: 94048 bytes
23:49:27 StationName: Best Oldies
23:49:27 bitRate:     128Kbit/s
23:49:27 icy-url: ..  http://www.shoutcast.com
23:49:27 AUDIO_info:  MP3Decoder has been initialized, free Heap: 57328 bytes , free stack 4588 DWORDs
23:49:27 lastURL: ..  http://c3.auracast.net:8010/stream
23:49:27 AUDIO_info:  stream ready
23:49:27 AUDIO_info:  syncword found at pos 0
23:49:27 AUDIO_info:  MP3 decode error -2 : MAINDATA_UNDERFLOW
23:49:27 AUDIO_info:  syncword found at pos 35
23:49:27 AUDIO_info:  syncword found at pos 0
23:49:27 AUDIO_info:  MP3 decode error -6 : INVALID_FRAMEHEADER
23:49:27 AUDIO_info:  syncword found at pos 59
23:49:27 AUDIO_info:  syncword found at pos 0
23:49:27 AUDIO_info:  MP3 decode error -8 : INVALID_SCALEFACT
23:49:27 AUDIO_info:  syncword found at pos 169
23:49:27 AUDIO_info:  syncword found at pos 0
23:49:27 AUDIO_info:  MP3 decode error -6 : INVALID_FRAMEHEADER
23:49:27 AUDIO_info:  syncword found at pos 131
23:49:27 AUDIO_info:  syncword found at pos 0
23:49:27 AUDIO_info:  MP3 decode error -6 : INVALID_FRAMEHEADER
23:49:27 AUDIO_info:  syncword found at pos 19
23:49:27 AUDIO_info:  syncword found at pos 0
23:49:27 AUDIO_info:  MP3 decode error -2 : MAINDATA_UNDERFLOW
23:49:27 AUDIO_info:  syncword found at pos 72
23:49:27 AUDIO_info:  syncword found at pos 0
23:49:27 AUDIO_info:  MP3 decode error -6 : INVALID_FRAMEHEADER
23:49:27 AUDIO_info:  syncword found at pos 173
23:49:27 AUDIO_info:  syncword found at pos 0
23:49:27 AUDIO_info:  MP3 decode error -9 : INVALID_HUFFCODES
23:49:27 AUDIO_info:  syncword found at pos 170
23:49:27 AUDIO_info:  syncword found at pos 0
23:49:27 AUDIO_info:  MP3 decode error -9 : INVALID_HUFFCODES
23:49:27 AUDIO_info:  syncword found at pos 417
23:49:27 AUDIO_info:  syncword found at pos 0
23:49:27 AUDIO_info:  MP3 decode error -9 : INVALID_HUFFCODES
23:49:27 AUDIO_info:  syncword found at pos 128
23:49:27 AUDIO_info:  syncword found at pos 0
23:49:27 AUDIO_info:  MP3 decode error -6 : INVALID_FRAMEHEADER
23:49:27 AUDIO_info:  syncword found at pos 6
23:49:27 AUDIO_info:  syncword found at pos 0
23:49:27 AUDIO_info:  Channels: 2
23:49:27 AUDIO_info:  SampleRate: 44100
23:49:27 AUDIO_info:  BitsPerSample: 16
23:49:27 AUDIO_info:  BitRate: N/A
23:49:27 Audiocodec:  MP3
23:49:28 AUDIO_info:  MP3 decode error -6 : INVALID_FRAMEHEADER
23:49:28 AUDIO_info:  syncword found at pos 95
23:49:28 AUDIO_info:  syncword found at pos 0
23:49:28 AUDIO_info:  MP3 decode error -9 : INVALID_HUFFCODES
23:49:28 AUDIO_info:  syncword found at pos 73
23:49:28 AUDIO_info:  syncword found at pos 0
23:49:28 AUDIO_info:  MP3 decode error -6 : INVALID_FRAMEHEADER
23:49:28 AUDIO_info:  syncword found at pos 28
23:49:28 AUDIO_info:  syncword found at pos 0
23:49:28 AUDIO_info:  MP3 decode error -8 : INVALID_SCALEFACT
23:49:28 AUDIO_info:  syncword found at pos 142
23:49:28 AUDIO_info:  syncword found at pos 0
23:49:28 StreamTitle: John Fred & His Playboy Band - Judy In Disguise
23:49:28 AUDIO_info:  MP3 decode error -9 : INVALID_HUFFCODES
23:49:28 AUDIO_info:  syncword found at pos 2
23:49:28 AUDIO_info:  syncword found at pos 0
[404662][I][mp3_decoder.cpp:1642] DecodeHuffmanPairs(): (tabType != invalidTab)
23:49:28 AUDIO_info:  MP3 decode error -9 : INVALID_HUFFCODES
23:49:28 AUDIO_info:  syncword found at pos 53
23:49:28 AUDIO_info:  syncword found at pos 0
23:49:28 AUDIO_info:  MP3 decode error -6 : INVALID_FRAMEHEADER
23:49:28 AUDIO_info:  syncword found at pos 114
23:49:28 AUDIO_info:  syncword found at pos 0
23:49:28 AUDIO_info:  Channels: 2
23:49:28 AUDIO_info:  SampleRate: 44100
23:49:28 AUDIO_info:  BitsPerSample: 16
23:49:28 AUDIO_info:  BitRate: 128000

TubaFonse, 2024-04-21:
@schreibfaul1 Achtung: Mein ESP32 TTGO-T8 V1.7 Modul, Chip: D0WDQ6-V3 hat ein 4M-Flash (ext) 
und ein ext **_8M_**-PSRAM (nicht 4M-PSRAM wie oben erwähnt). 

In der platformio.ini unter [ESP32] sind nur die unterschiedlichen Flash-Größen definiert, nicht jedoch die Größe und Art des PSRAM. Braucht man beim ESP32 keine zusätzlichen Optionen für die TTGO-Module mit unterschiedlichen Quad PSRAM Größen (4M, 8M) unter [ESP32]? 
[image]
Kann das evtl. die Ursache für mein o.g. IR-Problem sein, oder ist die Größe des PSRAM beim ESP32 für das MiniWebRadio V3 nicht relevant, sondern nur beim ESP32-S3? 

Gruß, Alfons.

schreibfaul1, 2024-04-21:
Hallo Alfons,
ziemlich viel text, ich fange mal bei Punkt 1  (IR Empfänger) an:
Wenn die Fernbedienung nur sporadisch funktioniert kann das mehrere Ursachen haben.
Es könnte z.B. ein konstruktives Problem sein. Die IR Empfänger sind für HF Störungen sehr empfindlich. Daher sind sie häufig in ein Metallgehäuse verkapselt. Vielleicht ist der Empfänger zu nahe an der WiFi Antenne? Oder der Signalanschluss führt an dieser vorbei? Ein Abstand von ~3cm wäre ausreichend. 
Es gibt die unterschiedlichsten IR Empfänger. Manche haben einen "open collector" Ausgang um mehrere parallel schalten zu können. Dann wird gegenüber einem TTL Ausgang ein PullUp Widerstand (1...10KOhm) benötigt. Gerade wichtig bei ESP32 "input only" Pin da dieser intern kein PullUp bereitstellt. Wie auch immer, es gibt manche, die funktionieren nicht richtig oder deren Frequenz ist nicht 38KHz.
Ich nehme gerne die:
[image]

gibts bei Am..on. Hatte noch nie Probleme damit. Du kannst die Fernbedienung auch gegen die Decke richten oder gegen die Wand, funktioniert immer.

Punkt2: ESP32 TTGO-T8 V1.7 habe ich auch im Einsatz in meiner Testumgebung. Nach meiner Erfahrung gibt es nur "hat PSRAM" oder "hat kein PSRAM". Es gibt eine Unterscheidung nach quad/octal wie beim S3. Die Größe wird automatisch erkannt, wenn dein Board 8MB hat, dann wird das auch iḿ Log stehen:
[image]
meins ist älter, hat nur 4MB

Punkt3: muss noch offen bleiben. Bisher wird anhand der Textlänge abgeschätzt welche Schriftgröße benutzt wird damit der Test in das Feld passt. Manchmal reicht es halt nicht, dann wird der Rest abgeschnitten und "txt overflow" erscheint.

Punkt4: die Fehlermeldungen der Dekoder. Im Prinzip wird ein Stream mit dem Beginn eines Rahmens eröffnet. Am Anfang eines Rahmens stehen Informationen für den Dekoder wir Samplerate oder Anzahl der Kanäle...
Wenn das nicht so ist, wird ein Synchronwort gesucht, der den Anfang eines Rahmens kennzeichnet. Manchmal ist das Synchronwort zufällig in der Audiosequenz enthalten. dann komme es zu diesen Fehlermeldungen. Das werde ich künftig reduzieren. Dafür gibt es Routinen wie `MP3FindFreeSync()` die hab ich bisher nicht in Benutzung.

beste Grüße
Wolle

TubaFonse, 2024-04-21:
Hallo Wolle, danke für Deine Infos zu Punkt 2 bis 4, 
Ergänzung zu meinem Punkt-1 (IR-Problem) und Deinen Tipps dazu: 
a) Meine IR-Fernbedienung und die beiden identischen IR-Empfänger funktionieren problemlos: Die FB sendet einwandfrei und beide IR-Empfänger geben die empfangenen Signale gleichzeitig an den jeweils konfigurierten IR-Eingang des ESP32 bzw. ESP32-S3 weiter! Es sieht so aus, wie wenn die decodierten IR-Signale immer am ESP32 GPIO 35 anliegen, jedoch vom ESP32 meist ignoriert werden, weil er zu "beschäftigt" mit anderen Tasks ist und dieser IR-"Interrupt" nur sporadisch "durchkommt", wenn gerade mal eine Lücke zwischen den Tasks ist.  
b) Um die Vermutung zu bestätigen, habe ich 2 Versuchsaufbauten im Abstand von 15cm hintereinander, die ich nur mit einer IR-Fernbedienung gleichzeitig steuere. Der eine Aufbau ist mit ESP32 TTGO-T8 V1.7, der andere mit ESP32-S3 WROOM-1-N16R8. 
Deshalb schließe ich mal Probleme auf der IR-Sender und IR-Empfänger-Seite aus, da ich untereinander die Bauteile beliebig tauschen kann - am ESP32-S3 funktionierts immer, am ESP32 nur ca. 1 mal von 10 Versuchen hintereinander. Ein Pull-Up von 10k am IR-Eingang des ESP32 ändert daran nichts. 
c) welche GPIOs verwendest Du in Deiner ESP32 TTGO-T8 V1.7 Testumgebung? Wenn bei Dir mit der aktuellen Konfiguration und FW die IR-Steuerung problemlos funktioniert, würde ich das mal auf dem Breadboard genauso nachbauen, um meinen Fehler einzugrenzen. Kann ja sein, dass genau mein Controller spinnt ... mein Board hat auch 4MB PSRAM laut Serial Monitor: 
[image]

Hier mein Versuchsaufbau: 
[image]
Grüße - Alfons.

TubaFonse, 2024-04-21:
Hallo Wolle, IR-Remote läuft jetzt mit meinem ESP32 TTGO-T8 V1.7 sauber !!!
Dank Deines Tipps "HF-Einstrahlung" habe ich nochmal meinen Aufbau kontrolliert. Ich hätte es wirklich nicht für möglich gehalten - mein IR-Empfänger am ESP32 war tatsächlich zu nah an der WiFi-Antenne. IR-Empfänger ein paar Zentimeter weiter weg platziert und schon funktioniert alles. Auch hat womöglich noch zusätzlich das Licht meiner LED-Deckenbeleuchtung Störungen im 38kHz-Bereich verursacht. Endlich wieder ein Erfolgserlebnis und ein Stück schlauer geworden!!! 

zu Punkt3 Problem "txt overflow" könnte man ja als durchlaufenden Text auf den Displays realisieren, wenn der Platz bei kleinster Schriftgröße nicht mehr ausreicht - wie bei anderen Internetradios auf dem Markt (wäre doch wieder eine Fleißaufgabe für Wolle ;-) 
zu Punkt4 "Fehlermeldungen der Decoder" hast Du ja auch schon Ideen zur Optimierung, freut mich ...

Dann werde ich dieses "IR Remote Issue" schließen, da ja Punkt3 und Punkt4 mit meinem ursprünglichen Problem nichts zu tun hat. 

Danke nochmals und Gruß - Alfons

closed 2024-04-21 by TubaFonse

schreibfaul1, 2024-04-21:
super, falls du die Möglichkeit hast, kannst du den IR Empfänger gegen den oben erwähnten TSOP4838 tauschen. Der ist um Größenordnungen empfindlicher und mach weniger HF Probleme.

TubaFonse, 2024-04-21:
danke, werde gleich welche bestellen - Gute Nacht!

schreibfaul1, 2024-04-22:
Ergänzung zu Punkt 4:
-ERR_MP3_MAINDATA_UNDERFLOW im MP3 Dekoder und Folgefehler

das habe ich jetzt abgefangen. Jetzt werden mehrere MP3 Frames "aufgefangen" bis eine Dekodierung sinnvoll ist. Die ganze Palette von Fehlermeldungen werden jetzt nicht mehr kommen.
Damit das wirksam wird ist  wie immer "Full Clean" erforderlich.