schreibfaul1/ESP32-MiniWebRadio #402: reset radio when using device monitor

clear-sky, 2024-04-22:
Hi Dear Wolle!
when i want to use device monitor radio drops in a loop of resets
this is my config in platformio.ini
```
[common]

monitor_speed = 115200  ; 921600, 512000, 460800, 256000, 115200
```
please look at my terminal output:
>> pio device monitor    
```

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Terminal on /dev/ttyUSB0 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
RON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
=> 0x403c9700: ?? ??:0
load:0x403c9704,len:0xad0
=> 0x403c9704: ?? ??:0
load:0x403cc700,len:0x29d8
=> 0x403cc700: ?? ??:0
entry 0x403c9880
=> 0x403c9880: ?? ??:0
[  1288][I][esp32-hal-psram.c:92] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-S3
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.01f  Apr 18/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 20 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Reset due to power-on event

00:00:00    
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00    
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ST7796
00:00:01 setup: ....  Init SD card
00:00:01 setup: ....  SD card found, 3827.0 MB by 3861.0 MB free
[  1889][I][esp32-hal-ledc.c:131] ledcAttachChannel(): LEDC attached to pin 10 (channel 0, resolution 8)
00:00:02 setup: ....  seek for stations.csv
00:00:02 setup: ....  stations.csv found
00:00:02 setup: ....  seek for WiFi networks
[  2926][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: darmstadt
[  9410][I][WiFiMulti.cpp:134] run(): [WIFI] scan done
[  9416][I][WiFiMulti.cpp:139] run(): [WIFI] 5 networks found
[  9423][I][WiFiMulti.cpp:263] run(): [WIFI] Connecting BSSID: 02:10:18:01:00:02 SSID: darmstadt Channel: 1 (-31)
[  9434][W][STA.cpp:548] disconnect(): STA already disconnected.
[  9538][I][WiFiMulti.cpp:283] run(): [WIFI] Connecting done.
00:00:09 WiFI_info:   Connecting WiFi...
00:00:09 WiFI_info:   mDNS name: MiniWebRadio
00:00:09 setup: ....  connected to darmstadt, IP address is 192.168.1.3
00:00:09 ftpServer:   Buffers allocated: 4622 bytes
00:00:09 rtime_info:  Initializing SNTP
03:30:09 rtime_info:  Waiting for system time to be set... (1/10)
03:30:11 rtime_info:  Waiting for system time to be set... (2/10)
00:37:02 rtime_info:  00:37:02
00:37:02 audiotask:   is pinned to core 0
00:37:02 audiotask:   priority is 2
00:37:02 setup: ....  Number of saved stations: 69
00:37:02 setup: ....  current station number: 4
00:37:02 setup: ....  current volume: 1
00:37:02 setup: ....  last connected host: http://uk3.internet-radio.com:8405/live
00:37:02 setup: ....  connection timeout: 500 ms
00:37:02 setup: ....  connection timeout SS�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
=> 0x403c9700: ?? ??:0
load:0x403c9704,len:0xad0
=> 0x403c9704: ?? ??:0
load:0x403cc700,len:0x29d8
=> 0x403cc700: ?? ??:0
entry 0x403c9880
=> 0x403c9880: ?? ??:0
[  1288][I][esp32-hal-psram.c:92] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-S3
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.01f  Apr 18/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 20 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Reset due to power-on event

00:00:00    
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00    
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ST7796
00:00:01 setup: ....  Init SD card
00:00:01 setup: ....  SD card found, 3827.0 MB by 3861.0 MB free
[  1889][I][esp32-hal-ledc.c:131] ledcAttachChannel(): LEDC attached to pin 10 (channel 0, resolution 8)
00:00:02 setup: ....  seek for stations.csv
00:00:02 setup: ....  stations.csv found
00:00:02 setup: ....  seek for WiFi networks
[  2926][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: darmstadt
```

clear-sky, 2024-04-25:
Hi dear wolle!
as you know I use a bare esp-s3-12k module 
i solved the problem but part of it!
soldered a 2.2uf + 100nf between EN pin and ground 
now reset has been solved but i give unclear characters in serial monitor please see:
>> pio device monitor     

```
Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Terminal on /dev/ttyUSB0 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
=> 0x403c9700: ?? ??:0
load:0x403c9704,len:0xad0
=> 0x403c9704: ?? ??:0
load:0x403cc700,len:0x29d8
=> 0x403cc700: ?? ??:0
entry 0x403c9880
=> 0x403c9880: ?? ??:0
[  1281][I][esp32-hal-psram.c:92] psramInit(): PSRAM enabled

ESP32 Chip: ESP32-S3
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.01f  Apr 18/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 20 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Reset due to power-on event

00:00:00    
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00    
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ST7796
00:00:01 setup: ....  Init SD card
00:00:01 setup: ....  SD card found, 3827.0 MB by 3861.0 MB free
[  1882][I][esp32-hal-ledc.c:131] ledcAttachChannel(): LEDC attached to pin 10 (channel 0, resolution 8)
00:00:02 setup: ....  seek for stations.csv
00:00:02 setup: ....  stations.csv found
00:00:02 setup: ....  seek for WiFi networks
[  2919][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: darmstadt
[ 10002][I][WiFiMulti.cpp:134] run(): [WIFI] scan done
[ 10008][I][WiFiMulti.cpp:139] run(): [WIFI] 8 networks found
[ 10015][I][WiFiMulti.cpp:263] run(): [WIFI] Connecting BSSID: 02:10:18:01:00:02 SSID: darmstadt Channel: 1 (-31)
[ 10026][W][STA.cpp:548] disconnect(): STA already disconnected.
[ 10120][I][WiFiMulti.cpp:283] run(): [WIFI] Connecting done.
00:00:09 WiFI_info:   Connecting WiFi...
00:00:09 WiFI_info:   mDNS name: MiniWebRadio
00:00:09 setup: ....  connected to darmstadt, IP address is 192.168.1.3
00:00:09 ftpServer:   Buffers allocated: 4622 bytes
00:00:09 rtime_info:  Initializing SNTP
03:30:09 rtime_info:  Waiting for system time to be set... (1/10)
03:30:11 rtime_info:  Waiting for system time to be set... (2/10)
03:30:13 rtime_info:  Waiting for system time to be set... (3/10)
23:17:47��W�}�nf� 0�3��7�MM���5)���17��_�^v�хϭ�����)��ngf toجrg �[��+���$��׿��)�����Ͻ铁�������is6�
��&��Ң݁͵�]\f��������ɕ�Ձstcv}on �ber� ��6�4��5-2��Γ�ׁ�]�?�vr�������W�t"vol����u������
  ��������^��ѥ���ݗ"��76�700�[�S�ms�[�mc!������Ձ����遹�������˿ctkn"timuowt Ӻ��[36m�^L�����
         �kmۘ�C��n���א�,���:�.>.  cu���t�v��][{�z?ڴ������5)��f���4]ȧ%�n��ɑ�R�����.Ҵ7���30���7���+Ӛ�j;�6Qӂ�[���l����.>�����Hֿ���������͖�W��
�:ڗ�i�e��atio.lשּׁk������5�2�����فX�w��["rvs��ݥ�c�"vo*sv,�o~ �[��[���[JJ����فժR굹��>�����jC�nngst"��nuw h��� �h�����빥��ɹ]]I�����com:840ӵ��o"
J��ʻҢ݉�U�S���f�������K*�[�ە됻�&&Ӵj_▒23:1���Size�ț��Ձ�y�]jKH�����ћ���%=����遁�ff�_��ꕹvo������S�}��Xv���K��t:"UDvw&�����
2�:3�:v7(AWTIO_y�녁m[���PR(�Ջ���=��-4"nowõp~gzit��+����r.����݃M��YX[�ձY��m�͉����ɥsnskr�s$�����(W�k+��*6vKj���ݭI]��m}��������ҫ͔����5�rvgr?�&�gdi�!�K�er�f�u~���}Y��ջ��^�׫����.���74��+��Hțښ���> S�X�o���ٯ����[*K�kӧ߃JuoY��I�e�#JkGH��݊*���=W9�j��L際ZZ��url>�r������hvvrs�//�ן�+je>�kj����
                                     �a�ڻ�����ӵ5)���1w:>6 A��Y__info�" �o�P(�k��롡�n�֫�
      )�+�����镓,0fzee �^p:*1���]&���Y�$        �K�����X֛���́��Z�om�m]
:3:5�����ast�ŵKs��m��w����\�멭ݹ��ѕۻ�ѽɅ����coo:8��Ӊ+����[C�.����م����Kٕ�鐙��������ח�w�_��fﹰ���mc�Z�+�˓ѽ�сm}ٽ��m��m/
���\ҫ��U5-=_鷳K����ɽ����X�$�V6V-*6&������
�T���nf[��[ۓ�{罍�ۓ���ջ���щ��́132;[1��%��ʺڢ��A�D٧�i��o�}m��[W����ɑ��չ���t��͑�=
2�:����
_��ꕹ��끑m�S�����׋����ɽr0/3 :$��VSLID_AVTS�H�A�ݵ�k5��/�ӻ�ݡ�]��ݹ��邁�[32osy��K�*�o��YXz]خk����ѻJj������݁]YZk�۫�:r";[33�SS������Ң݁A]LR�չ��鐑���j����ݽɑ��չсa���k��C��������!P�S�ꥹ���*�Czcnn��./$��+S��������݁�U�R����ˉ�mm�ɵU�
                                                      ��K�_�����0��+��l$�����7�Q�SRꥹ��遑��ɽ    ���C��]��������
```

clear-sky, 2024-04-25:
As soon as the system succeeds to get the time, the terminal crashes!!!

schreibfaul1, 2024-04-25:
I can't say exactly what this could be. It probably has nothing to do with the speed of the terminal. It could be a corrupted "settings.json". You can delete this file from the SD card and it will be recreated.

clear-sky, 2024-04-25:
thanks for the answer- your hint did not work ! problem is another thing

clear-sky, 2024-04-26:
Hi dear wolle!
thanks for the answer
yes it always happens in the same place!!! even i open device monitor after boot will give a crashed terminal
i changed time zone to berlin and newyork and toronto via web interface but problem remains

closed 2024-06-10 by github-actions[bot]