schreibfaul1/ESP32-MiniWebRadio #403: GARAMOND font problem

clear-sky, 2024-04-26:
Hi dear wolle!
some characters cant be shown correct in garamond font .
my ip is 192.168.1.4 (character 4 cont be shown correctly in ip but in station number and clock shown correctly )another problem you can see in program info(waer):
[image]

schreibfaul1, 2024-04-26:
I was able to reproduce this. The bug will be fixed with the next version. Then the equalizer will also appear on the display and a hard copy can be shown on the website. It will then no longer be necessary to take a photo of the display.

clear-sky, 2024-04-26:
_Thanks for your notice with respect_

closed 2024-04-26 by clear-sky

clear-sky, 2024-04-27:
Hi Dear Wolle!
sorry for posting this image but i thought it is important
[image]