schreibfaul1/ESP32-MiniWebRadio #405: Sleep Mode

valera39, 2024-04-26:
Hi Dear Wolle!
When you put the radio into sleep mode using the button on the remote control and then exiting it by pressing the same button, the radio reboots with this image
[image]
port monitor:

`Esp32ExceptionDecoder:` disabling, firmware at C:\Arduino\ESP32-MiniWebRadio\.pio\build\esp32s3\firmware.elf does not exist, rebuild the project?  
--- Terminal on COM5 | 512000 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
����x���x�x�x�xx��xxx�x��xxxxxx�x�x�xxxxx��xx�x�xxxxxxx��xx�x��xxx�x�xxx�xx�xx�xxx���xx�x�

ESP32 Chip: ESP32-S3
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.01f  Apr 18/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 3761.6 MB by 3781.5 MB free
[   856][I][esp32-hal-ledc.c:131] ledcAttachChannel(): LEDC attached to pin 10 (channel 0, resolution 8)
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  2002][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  7878][I][WiFiMulti.cpp:134] run(): [WIFI] scan done
[  7880][I][WiFiMulti.cpp:139] run(): [WIFI] 6 networks found
[  7881][I][WiFiMulti.cpp:263] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 1 (-54)
[  7884][W][STA.cpp:548] disconnect(): STA already disconnected.
[  7975][I][WiFiMulti.cpp:283] run(): [WIFI] Connecting done.
00:00:07 WiFI_info:   Connecting WiFi...
00:00:07 WiFI_info:   mDNS name: MiniWebRadio
00:00:07 setup: ....  connected to ASUS_AC66U, IP address is 192.168.1.130
00:00:07 ftpServer:   Buffers allocated: 4622 bytes
00:00:07 rtime_info:  Initializing SNTP
02:00:07 rtime_info:  Waiting for system time to be set... (1/10)
23:51:51 rtime_info:  23:51:51
23:51:51 audiotask:   is pinned to core 0
23:51:51 audiotask:   priority is 2
23:51:51 setup: ....  Number of saved stations: 92
23:51:51 setup: ....  current station number: 41
23:51:51 setup: ....  current volume: 3
23:51:51 setup: ....  last connected host: https://radio.sweetmelodies.tk:8000/stream1
23:51:51 setup: ....  connection timeout: 500 ms
23:51:51 setup: ....  connection timeout SSL: 2000 ms
23:51:51 action: ...  current volume is 3
23:51:51 WiFI_info:   RSSI is -55 dB
[ 10919][I][ArduinoOTA.cpp:141] begin(): OTA server at: MiniWebRadio.local:3232
23:51:52 action: ...  switch to station 41
23:51:52 AUDIO_info:  Connect to new host: "https://radio.sweetmelodies.tk:8000/stream1"
23:51:52 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
23:51:52 AUDIO_info:  buffers freed, free Heap: 178160 bytes
23:51:52 AUDIO_info:  SSL has been established in 384 ms, free Heap: 140040 bytes
23:51:52 StationName: Sweet Melodies
23:51:52 bitRate:     128Kbit/s
23:51:52 icy-url: ..  https://sweetmelodies.tk/
23:51:52 AUDIO_info:  AACDecoder has been initialized, free Heap: 140100 bytes , free stack 3228 DWORDs
23:51:52 lastURL: ..  https://radio.sweetmelodies.tk:8000/stream1
23:51:52 AUDIO_info:  stream ready
23:51:52 AUDIO_info:  syncword found at pos 0
23:51:52 AUDIO_info:  Channels: 2
23:51:52 AUDIO_info:  SampleRate: 44100
23:51:52 AUDIO_info:  BitsPerSample: 16
23:51:52 AUDIO_info:  BitRate: 160207
23:51:52 AUDIO_info:  AAC HeaderFormat: ADTS
23:51:52 AUDIO_info:  AAC Codec: MPEG-4 LowComplexity
23:51:53 StreamTitle: Mantovani - I Write The Songs
23:51:53 Audiocodec:  AAC
23:51:53 ftpServer:   Ftp server waiting for connection on port 21
23:51:54 BT-Emitter:  KCX_BT_Emitter not found 
23:51:59 DLNA_server: [0] 192.168.1.1:8200  RT-AC66U_B1-44F0
23:52:00 DLNA_server: 1 media server found
23:52:02 ir_code: ..  IR address 0x00, IR command 0x16
[ 23274][I][main.cpp:3224] ir_long_key(): long pressed key nr: 10
[ 23421][W][STA.cpp:157] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
23:52:05 falling asleep
23:52:37 ir_code: ..  IR address 0x00, IR command 0x16
23:52:37 awake
[ 56172][W][STA.cpp:548] disconnect(): STA already disconnected.
[ 57921][I][WiFiMulti.cpp:134] run(): [WIFI] scan done
[ 57922][I][WiFiMulti.cpp:139] run(): [WIFI] 4 networks found
[ 57924][I][WiFiMulti.cpp:263] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 1 (-56)
[ 57927][W][STA.cpp:548] disconnect(): STA already disconnected.
[ 58008][I][WiFiMulti.cpp:283] run(): [WIFI] Connecting done.
START WIFI
23:52:39 WiFi connected
23:52:39 AUDIO_info:  Connect to new host: "https://radio.sweetmelodies.tk:8000/stream1"
Guru Meditation Error: Core  0 panic'ed (InstructionFetchError). Exception was unhandled.

Core  0 register dump:
PC      : 0x3c247d74  PS      : 0x00060830  A0      : 0x820a711c  A1      : 0x3fcbe320
A2      : 0x3fcba6a8  A3      : 0x3c2fcaa0  A4      : 0x00000000  A5      : 0x00060a23
A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x8210c62c  A9      : 0x3fcbe2e0
A10     : 0x00000010  A11     : 0x3c2fcaa0  A12     : 0x3fceecc0  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x00000002
EXCVADDR: 0x3c247d74  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace: 0x3c247d71:0x3fcbe320 |<-CORRUPTED

ELF file SHA256: 11d0c1f7068867e1

Rebooting...
xxxxxxx�x�x�xxx�xxx���xxxx��x�xxx��xx�x�xxxxxxx�x�xxx�xxxxxx�xxxxxx�x��x�xx�xxxxxxxxxxxxxxxxxxxxxx�xx�xxx�x�������x�xxxxxx�xxxxx�x�x�xxx�xxx��x�xxxx�x�x�x�x�xx�xxxxxx�xxx�x�xxx�x�xx����x��x�xxx�x��x�x�xx����x���x�xxx�x�xx�x�xx����xx�x�xxx�x�xxxxx��xx���xx����x���x�x�x�x�xx��xxx�x��xxxxxx�x�x�xxxxx��xx�x�xxxxxxx��xx�x��xxx�x�xxx�xx�xx�xxx���xx�x�

ESP32 Chip: ESP32-S3
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.01f  Apr 18/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8388608 bytes
RESET_REASON: Software reset due to exception/panic

21:52:39
21:52:39        ***************************    
21:52:39        *     MiniWebRadio V3     *    
21:52:39        ***************************    
21:52:39
21:52:39 setup: ....  Arduino is pinned to core 1
21:52:39 tft_info: .  init ILI9488
21:52:40 setup: ....  Init SD card
21:52:40 setup: ....  SD card found, 3761.6 MB by 3781.5 MB free
[   863][I][esp32-hal-ledc.c:131] ledcAttachChannel(): LEDC attached to pin 10 (channel 0, resolution 8)
21:52:41 setup: ....  seek for stations.csv
21:52:41 setup: ....  stations.csv found
21:52:41 setup: ....  seek for WiFi networks
[  2009][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  7486][I][WiFiMulti.cpp:134] run(): [WIFI] scan done
[  7488][I][WiFiMulti.cpp:139] run(): [WIFI] 6 networks found
[  7490][I][WiFiMulti.cpp:263] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 1 (-52)
[  7492][W][STA.cpp:548] disconnect(): STA already disconnected.
[  7573][I][WiFiMulti.cpp:283] run(): [WIFI] Connecting done.
21:52:46 WiFI_info:   Connecting WiFi...
21:52:46 WiFI_info:   mDNS name: MiniWebRadio
21:52:46 setup: ....  connected to ASUS_AC66U, IP address is 192.168.1.130
21:52:46 ftpServer:   Buffers allocated: 4622 bytes
21:52:46 rtime_info:  Initializing SNTP
23:52:46 rtime_info:  23:52:46
23:52:46 audiotask:   is pinned to core 0
23:52:46 audiotask:   priority is 2
23:52:46 setup: ....  Number of saved stations: 92
23:52:46 setup: ....  current station number: 41
23:52:46 setup: ....  current volume: 3
23:52:46 setup: ....  last connected host: https://radio.sweetmelodies.tk:8000/stream1
23:52:46 setup: ....  connection timeout: 500 ms
23:52:46 setup: ....  connection timeout SSL: 2000 ms
23:52:47 setup: ....  volume is muted: (from 3)
23:52:47 WiFI_info:   RSSI is -54 dB
[  8519][I][ArduinoOTA.cpp:141] begin(): OTA server at: MiniWebRadio.local:3232
23:52:47 RESET_REASON:Software reset due to exception/panic
23:52:48 ftpServer:   Ftp server waiting for connection on port 21
23:52:49 BT-Emitter:  KCX_BT_Emitter not found 
23:52:55 DLNA_server: [0] 192.168.1.1:8200  RT-AC66U_B1-44F0
23:52:55 DLNA_server: 1 media server found
23:53:05 WiFI_info:   RSSI is -50 `dB`

Re-flashed with a full cleanup and re-downloaded the project again. 
Thanks for your hard work and hope you will add FM radio soon. I think [clear-sky](https://github.com/clear-sky) will be happy too :)

clear-sky, 2024-04-27:
Hi Dear valera39!
long press is for hardware buttons and using it with remote is rediculus !
due to ir remote #402 problem when you long press buttons in most cases it drops in mute mode before !
i had asked from wolle to add ir buttons to avoid this problems #245 #309 #359
hope he pay attention to your ask!

valera39, 2024-04-27:
Hi Dear [clear-sky](https://github.com/clear-sky)!
I still have the radio on the second version and there with a button on the remote it goes to sleep and wakes up with no problem. But in the third version it doesn't do that

valera39, 2024-04-27:
[clear-sky](https://github.com/clear-sky), I think a remote like this would be perfect for the radio. 
[image]
[https://aliexpress.ru/item/4000876148513.html?spm=a2g2w.home.10009201.5.75df5586PkmkDb&mixer_rcmd_bucket_id=aerabtestalgoRecommendAbV21_testRankingEnrichTarget3d&ru_algo_pv_id=7de200-65a512-090255-9725ba-1714244400&scenario=aerAppJustForYouNew&sku_id=12000037978151683&traffic_source=recommendation&type_rcmd=core](url)
Let's ask the esteemed Wolle to do so

clear-sky, 2024-04-28:
Hi Dear [valera39](https://github.com/valera39) !
Choosing the type of remote completely depends on everyone's taste!
These types of remotes have several problems that are unbearable for me. First, they have hard buttons
  Second, due to the use of a coin battery that has little power and therefore its range is very low. Third, if it is used a lot, the battery will be empty very quickly.
Larger remotes that work with AA or AAA batteries do not have the problems of the previous remotes. I use a special type of them that is raw and does not have a special code in them. You can save the codes of several remotes in this remote and use one remote instead of all. these remotes is called  "universal" or "learning"
[image]

clear-sky, 2024-04-28:
The next news is that the initial esp32-p4 boards were introduced and are being tested by early developers. The processor is dual-core and works at a speed of 400 MHz and can easily display full HD videos at a speed of 30 frames per second. Maybe Mr. Wolle, seeing this board, will leave the development of radio and start a new project called MWTV---who knows!!!
[image]

valera39, 2024-05-06:
Hi Dear Wolle!
I never heard your answer to my question. Can this problem be solved with a reboot?

schreibfaul1, 2024-05-06:
There will be a new version in which this and other errors are eliminated. If I change something in the master now, I have problems transferring the dev branch via PR. Please be patient.
The lowering of the CPU frequency and the disconnection of the WiFi has not proved successful and will no longer exist in "SLEEP mode".

valera39, 2024-05-06:
Thanks, I just thought you hadn't noticed the error.

valera39, 2024-05-24:
Hi Dear Wolle!
Is it possible to disable KCX_BT_Emitter activity in sleep mode?

ESP32 Chip: ESP32-S3
Arduino Version: 2.0.16
ESP-IDF Version: 4.4.7
MiniWebRadio     Version 3.1   May 21/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8381431 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 3761.4 MB by 3781.5 MB free
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  2345][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  2367][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Honor
00:00:03 setup: ....  5 WiFi networks found
00:00:03 setup: ....  ASUS_AC66U (-57)
00:00:03 setup: ....  TP-LINK_E345E0_RPT (-74)
00:00:03 setup: ....  TP-LINK_E345E0 (-75)
00:00:03 setup: ....  ASUS_AC66U_RPT (-76)
00:00:03 setup: ....  GG (-82)
[ 10080][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[ 10086][E][WiFiMulti.cpp:117] run(): [WIFI] no networks found
[ 10093][E][WiFiMulti.cpp:191] run(): [WIFI] no matching wifi found!
00:00:09 WiFI_info:   WiFi credentials are not correct
00:00:09 RECONNECTION try 1
[ 10100][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[ 10123][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Honor
00:00:11 setup: ....  5 WiFi networks found
00:00:11 setup: ....  ASUS_AC66U (-59)
00:00:11 setup: ....  TP-LINK_E345E0_RPT (-71)
00:00:11 setup: ....  TP-LINK_E345E0 (-75)
00:00:11 setup: ....  ASUS_AC66U_RPT (-76)
00:00:11 setup: ....  GG (-81)
[ 18380][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[ 18386][I][WiFiMulti.cpp:119] run(): [WIFI] 4 networks found
[ 18392][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 9 (-59)
[ 18473][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:17 WiFI_info:   Connecting WiFi...
00:00:17 WiFI_info:   mDNS name: MiniWebRadio
00:00:17 setup: ....  connected to ASUS_AC66U, IP address is 192.168.1.130
[ 18514][I][ArduinoOTA.cpp:141] begin(): OTA server at: MiniWebRadio.local:3232
00:00:17 ftpServer:   Buffers allocated: 4622 bytes
00:00:17 rtime_info:  Initializing SNTP
02:00:17 rtime_info:  Waiting for system time to be set... (1/10)
13:46:51 rtime_info:  13:46:51
13:46:51 audiotask:   is pinned to core 0
13:46:51 audiotask:   priority is 2
13:46:51 AUDIO_info:  commFMT = 1
13:46:51 setup: ....  Number of saved stations: 92
13:46:51 setup: ....  current station number: 42
13:46:51 setup: ....  current volume: 2
13:46:51 setup: ....  last connected host: http://163.172.77.142:8006/;stream/1
13:46:51 setup: ....  connection timeout: 500 ms
13:46:51 setup: ....  connection timeout SSL: 2000 ms
13:46:52 action: ...  switch to station 42
13:46:52 AUDIO_info:  Connect to new host: "http://163.172.77.142:8006/;stream/1"
13:46:52 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
13:46:52 AUDIO_info:  buffers freed, free Heap: 166804 bytes
13:46:52 AUDIO_info:  Connection has been established in 42 ms, free Heap: 165824 bytes
13:46:53 StationName: Ā«majorasĀ»
13:46:53 bitRate:     96Kbit/s
13:46:53 icy-url: ..  https://xocotakamu.jimdo.com/
13:46:53 AUDIO_info:  MP3Decoder has been initialized, free Heap: 165128 bytes , free stack 4996 DWORDs
13:46:53 lastURL: ..  http://163.172.77.142:8006/;stream/1
13:46:53 WiFI_info:   RSSI is -59 dB
13:46:53 Audiocodec:  MP3
13:46:53 ftpServer:   Ftp server waiting for connection on port 21
13:46:53 BT-Emitter:  KCX_BT_Emitter found 
13:46:53 AUDIO_info:  stream ready
13:46:53 AUDIO_info:  syncword found at pos 0
13:46:53 StreamTitle: Cicci Guitar Condor - Theme-from-shane
13:46:53 AUDIO_info:  Channels: 2
13:46:53 AUDIO_info:  SampleRate: 44100
13:46:53 AUDIO_info:  BitsPerSample: 16
13:46:53 AUDIO_info:  BitRate: 96000
13:46:53 BT-Emitter:  Version KCX_BT_RTX_V1.4
13:46:53 BT-Emitter:  scanned: MacAdd:5f673c5cfaec,Name:MIBOX4
13:46:53 BT-Emitter:  CON MATCH ADD
13:46:53 BT-Emitter:  CONNECT=>MacAdd:5f673c5cfaec,Name:MIBOX4 
13:46:53 BT-Emitter:  Volume -> 31
13:46:54 AUDIO_info:  slow stream, dropouts are possible
13:46:55 AUDIO_info:  slow stream, dropouts are possible
13:46:59 DLNA_server: [0] 192.168.1.1:8200  RT-AC66U_B1-44F0
[ 28547][E][DLNAClient.cpp:226] readHttpHeader(): content type not found
13:46:59 DLNA_server: [1] 192.168.1.4:6000  dlink-E247EB
13:46:59 DLNA_server: 2 media server found
13:47:07 BT-Emitter:  Status -> Scan...
13:47:08 ir_code: ..  IR address 0x00, IR command 0x16
13:47:08 BT-Emitter:  CON MATCH ADD 
13:47:08 BT-Emitter:  CONNECT=>MacAdd:5f673c5cfaec,Name:MIBOX4
[ 39472][I][main.cpp:2847] ir_long_key(): long pressed key nr: 10
13:47:10 falling asleep
13:47:23 BT-Emitter:  CON MATCH ADD 
13:47:23 BT-Emitter:  CONNECT=>MacAdd:5f673c5cfaec,Name:MIBOX4
13:47:39 BT-Emitter:  CON MATCH ADD 
13:47:39 BT-Emitter:  CONNECT=>MacAdd:5f673c5cfaec,Name:MIBOX4
13:47:54 BT-Emitter:  CON MATCH ADD 
13:47:54 BT-Emitter:  CONNECT=>MacAdd:5f673c5cfaec,Name:MIBOX4
13:48:10 BT-Emitter:  CON MATCH ADD 
13:48:10 BT-Emitter:  CONNECT=>MacAdd:5f673c5cfaec,Name:MIBOX4

schreibfaul1, 2024-05-25:
This is a good idea. There is an AT command which switches off the BT emitter "AT-POWER_OFF". But there is none to switch it on. This can be done with the button on the board or with a short "H" pulse at the Connect input.
You need another GPIO for this. I have realised this for the ESP32-S3 and adapted the schematic. The ESP32 lacks free pins.

ve2cfb, 2024-05-26:
Hello Wolle,
I did some test using the AT-POWER_OFF and CONNECT with the basic  KCX_BT_Emitter code that is on your site and ir was promissing... Keep reading.
i Really need this option because been in a condo, the rooms are quite close and every device wants to connect to everybody. So i need to POWER OFF the BT only.  Like you said, the esp32 is missing an extra pin. I had to remove the IR option for the test.
On my Web Radio there is only 2 buttons, the Power On/Off and the BT On/Off. Both are latched buttons with Led. At one point, i had to let go since i had problems with the code. (who ever invented hold age, should be sent to Jupiter...)

Anyway.¸..
Is the AT-POWER_OFF and CONNECT code usable with the ESP32-S3 ...?

Thanks again.

schreibfaul1, 2024-05-26:
With the ESP32, GPIO12 would be free, this is a "strapping pin" and must be pulled to GND at startup. GPIO12 is therefore hardly applicable (possibly for the display backlight?) The pin for IR is "input only". 
A short low pulse at CONNECT is required to wake up the BT emitter (~100ms)

ve2cfb, 2024-05-26:
Yes , now i remember why during my test, i stopped working with pin 35. 
Did a quick look in the code and found the CONNECT section. Will see if it can do what i want.
Thanks again.

schreibfaul1, 2024-05-26:
Perhaps GPIO12 can be used for TFT_BL. I haven't tried it, but then GPIO32 would be free for BT-CONNECT.

ve2cfb, 2024-05-26:
Hello Wolle,
Since almost all the BT fnctions are already done, i would do the test with a ESP32S3 and later try to implement it on a ESP32.
You can tell me if it is possible.

Use a SPST Latched Switch.

**Here is my idea of sequence of operation using 2 GPIO.**
1. (Switch is in OFF position)  do a AT-POWER_OFF . At startup time or during normal operation. 
2. (Switch is in ON position)  do a BT connect. At startup time or during normal operation. 
3. (Switch is in OFF position)  do AT-POWER_OFF again. During normal operation.

Many thanks...

valera39, 2024-05-26:
Hi Dear Wolle!
I would like to propose you an idea, to show the clock in sleep mode, with the ability to disable this option in the settings. What do you think about it?

schreibfaul1, 2024-05-26:
The idea is good

valera39, 2024-06-12:
Hi Dear Wolle!
In order not to open a new topic, I'll ask here.
How to make the clock brightness in sleep mode less?

schreibfaul1, 2024-06-13:
This requires another setting option. Another idea would be to use a photodiode to control the brightness depending on the ambient light. But unfortunately there is no free pin on the ESP32.

valera39, 2024-06-13:
Great idea with the photodiode, it can be used for esp32s3 which has free pins. Shall we try it?

schreibfaul1, 2024-06-15:
A photodiode is probably not the right solution after all. There are too many with different properties or characteristics. I think a brightness sensor would be better, preferably with I2C. I will first see what is available on the big online platforms, the sensor must be available everywhere at an affordable price.

valera39, 2024-06-15:
For example, you can use the available GY-302 BH1750 Light Sensor

clear-sky, 2024-06-15:
Hi Dear valera39!
simple LDR works fine for me in wolle's matrix clock!

valera39, 2024-06-15:
[clear-sky](https://github.com/clear-sky)Hey!
Exactly, I would like the brightness to be dimmed in sleep mode.

closed 2024-08-01 by github-actions[bot]