schreibfaul1/ESP32-MiniWebRadio #406: Problem with slow playback and Polish characters.

hevet, 2024-05-04:
In Version https://github.com/schreibfaul1/ESP32-audioI2S/commit/e2e4df30965e05c0ec18a68a1c7e9d2fc2728bf6 some stations are slower again.

http://ic1.smcdn.pl/2030-1.mp3
http://ic1.smcdn.pl/6120-1.mp3

From this one https://github.com/schreibfaul1/ESP32-audioI2S/commit/8d50fd3b897e050a0539e17dae9350f41f812e28  stations do not display Polish characters correctly.

schreibfaul1, 2024-05-04:
Yesterday I changed the conversion from latin-1 (iso8859) to UTF-8, which affects these letters:
### ÁÂÃÄÅÆÇÈßÉÊËÌÍÎÏÐÑÒÓÔÕÖ×ØÙÚÛÜÝÞßàáâãääæçèéêëìíîïðñòóôõö÷øùúûüýþÿ

Are there any other letters? How can I reproduce this?

hevet, 2024-05-04:
Sample station https://ic2.smcdn.pl/3990-1.mp3

hevet, 2024-05-04:
Polish characters are Ą Ć Ę Ł Ń Ś Ó Ź Ż ą ć ę  ł ń ś ó ź ż

schreibfaul1, 2024-05-04:
Thank you for your support. For metadata in Europe, the stream titles are sometimes still transmitted in Latin-1. UTF-8 is predominantly used. It is difficult to distinguish between the two encodings. I search for invalid UTF-8 characters in the stream. Hope that's okay now. At the sample station I now get:
  
[image]

hevet, 2024-05-04:
Ok, thanks, I'll check later when I get home.  What about the second problem, e.g. this station "Vox FM" plays at a slower tempo.

schreibfaul1, 2024-05-04:
That's strange, VOX FM plays at 44.1KHz and in the VLC player at 48KHz, the same URL https://ic2.smcdn.pl/3990-1.mp3

hevet, 2024-05-04:
@schreibfaul1 There is the same problem with Eska, Eska Rock, you already improved something because there was the same problem.
https://github.com/schreibfaul1/ESP32-MiniWebRadio/issues/383

schreibfaul1, 2024-05-04:
Thanks, that was a self-installed error. I have suppressed the error messages of the mp3 decoder if the mp3 frames are not sufficient for decoding. However, the sample rate 0 is output until the first successful decoding. This is ignored and the standard 44100Hz is set. Of course, the correct sample rate can only be determined when valid data is available for I2S.

hevet, 2024-05-04:
@schreibfaul1 thanks, is ok :)

closed 2024-06-18 by github-actions[bot]