schreibfaul1/ESP32-MiniWebRadio #407: Station stops playing after Time announcement on the hour

TubaFonse, 2024-05-05:
Hallo Wolle, 
nachdem die "Zeitansage zur vollen Stunde" erfolgt ist, spielt das Radio in der aktuellen Version nicht weiter. 
Auf dem Display wird nur der aktuelle Sender und der zuletzt gespielte Titel angezeigt, das VU-Meter zeigt keine Aktivität. 

Serial Monitor: 
13:59:49 StreamTitle: LUDWIG HIRSCH - ZART
13:59:53 Time: ...... play Audiofile /voice_time/14_00.mp3
13:59:53 AUDIO_info:  buffers freed, free Heap: 165320 bytes
13:59:53 AUDIO_info:  Reading file: "/voice_time/14_00.mp3"
13:59:53 AUDIO_info:  MP3Decoder has been initialized, free Heap: 160912 bytes , free stack 2156 DWORDs
13:59:54 AUDIO_info:  Content-Length: 143579
13:59:54 AUDIO_info:  ID3 framesSize: 55
13:59:54 AUDIO_info:  ID3 version: 2.4
13:59:54 AUDIO_info:  ID3 extended header
13:59:54 Audiocodec:  MP3
13:59:54 AUDIO_info:  Audio-Length: 143524
13:59:54 AUDIO_info:  stream ready
13:59:54 AUDIO_info:  syncword found at pos 0
13:59:54 AUDIO_info:  Channels: 2
13:59:54 AUDIO_info:  SampleRate: 44100
13:59:54 AUDIO_info:  BitsPerSample: 16
13:59:54 AUDIO_info:  BitRate: 128000
14:00:00 AUDIO_FILE:  AudioCurrentTime 0:06s, AudioFileDuration 0:08s  
14:00:01 AUDIO_info:  Closing audio file "14_00.mp3"
14:00:01 end of file: 14_00.mp3
14:00:01 AUDIO_info:  End of file "14_00.mp3"

Du findest bestimmt schnell die Ursache ...
Schönen Sonntag & bayrische Grüße nach Berlin, 
Alfons.

TubaFonse, 2024-05-05:
Hallo Wolle - habe Deine Korrektur soeben gesehen und gleich ausprobiert, es funktioniert wieder! 
Musste nur noch die volle Stunde abwarten und danke für die rasante "Lieferung" des Updates! 
Gruß, Alfons.

closed 2024-05-05 by TubaFonse

schreibfaul1, 2024-05-05:
Hallo Alfons,
bin gerade dabei, die Bilder, Grafiken etc. alles was auf den Display abgebildet wird als Objekte zu behandeln. Das bringt zunächst nicht viel Neues, aber der Quellkode ist einfacher zu warten. Dann werden für mich künftige Erweiterungen einfacher. 
Einige Mängel habe ich inzwischen beseitigt, auch das was hier als "Issues" angesprochen wurde. Aber das dauert noch, vermutlich im Juni wird der "Dev" Branch die Version 3.1 werden. Mal sehen ob bis dahin Espressif/PlatformIO in der Lage sind ein fehlerfreie "Platform" anzubieten, damit wieder Verbesserungen bezüglich der WiFi Performance möglich sind.
Bis dahin noch etwas Geduld,
beste Grüße aus Berlin
Wolle