schreibfaul1/ESP32-MiniWebRadio #408: How to connect ESP32-S3-Wroom-1 with N16R8

tzvetan65, 2024-05-06:
Hello everyone,
   I decided to replace the ESP32 D32 with the ESP32-S3-Wroom-1 with N16R8, but the resulting devboard has a different pineout.
[image]
[image]
[image]
 How should it be connected?

clear-sky, 2024-05-06:
[image]

tzvetan65, 2024-05-10:
Hello, can no one help? I cannot understand how to connect this ESP with DAC and the display. Which conclusion to who is in line with. Please help, I want this radio!

Arne65, 2024-05-10:
[image]

tzvetan65, 2024-05-10:
Hello,
Something weird, I don't see the answers under my question. When I write a new comment, I see an answer from before, I delete the comment as an out of date and the answer disappears. I apologize to the answers, I just don't see your answers, I don't know if the problem is on the site, it happens to me for the first time.

Jimbobnz, 2024-05-15:
I'm I needing to update the code to reflect the image above? 
//ESP32-S3
#define I2S_DOUT      9     /*  DIN connection */
#define I2S_BCLK      3     /*  BCLK Bit clock */
#define I2S_LRC       1     /*  WSEL Left Right Clock */

tzvetan65, 2024-05-15:
I'm sorry, but I can't help. I'm not doing well with the software when I have a problem and I ask here. I hope someone will answer, greetings and success.

schreibfaul1, 2024-05-15:
The pins in the **CONFIG_IDF_TARGET_ESP32S3** section in "config.h" are only suggestions. You have much more freedom in the choice of pins than with the ESP32. What exactly is causing you difficulties?

tzvetan65, 2024-05-15:
Have I correctly understood the ARNE65 scheme, should the T_CS connection be removed with SCK and connect T_DO to SDO (MISO)? Do I have to connect the ESP32S3 display?
[image]

Arne65, 2024-05-15:
Hello
make the bridges as in the original schema.Bridge
SPI_SCK with SCK and of course SPI_MOSI with SOI(MOSI) as shown in the original schematic.

You can assign the pins like in common.h

This definitely works.

With the ESP 32-S3 you are more independent when it comes to pin assignment than with the ESP 32 S2 .

tzvetan65, 2024-05-15:
Thanks for the quick answers. Have I correctly realized that there should be three bridges
SDO(MISO) to T_DO
SDI(MOSI) to T_DIN
SCK to T_CLK

Arne65, 2024-05-15:
You  only need 

DI(MOSI) to T_DIN
SCK to T_CLK

What kind of display do you use?

tzvetan65, 2024-05-16:
My display is 3.2 TFT SPI 240x320 V1.0. Maybe I'm misleading this scheme. Thanks.
[image]

Arne65, 2024-05-16:
[image]

tzvetan65, 2024-05-16:
Thank you, it's so much more understandable.

tzvetan65, 2024-05-17:
What can be the reason for this restart?

Arne65, 2024-05-17:
Send the output to the serial monitor, maybe you can see more.
All pins connected properly and voltage is correct?

tzvetan65, 2024-05-17:
I'm sorry, I don't know how to do it. For the first time I work with Arduino and everything is new and unknown to me.

tzvetan65, 2024-05-19:
Work !!! The Arne65 was right, the problem is in the supply voltage. Strange, the power source and the multimeter show me 4.95 volts, and I have 4.45 volts on the board.
   I have a question about ESP32, the RGB led shines very bright, the PWR red led shines normally, and the TX green led 
[image]
barely visible. Is it normal. Or there is a problem with the module.
[image]

Arne65, 2024-05-19:
The RGB LED is very bright on the boards, so that's normal, but in my opinion it's nonsense - no one needs a "headlight" on the boards.

Normally you should measure around 3.3 V after regulation.

Input voltage 5V after the voltage regulator 3.3V.

tzvetan65, 2024-05-19:
Yes, at a voltage of 4.45 volts after the regulator I had 3.2 volts. I raised the supply voltage to 5.2 volts and now after the regulator I have 3.3 volts. There is now another problem - with an increase over 8 radio restarts, although the supply voltage does not fall and consumption is about 250 miliampers.

Arne65, 2024-05-19:
[image]

Start the serial monitor, maybe you can find out where your problem is

tzvetan65, 2024-05-19:
I turned on the serial monitor, in the command of the amplification or reduction of the sound, everything turns red. I do not understand what it means, but with a serial monitor, I strengthen it to 21 and do not restart. After the serial monitor is switched off, it restarts again with a reinforcement exceeding 8.
[image]
[image]

schreibfaul1, 2024-05-19:
I wonder if you are using an external DAC with I2C control, like ES8388. If you have a DAC without I2C, e.g. PCM5102, the pins I2C_DATA and I2C_CLK in common.h must be set to -1. If you do have an I2C DAC, then the wiring (SDA/SDC) is wrong.

Arne65, 2024-05-19:
Are you using an SD card adapter?
I think you have write protection turned on

tzvetan65, 2024-05-19:
My DAC is PCM5102, do I have to change values here? I tried a locked and unlocked card, the effect is the same. But I have a problem with the card, I upload new stations, but they can't play.

schreibfaul1, 2024-05-19:
If you do not have an audio board (AI Thinker , Lyra, Olimex), set [image] to 1 to deactivate I2C.

tzvetan65, 2024-05-19:
I changed the Decoder setting = 1 but my restart remained. Did I correctly set the type of ESP in Platformio.ini. Why is this getting this?
[image]
[image]

schreibfaul1, 2024-05-19:
You probably have an ESP32-S3, so please set this in the taskbar
[image]

 and then send a log of the restart

tzvetan65, 2024-05-19:
I changed the appearance of the board, but the restart remains. How to upload Log of the Restart? When the serial monitor is turned on, there is no restart, can it really be a power problem?
[image]

Arne65, 2024-05-21:
You have to press the reset button :)

tzvetan65, 2024-05-21:
I'm sorry for the panic, the reason for the restart was the power source. It clearly causes any interference to the ESP32. With another power supply works well. The next problem is communication with SD card, I add new stations, they are recorded on the card, but some of them radio does not choose them. It just doesn't change the station. I tried an unlocked and locked card, with and without a pullup resistor but no result. Any idea for this problem?

tzvetan65, 2024-05-24:
Give at least some joker. What is the difference in recording and reading from the SD card? Why does he record and not read the records? Or are the records incorrect and does not recognize them?
[image]

tzvetan65, 2024-06-04:
Come on, has no one had such a problem? Why are the stations recorded on the card and not read? Help please!

steve6375, 2024-06-04:
So if you press Save and then Load and then Save xlsx, are the stations are all in the saved xlsx filee?
if you load the xlsx file is the list is correct?
If you reset esp32, are the stations selectable now?
Can stations be selected and played on web GUI?
Can stations be selected and played on TFT screen GUI?
Give more details of the problem
If others have the problem it would have been reported

tzvetan65, 2024-06-05:
Hello,
The new stations are recorded in the xlsx files and in the SD card, but they are not selected and reproduced. In my opinion, the recordings are correct, but I do not understand enough. I also restarted the restart button and the power, but nothing changes. The stations cannot be selected and played through the TFT or GUI interface. Only some new stations can be selected and reproduced, but I don't understand what the reason is. I have surrounded the newcomers to the attached screenshots.
[image]
[image]
[image]

steve6375, 2024-06-05:
Maybe you can upload here your xlsx file and someone can try it?

tzvetan65, 2024-06-05:
Thanks Steve6375, your last comment has helped me understand what is happening. I tried to upload Stations.xlsx and did not find a file with this extension. It turns out OpenOffice saves the file with an extension (.ods) because, you see, it was a newer format. I recorded the file in the correct format and everything worked. Thanks!

closed 2024-07-20 by github-actions[bot]