schreibfaul1/ESP32-MiniWebRadio #411: Problems with version 3.1.

hevet, 2024-05-09:
I installed version 3.1 as a test. The errors noticed are:
- the code does not compile for the ili9341 display, there is an error m_bgColor, it is undefined. This is helped by inserting ";" before uint32_t m_bgColor = 0;
- lack of Polish characters, bad encoding, e.g. instead of ó there is probably ć, etc.
- when I switch to the screen with brightness or equalizer settings, the icons load slowly

schreibfaul1, 2024-05-09:
Thank you for your contribution. V3.1 is currently under development. I want to create an object for everything that is drawn on the display. I create classes such as Textbox, Slider, Button or Picturebox.... for this purpose.
The semicolon has been inserted.
In the Garamond character set, the characters for "Latin Extended A" were shifted by two characters in the lookup table, which I have corrected. I would never have noticed that without your hint. For testing I took the characters from the other issue
old:
[image]

corrected:
[image]

hevet, 2024-05-10:
Thanks, Polish characters are now correct.
I noticed another thing, I don't know if it's supposed to be like that, but the left arrow switches the stations ascending and the right arrow switches the stations descending.

closed 2024-05-23 by hevet