schreibfaul1/ESP32-MiniWebRadio #418: ESP32-S3 WROOM-1 | corrupted text on the serial monitor

Jimbobnz, 2024-05-15:
Trying this code for the first time and I'm getting this issue on ESP32-S3 WROOM-1 (SP32-S3-DevKitC-1-N16R8) with corrupted text.

Serial Monitor:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29d8
entry 0x403c9880
[   414][I][esp32-hal-psram.c:92] psramInit(): PSRAM enabled
�&R2�"6�!y0y6��4
                �9%�4�%�gd'd�QC�'d
                                  $&$� c�4'�4�t$!$a1$A�46�$dP0$!U4$da1d 6$74fE$$6Ș4�$�It��4��t�8�� h{�rf�tg�f$�rfV%1�$8
                                                                                                                       nF�y61�g�84v�"$t�&cUd�� 7�
```

clear-sky, 2024-05-23:
Hi!
please see #402

schreibfaul1, 2024-05-23:
The TTL2USB chip (the CP2102 is often used) does not actually cause any problems. You can change the monitor speed in platfornio.ini at the top. The USB cable may be too long or of poor quality. You can reduce the speed for tests, e.g. 57600 Baud

clear-sky, 2024-05-23:
Hi dear Wolle!
I have this problem also!
my problem exactly happens when radio gets time from web due to #402 
you suspected to `Timezone_String` but i suspected to another thing : `espota`
because espota is added to project from version 3
With great difficulty, I managed to go back to version 2, where this problem does not exist at all!!!

clear-sky, 2024-05-23:
Hi dear wolle!
i dont use vscode !
if possible please install platformio cli and test `pio device monitor`
you will see the problem

schreibfaul1, 2024-05-25:
That may be the case. OTA means that I do not have a direct connection via USB. You can achieve this by setting the port of e.g. Com or USB to "Auto". OTA is deactivated if you delete the upload port in platfornio.ini
[image]

clear-sky, 2024-05-25:
Hi Dear Wolle!
thanks for the answer-it is impossible to flash the mcu unless I disable upload port 
did you test platformio cli?

closed 2024-07-09 by github-actions[bot]