schreibfaul1/ESP32-MiniWebRadio #421: The Mini Webradio work but no sound

ddrram2, 2024-05-20:
Hi everyone
everything is ok, sd, there are stations, everything is great, but there is no sound. I got these yellow lines while compile, maybe this is the reason why there is no sound:

C:/Users/ESP32-S3/.platformio/packages/framework-arduinoespressif32/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::tx_desc_auto_clear' [-Wmissing-field-initializers]
   };
   ^
C:/Users/ESP32-S3/.platformio/packages/framework-arduinoespressif32/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::fixed_mclk' [-Wmissing-field-initializers]
C:/Users/ESP32-S3/.platformio/packages/framework-arduinoespressif32/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::mclk_multiple' [-Wmissing-field-initializers]
C:/Users/ESP32-S3/.platformio/packages/framework-arduinoespressif32/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::bits_per_chan' [-Wmissing-field-initializers]
C:/Users/ESP32-S3/.platformio/packages/framework-arduinoespressif32/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::chan_mask' [-Wmissing-field-initializers]
C:/Users/ESP32-S3/.platformio/packages/framework-arduinoespressif32/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::total_chan' [-Wmissing-field-initializers]
C:/Users/ESP32-S3/.platformio/packages/framework-arduinoespressif32/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::left_align' [-Wmissing-field-initializers]
C:/Users/ESP32-S3/.platformio/packages/framework-arduinoespressif32/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::big_edin' [-Wmissing-field-initializers]
C:/Users/ESP32-S3/.platformio/packages/framework-arduinoespressif32/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::bit_order_msb' [-Wmissing-field-initializers]
C:/Users/ESP32-S3/.platformio/packages/framework-arduinoespressif32/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::skip_msk' [-Wmissing-field-initializers]
C:/Users/ESP32-S3/.platformio/packages/framework-arduinoespressif32/libraries/I2S/src/I2S.cpp: In member function 'int I2SClass::_applyPinSetting()':
C:/Users/ESP32-S3/.platformio/packages/framework-arduinoespressif32/libraries/I2S/src/I2S.cpp:324:5: warning: missing initializer for member 'esp_i2s::i2s_pin_config_t::mck_io_num' [-Wmissing-field-initializers]
     };
     ^
Compiling .pio\build\esp32s3\RMakerNode.o

this made the compile good:
Successfully created esp32s3 image.
==================== [SUCCESS] Took 313.80 seconds ================

and what I haven't said yet, I use CJMCU1334 and UDA1334 I2S DAC
the connection
 #define I2S_DOUT 9
 #define I2S_BCLK 3
 #define I2S_LRC 1
 #define I2S_MCLK -1
how is it working for you?
There is S3 orig. narrow and clone, both work the same.
what should I do to make it good?

schreibfaul1, 2024-05-20:
You can ignore the warnings about uninitialised variables. If the DAC is a board from "Adafuit" it works in any case, I have one myself. 
I2S_DOUT => DIN 
I2S_BCLK => BCLK 
I2S_LRC => WSEL 
You can choose the GPIOs freely, it doesn't have to be 9/3/1.

ddrram2, 2024-05-20:
Hello Wolle
Thank you very much for this quick response. Yes, I know that I can freely choose gpio, but because I get the yellow warning lines, some people said that it is a fault of the c++ extension. If I install VSC now, what plugins do I need?
I'm not a professional, I just like to find challenges for myself. Can you also tell me if the CJMCU1334 is good for this project?
and can I2S_MCLK be -1?

schreibfaul1, 2024-05-20:
The yellow warnings are not compiler faults. Something changes in the I2S area with almost every version change, which I would have to adapt. But that is not relevant for us. I have this board: [image] Maybe you have to put Mute on Vcc or SCk to GND on your board? MCLK only need very few DACs, that can stay -1

ddrram2, 2024-05-20:
I have this board.
Once the 9, 3, 1 gpio use and are sound for about 10 sec, but that didn't work anymore.
I don't know what to do, I've been working and reading for several weeks, but I'm not giving up
With esp32 devkit c 30pin works my 27, 25, 26 gpio pins.
[image]

schreibfaul1, 2024-05-21:
The board looks identical to mine. There are rarely problems with I2S and I have no solution for the behaviour of your board. Boards with the PCM5101a are best suited.

Arne65, 2024-05-21:
I don't know this board, but you've probably put MUTE on GND, haven't you?
MUTE off = Low
MUTE on = high

Not that the input gets high at some point and is therefore MUTE on.

ddrram2, 2024-05-21:
Hello dear Wolle

I woke up today thinking that today would be good, and it was. I used the original S3.
The settings are:_DOUT 9, BCLK 8, LRC 1, MCLK -1 and the sd card SD_MMC_D0 11, SD_MMC_CLK 13, SD_MMC_CMD 14.
Sometimes there are hiccups, but I'll make a motherboard and everything will be fine.
I will also try with the clone S3.
I don't use a display, I only control it from the web interface, and then there will be RC. But it could also be bluetooth, as I know myself.
Thank you very much for your help and work, I wish you all the best.

ddrram2, 2024-05-21:
Thanks for the tips, they are helpful, the next question is, where can I increase the web server connection time, because it happens that the radio boots and the last used
radio station is playing and I can't change radio station from the web interface because the web server: Ftp server waiting for connection on port ...
and press RESET and then it's good, or not, then reset until it's good. My other question is why I can't change the station from the phone using a web browser.

ddrram2, 2024-05-22:
Good morning
Now I can also connect with the radio webinterface phone. I noticed that I was connected from the PC with a phone
I can't connect, but if I close the webradio on the PC, I can connect by phone.

ddrram2, 2024-05-25:
I want to build a Mini Web Radio for a dear friend of mine and he wants to listen to these two stations:
Radio Beograd 1 - https://rtsradio-live.morescreens.com/RTS_2_001/audio/chunklist.m3u8 - homepage url: https://www.rts.rs/page/radio/sr/live.html
RTV - Radio Novi Sad 2 (Újvidéki Rádió) - http://212.200.230.50:1935/rns2/drugiprogram/livestream/stream.m3u8 - homepage url: https://www.rtv.rs/hu/program/magyar- language-adas
and if I add it to the list of stations and try to connect, Mini Webradio (3.1) make restart again
I opened VLC and the links work. VLC has a terminal and I typed info and got this:
Radio Belgrade 1
info
status change: ( new input: https://rtsradio-live.morescreens.com/RTS_2_001/audio/chunklist.m3u8 )
status change: ( play state: 3 )
+----[ Stream 0 ]
|
| Codec: ADTS
| Type: Audio
| Channels: Stereo
| Sample rate: 48000 Hz
| Bits per sample: 32
|
+----[ end of stream info ]

here is the content of Radio Beograd 1 - https://rtsradio-live.morescreens.com/RTS_2_001/audio/chunklist.m3u8:

#EXTM3U
#EXT-X-VERSION:3
#EXT-X-MEDIA-SEQUENCE:767133
#EXT-X-ALLOW-CACHE:NO
#EXT-X-TARGETDURATION:11
#EXTINF:10.005333,
/RTS_2_001/audio/2024-05-24-H08/audio-2024-05-24-08-13-36.ts
#EXTINF:9.984000,
/RTS_2_001/audio/2024-05-24-H08/audio-2024-05-24-08-13-46.ts
#EXTINF:10.005333,
/RTS_2_001/audio/2024-05-24-H08/audio-2024-05-24-08-13-56.ts
#EXTINF:10.005333,
/RTS_2_001/audio/2024-05-24-H08/audio-2024-05-24-08-14-06.ts
#EXTINF:10.005333,
/RTS_2_001/audio/2024-05-24-H08/audio-2024-05-24-08-14-16.ts
#EXTINF:9.984000,
/RTS_2_001/audio/2024-05-24-H08/audio-2024-05-24-08-14-26.ts
................................................... ................................................... ................
then RTV - Radio Novi Sad 2 (Újvidéki Radio)
info
status change: ( new input: http://212.200.230.50:1935/rns2/drugiprogram/livestream/stream.m3u8 )
status change: ( play state: 3 )
+----[ Stream 1 ]
|
| Codec: MPEG Audio layer 1/2 (mpga)
| Type: Audio
| Channels: Stereo
| Sample rate: 44100 Hz
| Bits per sample: 32
| Bitrate: 192 kb/s
|
+----[ end of stream info ]

here is the content of RTV - Radio Novi Sad 2 (Újvidéki Rádió) - http://212.200.230.50:1935/rns2/drugiprogram/livestream/stream.m3u8:
#EXTM3U
#EXT-X-VERSION:3
#EXT-X-STREAM-INF:BANDWIDTH=191976,CODECS="mp4a.40.34"
chunklist_w1231147418.m3u8

because it doesn't play these two stations, maybe it's a codec problem?

schreibfaul1, 2024-05-25:
Hi ddrram2,
I have found an error in the audioI2S library for the station RTV Radio Beograd. Please reload the library with "Full Clean". One of the streams I have tested is:
http://rtsradio-live.morescreens.com/RTS_2_001/playlist.m3u8     // Radio Beograd 1 Радио Београд
http://rtsradio-live.morescreens.com/RTS_2_004/playlist.m3u8    // Radio Beograd 202

What will foreseeably not work are the links below, in the playlist mp4a is announced and in the chunklist mp3 is in it, that's strange.
[image]

ddrram2, 2024-05-25:
Dear Wole
I'm sorry but I don't understand what to do. 
The stream links you sent work in VLC but not in the radio. The radio does not restart, but send the message:
19:12:49 action: ...  switch to station [36m30[0m
19:12:49 AUDIO_info:  [32mConnect to new host: "http://rtsradio-live.morescreens.com/RTS_2_001/playlist.m3u8"[0m
19:12:49 AUDIO_info:  [32mbuffers freed, free Heap: 149132 bytes[0m
19:12:49 AUDIO_info:  [32mConnection has been established in 95 ms, free Heap: 148784 bytes[0m
19:12:49 lastURL: ..  [37mhttp://rtsradio-live.morescreens.com/RTS_2_001/playlist.m3u8[0m
19:12:49 AUDIO_info:  [32mbandwidth: 128000 bit/s[0m
19:12:49 AUDIO_info:  [32mnew request: "http://rtsradio-live.morescreens.com/RTS_2_001/audio/chunklist.m3u8"[0m
19:12:49 AUDIO_info:  [32mThe host has disconnected, reconnecting[0m
19:12:49 AUDIO_info:  [32mnew request: "http://rtsradio-live.morescreens.com/RTS_2_001/audio//RTS_2_0 ... 4-05-25-19-11-46.ts"[0m
19:12:50 StreamTitle: [33mHTTP/1.1 404 Not Found[0m
19:12:50 StationName: [35m[0m
19:12:50 AUDIO_info:  [32murl is a webpage![0m
19:12:50 Audiocodec:  [33mAAC[0m

My problem is that I don't know where is the audioI2S library for the station RTV Radio Beograd
and how to reload the audioI2S library with "Full Clean"
Did you open the playlist with npp? Do you need to edit the playlist?
Please teach me how to solve these kinds of problems.
Mini Webradio is very fast, I can't freeze it.

schreibfaul1, 2024-05-25:
The audioI2S library is not a direct part of the project, but is loaded externally. To get the new version, please go to the ant and then to "Full Clean". Then compile and upload as usual.
[image]

ddrram2, 2024-05-25:
Dear Wole
Thank you very much for your quick reply, it helped.
Full Clean make the updates and I put the code again all ok.
Radio Beograd 1  https://rtsradio-live.morescreens.com/RTS_2_001/audio/chunklist.m3u8
work good.

21:21:50 action: ...  switch to station [36m30[0m
21:21:50 AUDIO_info:  [32mConnect to new host: "https://rtsradio-live.morescreens.com/RTS_2_001/audio/chunklist.m3u8"[0m
21:21:50 AUDIO_info:  [32mbuffers freed, free Heap: 149444 bytes[0m
21:21:51 AUDIO_info:  [32mSSL has been established in 951 ms, free Heap: 149040 bytes[0m
21:21:51 lastURL: ..  [37mhttps://rtsradio-live.morescreens.com/RTS_2_001/audio/chunklist.m3u8[0m
21:21:51 AUDIO_info:  [32mnew request: "https://rtsradio-live.morescreens.com//RTS_2_001/audio/2024-0 ... 4-05-25-21-20-46.ts"[0m
21:21:51 AUDIO_info:  [32mAACDecoder has been initialized, free Heap: 146752 bytes , free stack 2568 DWORDs[0m
21:21:51 AUDIO_info:  [32msyncword found at pos 0[0m
21:21:51 AUDIO_info:  [32mChannels: 2[0m
21:21:51 AUDIO_info:  [32mSampleRate: 48000[0m
21:21:51 AUDIO_info:  [32mBitsPerSample: 16[0m
21:21:51 AUDIO_info:  [32mBitRate: 143250[0m
21:21:51 AUDIO_info:  [32mAAC HeaderFormat: ADTS[0m
21:21:51 AUDIO_info:  [32mAAC Codec: MPEG-4 LowComplexity[0m
21:21:51 Audiocodec:  [33mAAC[0m
21:21:58 AUDIO_info:  [32mnew request: "https://rtsradio-live.morescreens.com//RTS_2_001/audio/2024-0 ... 4-05-25-21-20-56.ts"[0m
21:22:08 AUDIO_info:  [32mnew request: "https://rtsradio-live.morescreens.com//RTS_2_001/audio/2024-0 ... 4-05-25-21-21-06.ts"[0m
21:22:09 WiFI_info:   RSSI is [36m-67[37m dB[0m
21:22:10 WiFI_info:   RSSI is [36m-57[37m dB[0m
21:22:13 WiFI_info:   RSSI is [36m-66[37m dB[0m
21:22:15 WiFI_info:   RSSI is [36m-57[37m dB[0m
21:22:16 WiFI_info:   RSSI is [36m-66[37m dB[0m
21:22:17 WiFI_info:   RSSI is [36m-56[37m dB[0m
21:22:18 AUDIO_info:  [32mnew request: "https://rtsradio-live.morescreens.com//RTS_2_001/audio/2024-0 ... 4-05-25-21-21-16.ts"[0m
21:22:23 WiFI_info:   RSSI is [36m-67[37m dB[0m
21:22:24 WiFI_info:   RSSI is [36m-57[37m dB[0m
21:22:25 WiFI_info:   RSSI is [36m-66[37m dB[0m
21:22:28 AUDIO_info:  [32mnew request: "https://rtsradio-live.morescreens.com//RTS_2_001/audio/2024-0 ... 4-05-25-21-21-26.ts"[0m
21:22:29 WiFI_info:   RSSI is [36m-57[37m dB[0m
21:22:31 WiFI_info:   RSSI is [36m-68[37m dB[0m
21:22:34 WiFI_info:   RSSI is [36m-56[37m dB[0m
21:22:35 WiFI_info:   RSSI is [36m-66[37m dB[0m
21:22:38 WiFI_info:   RSSI is [36m-57[37m dB[0m
21:22:38 AUDIO_info:  [32mnew request: "https://rtsradio-live.morescreens.com//RTS_2_001/audio/2024-0 ... 4-05-25-21-21-36.ts"[0m
21:22:39 WiFI_info:   RSSI is [36m-68[37m dB[0m
21:22:40 WiFI_info:   RSSI is [36m-57[37m dB[0m
21:22:44 WiFI_info:   RSSI is [36m-66[37m dB[0m
21:22:48 WiFI_info:   RSSI is [36m-56[37m dB[0m
21:22:48 AUDIO_info:  [32mnew request: "https://rtsradio-live.morescreens.com/RTS_2_001/audio/chunklist.m3u8"[0m
21:22:48 AUDIO_info:  [32mnew request: "https://rtsradio-live.morescreens.com//RTS_2_001/audio/2024-0 ... 4-05-25-21-21-46.ts"[0m
21:22:49 WiFI_info:   RSSI is [36m-67[37m dB[0m
21:22:52 WiFI_info:   RSSI is [36m-58[37m dB[0m
21:22:53 WiFI_info:   RSSI is [36m-66[37m dB[0m
21:22:54 WiFI_info:   RSSI is [36m-57[37m dB[0m
21:22:55 WiFI_info:   RSSI is [36m-66[37m dB[0m
21:22:58 AUDIO_info:  [32mnew request: "https://rtsradio-live.morescreens.com//RTS_2_001/audio/2024-0 ... 4-05-25-21-21-56.ts"[0m
21:23:00 WiFI_info:   RSSI is [36m-57[37m dB[0m
21:23:01 WiFI_info:   RSSI is [36m-66[37m dB[0m
21:23:02 WiFI_info:   RSSI is [36m-57[37m dB[0m
21:23:08 WiFI_info:   RSSI is [36m-67[37m dB[0m
21:23:08 AUDIO_info:  [32mnew request: "https://rtsradio-live.morescreens.com//RTS_2_001/audio/2024-0 ... 4-05-25-21-22-06.ts"[0m
21:23:12 WiFI_info:   RSSI is [36m-58[37m dB[0m
21:23:13 WiFI_info:   RSSI is [36m-67[37m dB[0m
21:23:15 WiFI_info:   RSSI is [36m-57[37m dB[0m
21:23:16 WiFI_info:   RSSI is [36m-67[37m dB[0m
21:23:19 AUDIO_info:  [32mnew request: "https://rtsradio-live.morescreens.com//RTS_2_001/audio/2024-0 ... 4-05-25-21-22-16.ts"[0m
21:23:21 WiFI_info:   RSSI is [36m-57[37m dB[0m
21:23:27 WiFI_info:   RSSI is [36m-66[37m dB[0m
21:23:29 AUDIO_info:  [32mnew request: "https://rtsradio-live.morescreens.com//RTS_2_001/audio/2024-0 ... 4-05-25-21-22-26.ts"[0m
21:23:29 WiFI_info:   RSSI is [36m-59[37m dB[0m
21:23:30 WiFI_info:   RSSI is [36m-66[37m dB[0m
21:23:33 WiFI_info:   RSSI is [36m-56[37m dB[0m
21:23:35 WiFI_info:   RSSI is [36m-66[37m dB[0m
21:23:39 AUDIO_info:  [32mnew request: "https://rtsradio-live.morescreens.com//RTS_2_001/audio/2024-0 ... 4-05-25-21-22-36.ts"[0m
[... 20 lines cut ...]
21:23:46 AUDIO_info:  [32mID3 normal frames[0m
21:23:46 AUDIO_info:  [32mAudio-Length: 240384[0m
21:23:46 AUDIO_info:  [32msyncword found at pos 0[0m
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42030775  PS      : 0x00060f30  A0      : 0x8202328c  A1      : 0x3fcc3cd0  
A2      : 0x3fca5208  A3      : 0x3fcc3d38  A4      : 0x3d807ba0  A5      : 0x00003200  
A6      : 0x3d81601c  A7      : 0x00000000  A8      : 0x0002e629  A9      : 0x00000002  
A10     : 0xfa000000  A11     : 0x0000e629  A12     : 0xc5800000  A13     : 0xe0c40000  
A14     : 0x3fca861c  A15     : 0xa01fc000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[... cut ...]
Backtrace: 0x42030772:0x3fcc3cd0 0x42023289:0x3fcc3d10 0x420238c5:0x3fcc3d80 0x420299ae:0x3fcc3db0 0x42029be0:0x3fcc3e00 0x4200386d:0x3fcc3e20

ELF file SHA256: c8718ec85ebc027d

E (11149) esp_core_dump_flash: Not enough space to save core dump!
E (11154) esp_core_dump_elf: Failed to prepare core dump storage (257)!
E (11161) esp_core_dump_common: Core dump write binary failed with error=257
Rebooting...
 
I think this is also a codec problem.
Thank you very much for your help, I will come again.
If I have a question about the IR remote, should I open a new topic?

closed 2024-07-09 by github-actions[bot]