schreibfaul1/ESP32-MiniWebRadio #424: Change log 3.1

FranZatGIT, 2024-05-22:
Hi Wolle, what are the differences of 3.1 compared to 3.0. Is there something like a changelog?

Thank you

schreibfaul1, 2024-05-22:
The equaliser functions can now be set from the display. If you are using a BT transmitter, you can now also control it from the display. You can now set a separate alarm time for each day of the week. Internally, many elements are programmed in an object-oriented way, which increases reliability. Arduino is run as a component again, allowing for advanced settings. This increases the WiFi throughput, so that stations with up to 320KBit/s work smoothly again. Sometimes WiFi problems occur, such as unavailable DNS, which is now neutralised. The credentials are sent to the router up to 3 times, which is useful as some routers will deny access if the connection is briefly interrupted.

valera39, 2024-05-22:
And the mode of switching off when long pressing the * button has been fixed

schreibfaul1, 2024-05-23:
and I have added an "OFF" button, the red one,  to switch to sleep mode (sound off and display dark)
[image]

valera39, 2024-05-23:
Thanks, can I use the remote to put it in sleep mode?

FranZatGIT, 2024-05-23:
Hi Wolle, many thanks - all of that new things- really great. I uploaded the new version. A few things I noticed but I need to verify:

- I can add stations in the web tool , see it in the drop down list but i cannot select the new sender and its not in the list on the display
- The VU Meter are sometimese missing bars ( looks like an update issue)
- from time to time 192k sender (especially when changing to the sender ) have short interruptions/distortion
- I cannot see the new Off button ( 240*320 ILI9341) (equalizer and BT are there)

Just to mention it - I added some code lines for my implementation (Speaker relais, RGB LED, Power Control of Amplifier/Radio, Batterie Voltage check,...)  BUT for the above mentioned points that lines /Lib has been commented out

has anyone else recognized that point?

Thanks Franz

schreibfaul1, 2024-05-23:
Hi Valera,
that will work, the button you have assigned to "Mute" can switch sleep mode on and off.
[image]

schreibfaul1, 2024-05-23:
Hello Franz,

- if you have added a new station, the "new" station list is transferred to the NVS with the save button. You will find the station in the log. The NVS content is updated with the load button. This is the same routine as at startup. This is from the time when there was still a stations.cvs file. I will improve this
- The VU meter is not always drawn correctly, will be corrected soon
- I only recently added the Off button. The icon, [image], is probably missing and needs to be copied from the zip file.

FranZatGIT, 2024-05-25:
Hi Wolle,

- regarding Off Button:

well I again copied the whole content of the ZIP to the SD card   and the icon is there:
[image]
However  still no Icon on the display:
[image]

But please don't focus on that - that Button is'n really important to me ( I have the radio anyway running on battery and switch off the Power. 

- Regarding new stations

What is really a pain is that I 'm not able to add new stations. No matter if I press load or save.

- Feature request / help request

I have some kind of feature request - I would like to display the Battery Voltage or the percentage of Battery in the headline (somewhere between the Fuction Name e.g. "Internet" and "Vol"). The Volatage is already read and calculated in the code - just need to add another object in the headline. I will try to do it myself but your code is quite complex  here - would you give me a help if needed?

FranZatGIT, 2024-05-25:
Hi again, regarding the Icon - you said you recently added the icon - does that just mean the JPG or also in the code?
The last time I pulled the code was a couple of days ago... 

Thanks Franz

schreibfaul1, 2024-05-25:
All graphics are duplicated. Scaling an image with the ESP32 simply takes too long.
Therefore, there is always a (s)mall and (m)edium folder. If an image is not found, it can be seen in the log.

I have just added a station and created hardcopies.
[CommunityWebBrowser](https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/docs/Using%20the%20RadioBrowser.pdf) This is better than an explanation in text form.

How big is your battery? The project is an energy guzzler. Of course I can display the operating voltage.

FranZatGIT, 2024-05-26:
Hi Wolle, 
Yes I saw the 2 folders for different screen resolutions when I tested different displays at the  beginning. The off button icons were there in both folders. So I pulled a complete fresh clone of the code files ( not SD)  from github an reapplied my changes. The icon is there and also adding a new station is possible again. No Idea what exactly happened but I assume i used a main.ccp version  before you added the new button...
So all good so far :) maybe adding a build number counter to the " Version 3.1   May 21/2024  " in the file comment would be great to recognize that some changes were done. 

regarding the battery (see picture) - I use a 6s 18650 Battery for my project total of 66Wh. The ESP including Amp at moderate volume consumes about 2.5-4W  - so 15-20h play time - enough for me ;)  The device is not 100% ready but you can see also on the picture  (with Alu Heat sink) the charging board 140W PD3.1 - all USB C devices can be used to charge - from old Handy charger to a powerful Notebook power supply . And its bidirectional you can also use it as supply for the Handy or a notebook.
The Amp contains it's own BT receiver - The Radio can also be used a BT Box - in this case the ESP goes in sleep mode.

But back to the voltage - I would like to show the Battery Voltage / percentage in the headline. In my code amendment I already implemented a power down function when the battery voltage is to low. Meaning I already have the voltage in the code and calculation the percentage is also no rocket science - just adding a new object to the headline is a bit a challenge to me. (i'm just a DIY C++ "modifier" and don't have much knowledge/experience) I only partially understand your code. Any help much appreciated :)

[image]

FranZatGIT, 2024-05-27:
Hi Wolle,

i was able to find the relevant code lines :) The Battery percentage is now there.
[image]

schreibfaul1, 2024-05-27:
Very good!

schreibfaul1, 2024-05-27:
and very clean construction, may I use your photos for the gallery of user devices?

FranZatGIT, 2024-05-27:
Hi Wolle, sure - here are some more recent pics with LWL from charger board.  However the box itself is not finished yet - but you can imagine how i will look like...
[image]
![WhatsApp Bild 2024-05-27 um 23 38 18_fbd8ece2](https://github.com/schreibfaul1/ESP32
[image]
-MiniWebRadio/assets/162827986/ed84d3ea-7769-48bc-bfab-bb0aabcacd96)
[image]
[image]
[image]

schreibfaul1, 2024-05-28:
thank you, I have taken the photos
https://github.com/schreibfaul1/ESP32-MiniWebRadio/wiki/User-devices

FranZatGIT, 2024-05-28:
Thank you - I will send you updated pics when building of box is progressing 

I studied the way you change the RSSI Symbol and did the same for the battery :)
[image]
[image]
[image]
[image]

[image]

closed 2024-07-12 by github-actions[bot]