schreibfaul1/ESP32-MiniWebRadio #427: The entire title is missing from the display.

hevet, 2024-05-23:
I have a display on ili9341. Sometimes the full title is not displayed on the display, but the full title is displayed on the web interface. An example of what it looks like in the photos below.

[image]

[image]

schreibfaul1, 2024-05-23:
At the moment, the possible length of the text is estimated based on the number of letters and then the character size is selected. Sometimes it doesn't fit. I will do this differently in future.
P.S. From V3.1 you can click on the Into image via the web interface (tab "Info") and a hardcopy of the display will be created. This saves you having to take a photo.

schreibfaul1, 2024-05-29:
I have now changed this. We have 10 font sizes. Starting with the largest, the system checks whether the text fits into the given window. If this is not the case, the font size is reduced, checked ... etc. The font size is no longer based on an estimate, the font is now often larger than in the previous version.

hevet, 2024-05-29:
Thanks @schreibfaul1, When I come back, I'll check how it works now :)

hevet, 2024-05-29:
@schreibfaul1 Something is wrong. When I turn on the radio to any station, the information is displayed correctly. I will switch to another one and the information will no longer be displayed on the next one. I go back to the previous ones and it doesn't display anything anymore. You have to turn the radio off and on again for the information to appear. After switching to other stations, the information disappears again and you have to restart it to reappear.
From what I can see, the information only appears when I switch with the remote control and select the station using the numeric keypad. The information is always displayed in the web interface.

hevet, 2024-05-29:
@schreibfaul1 Something is wrong. When I turn on the radio that has no information, it will mean that the information will not be displayed on any station if I switch up and down. If I select a station using the numeric keypad, the information will appear again. The information is always correct in the web interface.

hevet, 2024-05-29:
@schreibfaul1 I added a link to a video showing the described problem.
https://drive.google.com/file/d/17vC1TxRus39dR8XoQojxhxOhG3NypZEq/view?usp=drivesdk

schreibfaul1, 2024-05-29:
I didn't notice this during my tests. I had hidden the text box for the stream tiltles when there is nothing to display. After that, nothing is visible. I think it will be better now.

hevet, 2024-05-29:
@schreibfaul1 It's better, it looks good :)

closed 2024-07-13 by github-actions[bot]