schreibfaul1/ESP32-MiniWebRadio #429: Detected a whitespace character in project paths.

JozefKc, 2024-05-23:
Hi Wolle!
I tried to install today's /May 23th/ version. As usual, I corrected necessary things in common.h and tried to build the project. After few seconds I received a message: **Detected a whitespace character in project paths**. 
I am using ESP32-S3N16R8 with ILI9488, PCM5102, BT card. Till now I had no problem. Everything worked till today.
I made a screenshot:
[image]
It seems like the compiler does not know where to find h-libraries....

Thank you for your help in advance.

schreibfaul1, 2024-05-23:
I'm not quite sure why the error is occurring.
The project name has not changed, it is still "ESP32-MiniWebRadio".
Maybe the path where you copied the project contains spaces?

JozefKc, 2024-05-25:
Thank you Wolle. The folder  name did consist space character.....

closed 2024-07-09 by github-actions[bot]