schreibfaul1/ESP32-MiniWebRadio #434: slow stream, dropouts are possible

valera39, 2024-05-26:
Hi Dear Wolle!
In version 3 of the radio, on some stations playback is interrupted, look at the log please. On the receiver version 2 all streams work without fail.

ESP32 Chip: ESP32-S3
Arduino Version: 2.0.16
ESP-IDF Version: 4.4.7
MiniWebRadio     Version 3.1   May 21/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8381399 bytes
RESET_REASON: Reset due to power-on event

00:00:00    
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 3761.4 MB by 3781.5 MB free
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  2392][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  2414][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Honor
00:00:03 setup: ....  6 WiFi networks found
00:00:03 setup: ....  ASUS_AC66U (-54)
00:00:03 setup: ....  TP-LINK_E345E0_RPT (-67)
00:00:03 setup: ....  TP-LINK_E345E0 (-77)
00:00:03 setup: ....  ASUS_AC66U_RPT (-80)
00:00:03 setup: ....  PS4network (-84)
00:00:03 setup: ....  GG (-88)
[ 10928][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[ 10934][I][WiFiMulti.cpp:119] run(): [WIFI] 6 networks found
[ 10941][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 1 (-53)
[ 11021][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:10 WiFI_info:   Connecting WiFi...
00:00:10 WiFI_info:   mDNS name: MiniWebRadio
00:00:10 setup: ....  connected to ASUS_AC66U, IP address is 192.168.1.130
[ 11062][I][ArduinoOTA.cpp:141] begin(): OTA server at: MiniWebRadio.local:3232
00:00:10 ftpServer:   Buffers allocated: 4622 bytes
00:00:10 rtime_info:  Initializing SNTP
02:00:10 rtime_info:  Waiting for system time to be set... (1/10)
18:28:22 rtime_info:  18:28:22
18:28:22 audiotask:   is pinned to core 0
18:28:22 audiotask:   priority is 2
18:28:22 AUDIO_info:  commFMT = 1
18:28:22 setup: ....  Number of saved stations: 92
18:28:22 setup: ....  current station number: 56
18:28:22 setup: ....  current volume: 2
18:28:22 setup: ....  last connected host: https://cast1.torontocast.com:1335/stream
18:28:22 setup: ....  connection timeout: 1000 ms
18:28:22 setup: ....  connection timeout SSL: 2500 ms
18:28:23 action: ...  switch to station 56
18:28:23 AUDIO_info:  Connect to new host: "http://harddanceradio.ddns.is74.ru:8000/eurodisco.m3u"
18:28:23 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
18:28:23 AUDIO_info:  buffers freed, free Heap: 166580 bytes
18:28:23 AUDIO_info:  Connection has been established in 458 ms, free Heap: 165304 bytes
18:28:23 AUDIO_info:  Connect to new host: "http://harddanceradio.ddns.is74.ru:8000/eurodisco"
18:28:23 AUDIO_info:  buffers freed, free Heap: 165848 bytes
18:28:23 AUDIO_info:  Connection has been established in 83 ms, free Heap: 164860 bytes
18:28:23 bitRate:     256Kbit/s
18:28:23 icy-descr:   Eurodisco Syntpop Italodisco SynthWave 80s 90s
18:28:23 StationName: EURODISCO '80 [HDR]
18:28:23 icy-url: ..  http://harddanceradio.ddns.is74.ru:8000
18:28:23 AUDIO_info:  AACDecoder has been initialized, free Heap: 164336 bytes , free stack 5068 DWORDs
[... 244 lines cut ...]
[ 11074][I][WiFiMulti.cpp:119] run(): [WIFI] 6 networks found
[ 11081][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 1 (-63)
[ 16121][E][WiFiMulti.cpp:187] run(): [WIFI] Connecting Failed (6).
00:00:15 WiFI_info:   WiFi credentials are not correct
00:00:15 RECONNECTION try 1
[ 16129][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[ 16152][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Honor
00:00:15 setup: ....  -2 WiFi networks found
00:00:15 WiFI_info:   WiFi credentials are not correct
00:00:15 RECONNECTION try 2
[ 16190][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[ 16205][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Honor
00:00:15 setup: ....  -1 WiFi networks found
00:00:15 WiFI_info:   WiFi credentials are not correct
00:00:15 RECONNECTION try 3
[ 16391][W][tft.cpp:2669] validCharsInString(): char is not printable 0x0A
00:00:15 Accesspoint: IP: 192.168.4.1
00:00:22 setup: ....  6 WiFi networks found
00:00:22 setup: ....  TP-LINK_E345E0_RPT (-61)
00:00:22 setup: ....  ASUS_AC66U (-62)
00:00:22 setup: ....  ASUS_AC66U_RPT (-73)
00:00:22 setup: ....  TP-LINK_E345E0 (-83)
00:00:22 setup: ....  GG (-85)
00:00:22 setup: ....  PS4network (-87)
00:00:22 ftpServer:   Ftp server waiting for connection on port 21

schreibfaul1, 2024-05-26:
I have adopted the sdkconfig from V2. But somehow the settings have been reset. Now it will be okay.

valera39, 2024-05-27:
With the latest changes there are improvements, but still I can't get the bluetooth module and ESP32 to work together. If I do not supply power to the bluetooth module, there are no records "slow stream, dropouts are possible" at all, but when working together this error is present all the time. I changed the position of the module relative to ESP32 in different positions, I moved it to a distance of about 20cm, but there is no result. What else can I think of?

ESP32 Chip: ESP32-S3
Arduino Version: 2.0.16
ESP-IDF Version: 4.4.7
MiniWebRadio     Version 3.1   May 21/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 8388608 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 40 MHz
PSRAM total size: 8381399 bytes
RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************    
00:00:00        *     MiniWebRadio V3     *    
00:00:00        ***************************    
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9488
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 3761.4 MB by 3781.5 MB free
00:00:01 setup: ....  seek for stations.csv
00:00:01 setup: ....  stations.csv found
00:00:01 setup: ....  seek for WiFi networks
[  2530][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ASUS_AC66U
[  2552][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: Honor
00:00:03 setup: ....  5 WiFi networks found
00:00:03 setup: ....  ASUS_AC66U (-57)
00:00:03 setup: ....  TP-LINK_E345E0_RPT (-60)
00:00:03 setup: ....  ASUS_AC66U_RPT (-72)
00:00:03 setup: ....  TP-LINK_E345E0 (-78)
00:00:03 setup: ....  GG (-82)
[ 11068][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[ 11074][I][WiFiMulti.cpp:119] run(): [WIFI] 7 networks found
[ 11081][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 2C:FD:A1:C2:44:F0 SSID: ASUS_AC66U Channel: 6 (-56)
[ 11169][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:10 WiFI_info:   Connecting WiFi...
00:00:10 WiFI_info:   mDNS name: MiniWebRadio
00:00:10 setup: ....  connected to ASUS_AC66U, IP address is 192.168.1.130
[ 11209][I][ArduinoOTA.cpp:141] begin(): OTA server at: MiniWebRadio.local:3232
00:00:10 ftpServer:   Buffers allocated: 4622 bytes
00:00:10 rtime_info:  Initializing SNTP
02:00:10 rtime_info:  Waiting for system time to be set... (1/10)
02:00:12 rtime_info:  Waiting for system time to be set... (2/10)
02:00:14 rtime_info:  Waiting for system time to be set... (3/10)
17:18:10 rtime_info:  17:18:10
17:18:10 audiotask:   is pinned to core 0
17:18:10 audiotask:   priority is 2
17:18:10 AUDIO_info:  commFMT = 1
17:18:10 setup: ....  Number of saved stations: 92
17:18:10 setup: ....  current station number: 56
17:18:10 setup: ....  current volume: 2
17:18:10 setup: ....  last connected host: http://harddanceradio.ddns.is74.ru:8000/eurodisco
17:18:10 setup: ....  connection timeout: 1000 ms
17:18:10 setup: ....  connection timeout SSL: 2500 ms
17:18:11 action: ...  switch to station 56
17:18:11 AUDIO_info:  Connect to new host: "http://harddanceradio.ddns.is74.ru:8000/eurodisco.m3u"
17:18:11 AUDIO_info:  PSRAM found, inputBufferSize: 638965 bytes
17:18:11 AUDIO_info:  buffers freed, free Heap: 164996 bytes
17:18:11 AUDIO_info:  Connection has been established in 110 ms, free Heap: 163956 bytes
17:18:11 AUDIO_info:  Connect to new host: "http://harddanceradio.ddns.is74.ru:8000/eurodisco"
17:18:11 AUDIO_info:  buffers freed, free Heap: 159604 bytes
17:18:11 AUDIO_info:  Connection has been established in 96 ms, free Heap: 164016 bytes
17:18:11 bitRate:     256Kbit/s
17:18:11 icy-descr:   Eurodisco Syntpop Italodisco SynthWave 80s 90s
17:18:11 StationName: EURODISCO '80 [HDR]
17:18:11 icy-url: ..  http://harddanceradio.ddns.is74.ru:8000
17:18:11 AUDIO_info:  AACDecoder has been initialized, free Heap: 162768 bytes , free stack 5072 DWORDs
[... 215 lines cut ...]
17:19:49 AUDIO_info:  slow stream, dropouts are possible
17:19:50 AUDIO_info:  slow stream, dropouts are possible
17:19:51 AUDIO_info:  slow stream, dropouts are possible
17:19:51 AUDIO_info:  Stream lost -> try new connection
17:19:51 AUDIO_info:  Connect to new host: "http://harddanceradio.ddns.is74.ru:8000/eurodisco"
17:19:51 AUDIO_info:  buffers freed, free Heap: 159836 bytes
17:19:52 AUDIO_info:  Connection has been established in 370 ms, free Heap: 160936 bytes
17:19:52 bitRate:     256Kbit/s
17:19:52 icy-descr:   Eurodisco Syntpop Italodisco SynthWave 80s 90s
17:19:52 StationName: EURODISCO '80 [HDR]
17:19:52 icy-url: ..  http://harddanceradio.ddns.is74.ru:8000
17:19:52 AUDIO_info:  AACDecoder has been initialized, free Heap: 159724 bytes , free stack 5072 DWORDs
17:19:52 lastURL: ..  http://harddanceradio.ddns.is74.ru:8000/eurodisco
17:19:52 AUDIO_info:  stream ready
17:19:52 AUDIO_info:  syncword found at pos 703
17:19:52 AUDIO_info:  syncword found at pos 0
17:19:52 AUDIO_info:  Channels: 2
17:19:52 AUDIO_info:  SampleRate: 44100
17:19:52 AUDIO_info:  BitsPerSample: 16
17:19:52 AUDIO_info:  BitRate: 264944
17:19:52 AUDIO_info:  AAC HeaderFormat: ADTS
17:19:52 AUDIO_info:  AAC Codec: MPEG-4 LowComplexity
17:19:53 StreamTitle: On-Line: 3 / Secchi & Orlando Johnson - I Say Yeah (shaman.edit) (05:58)
17:19:54 AUDIO_info:  slow stream, dropouts are possible
17:19:56 AUDIO_info:  slow stream, dropouts are possible

valera39, 2024-05-27:
Maybe prohibit bluetooth and wifi to work together, disconnect the module by pulsing "connect" when wifi is running or the reason is different?

schreibfaul1, 2024-05-27:
Your reception field strength of -56dB is very weak. BT and WiFi use the same band, BT performs "frequency hopping" to occupy a different channel. Try to position the BT emitter and the ESP32 as far apart as possible, the antennas should be orthogonal to each other in the x, y and z planes. If this is not possible, insert some antistatic foil between the two antennas.

valera39, 2024-05-27:
Dear Wolle!
Thanks for the quick response and advice. I will continue my experiments. It seems to me that to accommodate a radio in a small enclosure, the screen option might work well

valera39, 2024-05-27:
Dear Wolle!
Unfortunately all my experiments to eliminate the above error when using bluetooth and wifi module together have failed. I have checked all your tips on the location of the modules, used grounding shield, placed the radio near the router, changed "AUDIOTASK_CORE" values, but in vain. Please check this station http://harddanceradio.ddns.is74.ru:8000/eurodisco with bluetooth module connected, what will be your result? Even without connected bluetooth module this station sometimes writes in the log " AUDIO_info: slow stream, dropouts are possible".
I am left with the last option, to use ESP 12K-S3 Ai-Thinker, otherwise to my regret I will have to give up bluetooth, or what is extremely undesirable, to use its mechanical disconnection with a switch.
So please add to the documentation a working diagram with the correct connection pins for ESP 12K-S3.
Thank you, with best wishes for success!

schreibfaul1, 2024-05-28:
EURODISCO '80 has been running for an hour !
 
[image]

The problem is the weak WiFi signal. Do you have the possibility to open a second access point, e.g. a WiFI repeater?

valera39, 2024-05-28:
Why it works for you and not for me, because we have the same components, I am very puzzled.
Yes, I've tried connecting to both a cell phone internet distribution and a repeater from the next room, together with the bluetooth module does not work. Without it, if the radio is located 1-2 meters away from the access point, playback goes smoothly. What is interesting, the receiver of the second version on ESP32 wrower, stands much farther from the access point and works without fail at a lower signal level on the internal antenna. Perhaps the ESP32S3 boards I use (and I tried N8 and N16) have defects in sensitivity. I have a proven ESP-12k -S3 with an external antenna, and if you can tell me the correct pins to connect it, then maybe this is a chance.

schreibfaul1, 2024-05-28:
1-2 metres to the access point is perfect, then the received field strength should be estimated at -25 ... -30dB. But less than -50 is bad.

Jefke1, 2024-05-28:
Hi valera39,
At the beginning of April I had problems with WiFiMultie. I had an rssi of -46 but no connection.When connected, i also suffered from dropouts, my modem was outdated.
I replaced it with a recent model (Wifi6) and everything works properly now. 
While V1 and V2 had no problem at all. 
I hope this might help you.

valera39, 2024-05-28:
[Jefke1](https://github.com/Jefke1), Thanks, I have already tried that option. I think I've tried everything, but maybe I'm missing something. In my case, I have a strange situation: two exactly the same radios with different firmware work differently from the same router. And even if I swap the firmware, the second version still works like a Swiss watch without fail.
I will think, but my knowledge is limited. Maybe Mr. Volle will change something. I hope so. But he has everything working perfectly.

schreibfaul1, 2024-06-23:
Hi Valera,
BT and WiFi use the same frequency range. The antennas can influence each other, especially if the WiFi signal is not very strong. I suspect that the antennas of the ESP32 and the BT emitter "see" each other. You can cover the BT emitter with antistatic film without changing anything in the design. You can simply wrap the BT emitter with it, the BT range is then only slightly lower. Such film can also shield the audio amplifier against interference from radio signals.
This is what I mean by antistatic film:
[image]
Electronic components are often packaged with it by the manufacturer.

valera39, 2024-06-23:
Do you use antstatic film in your radio too?

closed 2024-08-07 by github-actions[bot]