schreibfaul1/ESP32-MiniWebRadio #440: W5500

valera39, 2024-05-30:
Hi Dear Wolle!
Is it possible to add support for a module like the W5500 [https://www.amazon.de/-/en/AZDelivery-Ethernet-Network-Internet-Arduino/dp/B07ZRF96Y1](url) and, for example, provide in the settings the option to choose between LAN and WiFi.
This may seem irrelevant, but for me and hopefully for some other users it would be necessary.
Thanks!

TubaFonse, 2024-05-30:
Hi Valera39, Hi Wolle, 
good idea! I'm also interested to connect the MWR-V3 direct to Ethernet. I have the LAN8720 Ethernet Board. 
Best regards, Alfons. 
[image]

valera39, 2024-05-30:
[TubaFonse](https://github.com/TubaFonse), Very interesting, I didn't know such a board existed.

schreibfaul1, 2024-05-31:
I don't have any experience with the network components, nor do I have this HW.

To come back to the poor WiFi reception quality: if I'm interpreting this correctly, was it better in previous versions? Then it can probably only be something deep under the surface. You could go back to January in platformio.ini with:
````

;platform = https://github.com/platformio/platform-espressif32.git#v6.7.0
;platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.16
;framework = arduino, espidf

platform = https://github.com/platformio/platform-espressif32.git#v6.4.0
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.14
framework = arduino, espidf
````
Would that help?
We have three access points in our house. The ESP32 sometimes connects to the AP that is furthest away. I have added a _scanNetworks_, an _autoReconnect_ and _MDS.addService_ to the `connectToWiFi()`. What happens if this is deleted?

valera39, 2024-05-31:
Hi Dear Wolle!
I tried your tips with changes to platformio.ini. You are a genius!!! I'm afraid to jinx it, but it seems to be working.
The signal strength has increased a bit, it has become stable and does not fluctuate, drops in the flow at the problem station have almost stopped. Stable operation with bluetooth module and fiwi is now possible. 
I didn't touch scanNetworks, autoReconnect and MDS.addService that you added to connectToWiFi().
Special thanks for the sleep mode clock. Only if possible, is it possible to remove all the extra information from the screen and leave only the clock?

Translated with DeepL.com (free version)

schreibfaul1, 2024-05-31:
That's great. I now wonder what the actual cause is, the IDF version is now 4.4.5 and was previously 4.4.7.
The header and footer in clock sleep mode is now hidden.

closed 2024-07-16 by github-actions[bot]