schreibfaul1/ESP32-MiniWebRadio #443: Problem when scrolling long lists on DLNA server

cschwick, 2024-05-31:
Dear Wolle, 

First of all thanks a lot for this spectacular project! ! All my compliments for this masterpiece!!

I think I identified a small problem when trying to scroll long lists on my DLNA server (with the touch screen). If I understood the code right, it should scroll 9 entries towards the top or the bottom of the list if you "drag a touch" more than 2 rows on the display. However, in my case, many scroll steps were executed one after the other.
I solved the problem in the class dnlaList in the function "hasClicked" : When scrolling occurs I added the line
m_oldY = Y;
directly after the each of the statements 
mcurrItemNr += (or -=) 9;

The code reads now (starts at line 1591 in my case, not sure if I added some lines above or not...):

 ---
    void hasClicked(uint16_t x, uint16_t y){
        if(!m_buff) m_buff = x_ps_malloc(512);
        m_itemListPos = y / (m_h / 10);

        if(m_oldY && (m_oldY + 2 *m_lineHight < y)) {
            m_ra.val1 = 0;
            *m_dlnaLevel = m_oldDlnaLevel;
            if(m_currItemNr == 0) goto exit;
            if(m_currItemNr >  9) {
                SerialPrintfln("9 items further up");
                m_currItemNr -= 9;
                m_oldY = y;
            }
            else m_currItemNr = 0;
            m_browseOnRelease = 4;
            m_chptr = NULL;
        }
        if(m_oldY && (m_oldY - 2* m_lineHight > y)) {
            m_ra.val1 = 0;
            *m_dlnaLevel = m_oldDlnaLevel;
            if(m_currItemNr + 9 >= m_dlnaMaxItems) {
                goto exit;
            }
            SerialPrintfln("9 items further down");
            m_currItemNr += 9;
            m_browseOnRelease = 4;
            m_chptr = NULL;
            m_oldY = y;
        }

        if(m_oldX || m_oldY) goto exit;

              ... no further changes ...
---

This change makes that I scroll systematically 9 lines up or down if I drag between 2 and 4 lines vertically on the screen. I can even scroll "faster" if I drag larger distances which is convenient. 

Not sure if my touch display behaves differently than yours or if I miss something, but I think the behavior with the change is as it was intended, no? 

Again thanks lot for all your work!

Best wishes, Chris

schreibfaul1, 2024-06-01:
Hi Chris,
thanks for your contribution. I have tested your sample code and it works so far.
In order not to lose the overview with long lists, I will still display the position, e.g. 205-214/930.
I will implement your suggestion in the next few days.

cschwick, 2024-06-01:
Thank you Wolle, whenever it is convenient for you !
Best wishes, Christoph

schreibfaul1, 2024-06-05:
I have added fast scrolling. At least over two entries is normal scrolling (up, down) to the next page. Over 6 or more lines is fast scrolling over ~4 pages. In addition, there is now information at the top right where you are in the list.
[image]

closed 2024-07-20 by github-actions[bot]