schreibfaul1/ESP32-MiniWebRadio #444: Stations not updated

BBrz, 2024-06-01:
Hello,
I've added some more stations in stations.csv.
In web interface all of them are showing but if I select one of the new ones it plays one of the original's one.
If I use the display buttons for next and previous station, it acts like the new stations aren't there and plays only the original 14 stations provided initially.
Any ideas?

Thanks

schreibfaul1, 2024-06-01:
Changing something in the stations.csv is tedious.
It is better to add the stations via the web interface.
https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/docs/Using%20the%20RadioBrowser.pdf

BBrz, 2024-06-08:
Looks like it's working from the web interface.
Thank you!

closed 2024-06-08 by BBrz