schreibfaul1/ESP32-MiniWebRadio #445: I have a problem with the platformIO.ini MissingPackageManifestError:

ats3788, 2024-06-02:
MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package
Hello  
I can't compile this Project any clue. Thx

schreibfaul1, 2024-06-02:
The "packages" were probably loaded incompletely Please search for the .platfirnio folder and delete the packages and platforms folders. The folders will be recreated automatically.

ats3788, 2024-06-11:
Thank you but you dis a syntax error it's .platformio not .platfirnio
I don't see any .platformio I opened a case in the PlatformIO Community 
I hope I get it work.

ats3788, 2024-06-11:
Sorry Schreibfaul
I deleted the whole .platformIO Folder and let PlatformIO do the work and now I can compile it. It was not clear for me to delete this folder because my minigw was in it. Cool Thank you for this awesome Project. Alles gute wünsche ich Dir.

closed 2024-07-26 by github-actions[bot]