schreibfaul1/ESP32-MiniWebRadio #446: MP3Decoder could not be initialized

Namzor1955, 2024-06-04:
Hello,
On Version 3.01f Apr 18/2024, all selected stations work, but when I replace the update with Version 3.1 May 29/2024, the stations do not work.
An error occurs

[ 17247][E][mp3_decoder.cpp:1600] MP3Decoder_AllocateBuffers(): not enough memory to allocate mp3decoder buffers
11:42:25 AUDIO_info: The MP3Decoder could not be initialized

What to do?

ESP32 Chip: ESP32-D0WD
Arduino Version: 3.0.0
ESP-IDF Version: 5.1.3
MiniWebRadio     Version 3.01f  Apr 18/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 4194304 bytes, speed 40 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 27 MHz
PSRAM not found! MiniWebRadio doesn't work properly without PSRAM!RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************
00:00:00        *     MiniWebRadio V3     *
00:00:00        ***************************
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 3751.2 MB by 3789.0 MB free
[   328][I][esp32-hal-ledc.c:131] ledcAttachChannel(): LEDC attached to pin 32 (channel 0, resolution 8)
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
[   708][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: wirerozman
[   714][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: wirerozman
[   716][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: wifirozman
[ 10038][I][WiFiMulti.cpp:134] run(): [WIFI] scan done
[ 10040][I][WiFiMulti.cpp:139] run(): [WIFI] 5 networks found
[ 10041][I][WiFiMulti.cpp:263] run(): [WIFI] Connecting BSSID: 64:6E:EA:2D:3B:0B SSID: wirerozman Channel: 11 (-66)
[ 10044][W][STA.cpp:548] disconnect(): STA already disconnected.
[ 10130][I][WiFiMulti.cpp:283] run(): [WIFI] Connecting done.
00:00:10 WiFI_info:   Connecting WiFi...
00:00:10 WiFI_info:   mDNS name: MiniWebRadio
00:00:10 setup: ....  connected to wirerozman, IP address is 192.168.1.10
00:00:10 ftpServer:   Buffers allocated: 4622 bytes
00:00:10 rtime_info:  Initializing SNTP
01:00:10 rtime_info:  Waiting for system time to be set... (1/10)
11:56:07 rtime_info:  11:56:07
11:56:07 audiotask:   is pinned to core 0
11:56:07 audiotask:   priority is 2
11:56:07 AUDIO_info:  commFMT = Philips
11:56:07 setup: ....  Number of saved stations: 14
11:56:07 setup: ....  current station number: 6
11:56:07 setup: ....  current volume: 7
11:56:07 setup: ....  last connected host: http://macslons-irish-pub-radio.stream.laut.fm/macslons-irish-pub-radio
11:56:07 setup: ....  connection timeout: 500 ms
11:56:07 setup: ....  connection timeout SSL: 2000 ms
11:56:07 setup: ....  volume is muted: (from 7)
11:56:08 WiFI_info:   RSSI is -68 dB
11:56:08 AUDIO_info:  file "/common/s/level_bar.jpg" not found
[ 12380][I][ArduinoOTA.cpp:141] begin(): OTA server at: MiniWebRadio.local:3232
11:56:08 action: ...  switch to station 6
11:56:08 AUDIO_info:  Connect to new host: "http://macslons-irish-pub-radio.stream.laut.fm/macslons-irish-pub-radio"
11:56:08 AUDIO_info:  PSRAM not found, inputBufferSize: 13951 bytes
11:56:08 AUDIO_info:  buffers freed, free Heap: 84120 bytes
11:56:08 AUDIO_info:  Connection has been established in 49 ms, free Heap: 83128 bytes
11:56:08 bitRate:     128Kbit/s
11:56:08 StationName: Macslons Irish Pub-Radio
11:56:08 icy-url: ..  http://laut.fm/macslons-irish-pub-radio
11:56:08 icy-descr:   Celtic Rock, Celtic Punk bis hin zum traditionellen Irish Folk.
11:56:08 AUDIO_info:  MP3Decoder has been initialized, free Heap: 58632 bytes , free stack 5120 DWORDs
11:56:08 lastURL: ..  http://macslons-irish-pub-radio.stream.laut.fm/macslons-irish-pub-radio
11:56:08 ftpServer:   Ftp server waiting for connection on port 21
11:56:09 Audiocodec:  MP3
11:56:09 AUDIO_info:  stream ready
11:56:09 AUDIO_info:  syncword found at pos 0
11:56:09 AUDIO_info:  Channels: 2
11:56:09 AUDIO_info:  SampleRate: 44100
11:56:09 AUDIO_info:  BitsPerSample: 16
11:56:09 AUDIO_info:  BitRate: 128000
11:56:09 StreamTitle: 100 Raisons - Sang Mélé
11:56:14 WiFI_info:   RSSI is -63 dB

ESP32 Chip: ESP32-D0WD-V3
Arduino Version: 2.0.14
ESP-IDF Version: 4.4.5
MiniWebRadio     Version 3.1   May 29/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 4194304 bytes, speed 40 MHz
CPU speed 240 MHz
SDMMC speed 80 MHz
TFT speed 27 MHz
PSRAM not found! MiniWebRadio doesn't work properly without PSRAM!RESET_REASON: Reset due to power-on event

00:00:00
00:00:00        ***************************
00:00:00        *     MiniWebRadio V3     *
00:00:00        ***************************
00:00:00
00:00:00 setup: ....  Arduino is pinned to core 1
00:00:00 tft_info: .  init ILI9341
00:00:00 setup: ....  Init SD card
00:00:00 setup: ....  SD card found, 3751.2 MB by 3789.0 MB free
00:00:00 setup: ....  seek for stations.csv
00:00:00 setup: ....  stations.csv found
00:00:00 setup: ....  seek for WiFi networks
[   757][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: wirerozman
[   765][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: wirerozman
[   770][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: wifirozman
00:00:07 setup: ....  5 WiFi networks found
00:00:07 setup: ....  wirerozman (-62)
00:00:07 setup: ....  mrhappy (-73)
00:00:07 setup: ....  AndroidAP8724 (-76)
00:00:07 setup: ....  VUKASIN (-85)
00:00:07 setup: ....  A1-B0936E (-86)
[  9254][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[  9255][I][WiFiMulti.cpp:119] run(): [WIFI] 5 networks found
[  9256][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 64:6E:EA:2D:3B:0B SSID: wirerozman Channel: 11 (-65)
[  9979][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:00:09 WiFI_info:   Connecting WiFi...
00:00:09 WiFI_info:   mDNS name: MiniWebRadio
00:00:09 setup: ....  connected to wirerozman, IP address is 192.168.1.10
[ 10013][I][ArduinoOTA.cpp:141] begin(): OTA server at: MiniWebRadio.local:3232
00:00:10 ftpServer:   Buffers allocated: 4622 bytes
00:00:10 rtime_info:  Initializing SNTP
01:00:10 rtime_info:  Waiting for system time to be set... (1/10)
01:00:12 rtime_info:  Waiting for system time to be set... (2/10)
01:00:14 rtime_info:  Waiting for system time to be set... (3/10)
11:42:23 rtime_info:  11:42:23
11:42:23 audiotask:   is pinned to core 0
11:42:23 audiotask:   priority is 2
11:42:23 AUDIO_info:  commFMT = Philips
11:42:23 setup: ....  Number of saved stations: 14
11:42:23 setup: ....  current station number: 6
11:42:23 setup: ....  current volume: 7
11:42:23 setup: ....  last connected host: http://macslons-irish-pub-radio.stream.laut.fm/macslons-irish-pub-radio
11:42:23 setup: ....  connection timeout: 1000 ms
11:42:23 setup: ....  connection timeout SSL: 2500 ms
11:42:24 action: ...  switch to station 6
11:42:24 AUDIO_info:  Connect to new host: "http://macslons-irish-pub-radio.stream.laut.fm/macslons-irish-pub-radio"
11:42:24 AUDIO_info:  PSRAM not found, inputBufferSize: 13951 bytes
11:42:24 AUDIO_info:  buffers freed, free Heap: 18276 bytes
11:42:24 AUDIO_info:  Connection has been established in 60 ms, free Heap: 17004 bytes
11:42:24 setup: ....  volume is muted: (from 7)
11:42:25 bitRate:     128Kbit/s
11:42:25 StationName: Macslons Irish Pub-Radio
11:42:25 icy-url: ..  http://laut.fm/macslons-irish-pub-radio
11:42:25 icy-descr:   Celtic Rock, Celtic Punk bis hin zum traditionellen Irish Folk.
[ 17247][E][mp3_decoder.cpp:1600] MP3Decoder_AllocateBuffers(): not enough memory to allocate mp3decoder buffers
11:42:25 AUDIO_info:  The MP3Decoder could not be initialized
11:42:25 WiFI_info:   RSSI is -63 dB
11:42:25 ftpServer:   Ftp server waiting for connection on port 21
11:42:26 WiFI_info:   RSSI is -69 dB
11:42:26 BT-Emitter:  KCX_BT_Emitter not found 
11:42:29 WiFI_info:   RSSI is -63 dB
11:42:31 DLNA_server: 0 media server found
11:42:39 WiFI_info:   RSSI is -62 dB

schreibfaul1, 2024-06-04:
The project is getting bigger and bigger. The limit of what is feasible for the ESP32 has already been exceeded. If you have an ESP32 or ESP32-S3 with PSRAM, a lot of dynamic memory is shifted to the PSRAM.

Namzor1955, 2024-06-04:
> Za: "schreibfaul1/ESP32-MiniWebRadio" ***@***.***>
> Poslano: Torek, 4. Junij 2024 20:57:19

Thanks for the quick reply. I was expecting something like that. 
I also built a version with esp32s3 (N16R8) with the same components, but there is a wifi connection problem. Esp32 was purchased via Aliexspres (https://www.aliexpress.com/item/1005006446406795.html?spm=a2g0o.order_list.order_list_main.16.ddf118027GXO2x). 
I also checked the settings in menuconfiguration and they are the same as your settings in the documentation. 
I am attaching the process and I am asking you if you can help me. 
Thanks for the reply 

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ESP32-S3-DevKitC-1-N16R8.html 
PLATFORM: Espressif 32 (2024.3.10+sha.348a508) > Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM) 
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash 

Connected! 
[ 414][I][esp32-hal-psram.c:92] psramInit(): PSRAM enab[ 1980][I][WiFiMulti.cpp:90] addAP(): [WIFI][APlistAdd] add SSID: wirerozman 
[ 7241][I][WiFiMulti.cpp:134] run(): [WIFI] scan done 
[ 7243][I][WiFiMulti.cpp:139] run(): [WIFI] 5 networks found 
[ 7244][I][WiFiMulti.cpp:263] run(): [WIFI] Connecting BSSID: 64:6E:EA:2D:3B:0B SSID: wirerozman Channel: 11 (-49) 
[ 7247][W][STA.cpp:548] disconnect(): STA already disconnected. 
[ 8305][W][STA.cpp:157] _onStaArduinoEvent(): Reason: 2 - AUTH_EXPIRE 
[ 8307][W][STA.cpp:548] disconnect(): STA already disconnected. 
[ 9329][W][STA.cpp:157] _onStaArduinoEvent(): Reason: 2 - AUTH_EXPIRE 
[ 9330][W][STA.cpp:548] disconnect(): STA already disconnected.

closed 2024-07-19 by github-actions[bot]