schreibfaul1/ESP32-MiniWebRadio #447: v3.1 fresh install problems

clear-sky, 2024-06-04:
Hello everyone
Because of the problems that I will say later, I thought that the esp32-s3 module was broken, so I changed it and installed another one that was completely new, but I encountered a very strange problem. After downloading the radio program, it connects to the Internet, but no sound is played. Viometer has no activity thus there is  a problem. I couldn't connect to the radio even with the browser. I pinged the radio's IP, only when the radio starts to working, it responds to the ping for a few moments, and then the connection is disconnected.
I saw that I can't do anything, I had to go back to version 2, **everything works fine there**, those who use version 3 with the new module will definitely get caught in this hell, especially if they are newbies!
I think the lack of default files in nvs is the cause of this problem
Well, now it's time to tell what problems I encountered. Due to the device monitor not working properly in version 3 #402 and #418, I couldn't see the problem. I haven't found a solution for it yet. I don't know how many other people have this problem.

Solar320, 2024-06-05:
I can't get an connection to my ASUS Router anymore, first I thought it relies to my ESP32S3 with faulty w-lan but then I tryed it with the good ESP32S3 and there it also do not work anymore. With the old Radio there was no problem.
For some reason It is working when I connect to my MobilePhone AP.
I double checked all settings include the Password and SSID, all is correct.
The Miniradio opens an own AP when I set the password there it also do not work. Only when I use the Mobile....

Since I was not not able to find an usable method to make an flash backup from my ESP32S3 I do not have an working version now.... Also there is now an problem to compile the Backup of the old MiniradioVersion.  
There is something going wrong with all the libraries... I will look tomorrow if I find an older release of the radio at github.

clear-sky, 2024-06-06:
Hi Dear [Solar320](https://github.com/Solar320)
it is better to format your sd card to FAT32 - download latest [Content_on_SD_Card.zip](https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/Content_on_SD_Card.zip) and extract it in your sd card
set your username and password in [common.h](https://github.com/schreibfaul1/ESP32-MiniWebRadio/blob/master/src/common.h)
hope this help you!

Solar320, 2024-06-09:
This doesn't change anything. I format the sdcard, copy all files on it in original. Then the Radio open up an AP and also find my router.
When I put in my password it restarts and can not connect. I also try to put my password directly into the networs.csv, no difference.

I flashed an Antenna Test tool and it instantly connect to my router and send back the signal level over serial interface.
Like I said, the old Radio version will connect without problems.

I will clean the flash and flash again, when this also do not change something I will make an Backup of the source code and clone the whole radio again from Github, and try again with an clean install.
For the first time I try to only setup the SDCARD, not PCM and no Display. I should see the Infos over the serial IF.

18:05:28.353 -> entry 0x403c9880
18:05:29.081 -> ...........
18:05:34.085 -> WiFi connected
18:05:34.085 -> IP address: 
18:05:34.085 -> 192.168.50.25
18:05:34.085 -> 
18:05:34.085 -> - Signal Level: -61
18:05:36.044 -> - Signal Level: -60
18:05:38.045 -> - Signal Level: -60
18:05:40.081 -> - Signal Level: -61
18:05:42.083 -> - Signal Level: -60
18:05:44.047 -> - Signal Level: -60
18:05:46.092 -> - Signal Level: -59
18:05:48.079 -> - Signal Level: -60
18:05:50.087 -> - Signal Level: -60
18:05:52.084 -> - Signal Level: -58
18:05:54.051 -> - Signal Level: -54
18:05:56.093 -> - Signal Level: -58
18:05:58.053 -> - Signal Level: -59
18:06:00.054 -> - Signal Level: -59
18:06:02.075 -> - Signal Level: -59
18:06:04.084 -> - Signal Level: -59
18:06:06.054 -> - Signal Level: -59
18:06:08.083 -> - Signal Level: -58
18:06:10.091 -> - Signal Level: -58
18:06:12.057 -> - Signal Level: -59
18:06:14.079 -> - Signal Level: -60
18:06:16.059 -> - Signal Level: -54
18:06:18.095 -> - Signal Level: -50
18:06:20.078 -> - Signal Level: -53
18:06:22.109 -> - Signal Level: -54
18:06:24.065 -> - Signal Level: -52
18:06:26.064 -> - Signal Level: -51

Solar320, 2024-06-15:
Still the same....I opened an Guest Network at my ASU Router and sometimes randomly it will connect to it. 
To the Main W-Lan It do not connect. There must be any timing problem or something else. 

The compiling log was full of errors until I installed the Espressif-IDF manually.

I still get the error:

Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...

And via serial Monitor I get:

[WIFI][APlistAdd] no ssid or ssid too long

I still get an instant connection to my Mobile Phone HotSpot!  I will go back to an older vesion of this Radio if Available.

valera39, 2024-06-15:
[Solar320](https://github.com/Solar320)
Try changing the channel on your router. I also have an ASUS and had a similar problem. And with a TPlink router, for example, the radio connects the first time. You may also need to change the wifi tab settings for the ASUS

Solar320, 2024-06-15:
I had channel 13 and tried allready channel 12, without any change. I can try some other channels for testing ...
But I try the old radio ESP32-MiniWebRadio-V.2.15.0 and it is connecting. 
Any other sketch which uses the same ESP32S3 chip and router  is working without problems.

I have an Fusion V2 Board for the DTU on Battery, it has an ESP32N8R2. This one connects instantly.
I do not try to flash the new radio on the fusion board because I don't want to change anything.

Its extremly unlikely that the router is the problem when the same chip is working with any other Sketch. I think it has to do with changes into the sdkconfig.esp32s3.   Unfortunately the old config do not compile with the new radio version.

I would use the old Radio version instead of the new one, but it has problems with the HTML interface and also with the Touchscreen. Inverting the touchscreen do not work on this version and I have an V and H inverted Touch screen.

Now I will try to use my old V2 version, if I can manage to get it working to compile. I will wait until the wifi problem is fixed.

valera39, 2024-06-15:
[Solar320](https://github.com/Solar320)
I agree with you, my V2 connects with no problems either. Try channel 2, the third version connects to the network almost immediately. But in general, the problem does exist and maybe Mr. Wohl will pay attention to it. But I think he has no way to test the radio performance with different routers, and we should help him with that.
I don't want to change my Asus AC66U for another one, it works fine with other internet radio projects.

schreibfaul1, 2024-06-15:
I cannot reproduce the error. I have not changed anything in the WiFi area. But if older versions worked, you can try changing the Arduino version in platformpo.ini. Now 2.0.14 is set, you can go back to 2.0.8. The guest access of a router does not allow communication in the network, sets restrictions and only allows the connection to the Internet for certain ports.
Perhaps something can be changed on the router, WPA2, WPA2, with or without CCMP?
[image]

Solar320, 2024-06-15:
On my ASUS I can allow the communication to the normal w-lan.

I test my ESP with the ESPCAM WIFI Testprogram and it connect:

There must be some error in the libs.  I was now able to compile the older V2 version of the radio and it also do not connect to the router, before it does. 
So there must be some Update in the between that destroy this.
I now try the 2.0.8 framework, I know how to do so but I am not an pro in this area. I can only Try and error...

This ist the Output of the ESPCAM WIFI Testprogram

Project:  Wi-Fi Signalstärke messen mit dem ESP32-CAM Modul
Author:   Thomas Edlinger for www.edistechlab.com
Date:     Created 10.12.2020
Version:  V1.0
IDE:      Arduino IDE 1.8.13
 
Required Board (Tools -> Board -> Boards Manager...)
 - Board: esp32   by Espressif Systems   V1.0.4
*/

22:37:58.702 -> .
22:37:58.702 -> WiFi connected
22:37:58.702 -> IP address: 
22:37:58.702 -> 192.168.50.109
22:37:58.702 -> 
22:37:58.702 -> - Signal Level: -47
22:38:00.663 -> - Signal Level: -49
22:38:02.686 -> - Signal Level: -47
22:38:04.682 -> - Signal Level: -49
22:38:06.712 -> - Signal Level: -47
22:38:08.702 -> - Signal Level: -49
22:38:10.685 -> - Signal Level: -48
22:38:12.668 -> - Signal Level: -50
22:38:14.669 -> - Signal Level: -47
22:38:16.670 -> - Signal Level: -47
22:38:18.671 -> - Signal Level: -48
22:38:20.692 -> - Signal Level: -47
22:38:22.704 -> - Signal Level: -46
22:38:24.718 -> - Signal Level: -49
22:38:26.701 -> - Signal Level: -48
22:38:28.706 -> - Signal Level: -49

I go back to 2.0.8.  I reset 3 times and every time it connect to the guest network.
I now test, if the radio also will connect to the normal W-lan.
Note that there still many "no ssid or too long " errors are!

ESP32 Chip: ESP32-S3
Arduino Version: 2.0.8
ESP-IDF Version: 4.4.7
MiniWebRadio     Version 3.2 Jun 13/2024
ARDUINO_LOOP_STACK_SIZE 8192 words (32 bit)
FLASH size 16777216 bytes, speed 80 MHz
CPU speed 240 MHz
SDMMC speed 16 MHz
TFT speed 40 MHz
PSRAM total size: 8381319 bytes
RESET_REASON: Software reset via esp_restart

00:02:06
00:02:06        ***************************
00:02:06        *     MiniWebRadio V3     *    
00:02:06        ***************************
00:02:06
00:02:06 setup: ....  Arduino is pinned to core 1
00:02:06 tft_info: .  init ILI9341
00:02:07 setup: ....  Init SD card
00:02:07 setup: ....  SD card found, 30410.3 MB by 30436.5 MB free
00:02:08 setup: ....  seek for stations.csv
00:02:08 setup: ....  stations.csv found
00:02:08 setup: ....  seek for WiFi networks
[  2743][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: mySSID
[  2749][I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: A290_Guest
[  2784][E][WiFiMulti.cpp:55] addAP(): [WIFI][APlistAdd] no ssid or ssid too long
[  2784][E][WiFiMulti.cpp:55] addAP(): [WIFI][APlistAdd] no ssid or ssid too long
[  2793][E][WiFiMulti.cpp:55] addAP(): [WIFI][APlistAdd] no ssid or ssid too long
00:02:14 setup: ....  5 WiFi networks found
00:02:14 setup: ....  A920 (-50)  <<<<<<<<<<<<<<<<normal Network
00:02:14 setup: ....  A290_Guest (-52) <<<<<<<<<<<<<<<GUEST Network
00:02:14 setup: ....  A1-4a9861 (-85)
00:02:14 setup: ....  DreiNeo_2.4Ghz_2464 (-90)
00:02:14 setup: ....  A1-e9de19 (-93)
[ 14877][I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[ 14877][I][WiFiMulti.cpp:119] run(): [WIFI] 6 networks found
[ 14878][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 08:BF:B8:64:BD:19 SSID: A290_Guest Channel: 13 (-51)
[ 16398][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 2 - AUTH_EXPIRE <<<<<<NOTE THIS!!!!!!
[ 16658][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
00:02:22 WiFI_info:   Connecting WiFi...
00:02:22 WiFI_info:   mDNS name: MiniWebRadio
00:02:22 setup: ....  connected to A290_Guest, IP address is 192.168.50.109
00:02:22 ftpServer:   Buffers allocated: 4622 bytes
00:02:22 rtime_info:  Initializing SNTP
01:02:22 rtime_info:  Waiting for system time to be set... (1/10)
01:02:24 rtime_info:  Waiting for system time to be set... (2/10)
22:58:20 rtime_info:  22:58:20

Solar320, 2024-06-15:
OK: As I removed the guest network it now connect again to my normal network.

Normal the radio should look at the network entry first, which was into the common.h (my normal network) 
But It has choosen to connect everytime to my guest network. So i had to disable the guestnetwork first, then it connect to the 
normal network again. (which is into the networks.csv)
The 2.0.8 framework seems to be the Problem. 

I will now restart many times to test it again and then I will test the new framework 2.0.14 again...

Solar320, 2024-06-15:
> [Solar320](https://github.com/Solar320) I agree with you, my V2 connects with no problems either. Try channel 2, the third version connects to the network almost immediately. But in general, the [...]

At the moment I can't explain but now it works. I will now flash the second ESP32S3 to test it also.

Solar320, 2024-06-16:
1.The last Update removed the "SSID too long Message" The connection process is now very good. 

2.In the past I have to set the SDCard speed to 20 or 16Mhz to get an connection. Now I tryed 40Mhz and even 80Mhz.
With 16 or 20Mhz I had problems with performance with Radio V3 and also many readerrors, for example it couldn't read the picture /png/Button_Test_Yellow.png if I press the "?" Button .  
In the start phase there were no big problems but when the radio was playing it may take too long to get the pictures from the SD card and the error occurs. Now with 40 or 80Mhz this errors are gone.
Its an wonder that my 15cm long wires from the SD-Card to the ESP are working. 

3. At the moment I can only select 1-14 presets in my List, I have 25 in my list, when I select an title higher than 14 it will not play.
   It shows: 
[1793779][W][common.h:1159] GetImageSize(): file /logo/s/KISS FM.jpg not exists
[1793802][W][common.h:1159] GetImageSize(): file /logo/s/KISS FM.jpg not exists
19:47:28 Audiocodec:  MP3
[1835750][E][Preferences.cpp:483] getString(): nvs_get_str len fail: station_000 NOT_FOUND
19:48:10 action: ...  switch to station 0
[1835813][W][common.h:1159] GetImageSize(): file /logo/s/KISS FM.jpg not exists
[1835838][W][common.h:1159] GetImageSize(): file /logo/s/KISS FM.jpg not exists
19:48:10 webSrv: ...  File not found "/logo/KISS FM.jpg"

If I copy and past the link http://149.56.185.82:8131 in and start it manually it works.
In the old Radio V2 it was impossible to play this Kiss FM Stream. But also in V3 it is not stable enough but usable.

4. I found an 500-1000Kbit FLAC stream for testing the radio, it works better than the KISS FM stream.
  https://stream10.xdevel.com/audio17s976748-2218/stream/icecast.audio

Solar320, 2024-06-23:
Hello, I now installed EDs Radio and had again problems to connect to my router. It uses the ESP32 not ESP32-S3.

I found that it has an problem to use the channels 12 and 13.

I disabled the additionally feature "Auto select channel including channel 12, 13" in my router and I have instantly an connection.
The reason because of it connect instantly to my Mobilphone hotspot is, that it do not use these channels.

May be there is an option to switch the Country code of the WIFI for using in europe. May be over "MenueConfig".

closed 2024-08-07 by github-actions[bot]