schreibfaul1/ESP32-MiniWebRadio #448: screen update in ver 3.1

clear-sky, 2024-06-05:
Hi dear wolle!
this problem can be seen in ver 3 and 3.1
if I unplug usb to serial cable, screen update is stopped or very slow

schreibfaul1, 2024-06-05:
I don't know exactly what the cause could be. It may have something to do with OTA. What happens if you remove the _uoload_port_ from platformio. Or enter a port like COM1, USB0, ACM0... depending on the operating system.

clear-sky, 2024-06-05:
Hi dear wolle!
my os is linux
upload_port is removed before ,if i dont remove upload_port from platformio flashing of mcu stops with this error `message : Host MiniWebRadio.local Not Found`
changing this line to `upload_port     = /dev/ttyUSB* ` allows flashing but problem remains

closed 2024-07-20 by github-actions[bot]