schreibfaul1/ESP32-MiniWebRadio #450: Turn display off

BBrz, 2024-06-08:
Hello,
I was searching for an option to turn the display off but keep the radio playing.
There is such an option?

Thanks

schreibfaul1, 2024-06-09:
This function does not exist, but it is not difficult to make this yourself.

BBrz, 2024-06-29:
Hello @schreibfaul1 

I managed to solve it somehow.
I added two functions:
```
void turn_off_display() {
    clearAll();
    setTFTbrightness(0);
    SerialPrintfln("turn display off");
    dispHeader.hide();
    dispFooter.hide();
    display_on = false;
}

void turn_on_display() {
    clearAll();
    setTFTbrightness(_brightness);
    showLogoAndStationName(true);
    dispHeader.show();
    dispFooter.show();
    display_on = true;
}
```
In `void ir_key` and `void tp_released` I added:
`if(!display_on){ turn_on_display(); inactivity_timer = 0; return;}`

Here: `if(_f_1sec) { // calls every second`
```
    inactivity_timer++;

    if (inactivity_timer >= 180) { 
        if (display_on) {
            turn_off_display();
        }
    }
```
Now if I wake up the radio by pressing one of the buttons on the remote it turns on the display but without the song name and the Vu-meter shows just the lit segments (attached photo - [image] ).
If I wake it up by touch, after a second or two, it shows everything as it should.

Any idea what is wrong?

Thank you!

BBrz, 2024-06-29:
**Solved**
I changed to this, since I'm interested just in Clock and Radio:
```
    void turn_on_display() {
    clearAll();
    setTFTbrightness(_brightness);
    if(_state == CLOCK) {
        changeState(CLOCK);
        dispHeader.show();
        dispFooter.show();
        inactivity_timer = 0;
        } else {
        changeState(RADIO);
        dispHeader.show();
        dispFooter.show();
        VUmeter_RA.show();
        txt_RA_sTitle.setText("");
        txt_RA_sTitle.show();
        _f_newStreamTitle = true;
        showLogoAndStationName(true);
        inactivity_timer = 0;
        } 
    
    
    display_on = true;
}
```
and everything is ok.

closed 2024-06-29 by BBrz